ros-navigation/navigation2 #2714: Move nav2_core plugin implementation out of base packages

SteveMacenski, 2021-11-30:
- waypoint follower task executors --> `nav2_waypoint_task_executors`
- controller progress and goal checker plugins --> `nav2_goal_checkers` & `nav2_progress_checkers`
- costmap layers --> `nav2_costmap_2d_layers`
- recovery (behaviors) --> `nav2_behavior_plugins`

jwallace42, 2022-04-20:
So we would separate out the server from the plugins. That would be great!

SteveMacenski, 2022-04-21:
yeah, my concern though is adding upwards of 6 new packages. While it would be not a big issue from a user perspective, this repository is starting to already get a little long on the browser / directory viewer from the packages and this would increase that count a bit.

Might be an excuse to do a superficial clustering of capabilities by directory to group more in a 'tree' format than a 'flat' format we have now, if that makes sense.

The major reason I haven't just done this already myself is I don't really know what I think is best: what we have now, adding 6 packages, or adding the packages and reorganizing the project directories a little bit. 

What do you think? Maybe I should make this decision based on user / developer consensus.

jwallace42, 2022-04-21:
I think the tree directory structure with separate packages for plugins would be the best. I developer consensus would be good. I can think of a bunch of different ways to break it up! I am not really sure which is the best.

SteveMacenski, 2022-04-26:
Kind of the same boat I'm in, so that's why I haven't taken action yet. I figured I'd wait until I settled on something or there was a consensus vs "just trying something" and making people's lives difficult with multiple reorgs.

SteveMacenski, 2022-05-02:
For now, I think we should just leave as is until we have a strong consensus of what change makes most sense. While I don't love the status quo, its better than a reorg that makes things even worst. For now, all of the algorithms are listed on https://navigation.ros.org/plugins/index.html so there is still a central index to find them all. I think maybe we should wait another year or two when we have some more packages around and maybe once we have a critical mass of algorithms there will be a clear grouping we should consider.

closed 2022-05-02 by SteveMacenski