ros-navigation/navigation2 #2716: Issue building nav2 from binaries for ROS rolling

klewisBE, 2021-11-30:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 20.04 on Mac M1 through parallels
- ROS2 Version:
  Rolling binaries
- Version or commit hash:
  main

#### Steps to reproduce issue
I started a new VM through parallels and installed rolling as described in the [docs](https://docs.ros.org/en/rolling/Installation/Ubuntu-Development-Setup.html). Then i ran talker and listener to confirm it worked. Then I followed the main branch development directions for [nav2](https://navigation.ros.org/build_instructions/index.html). After running `colcon build --symlink-install` i get this error

```
---stderr: nav2_behavior_tree                                                                               
/usr/bin/ld: CMakeFiles/test_action_spin_action.dir/test_spin_action.cpp.o: in function `rclcpp_action::Server<nav2_msgs::action::Spin>::SharedPtr rclcpp_action::create_server<nav2_msgs::action::Spin>(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeWaitablesInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp_action::Server<nav2_msgs::action::Spin>::GoalCallback, rclcpp_action::Server<nav2_msgs::action::Spin>::CancelCallback, rclcpp_action::Server<nav2_msgs::action::Spin>::AcceptedCallback, rcl_action_server_options_t const&, std::shared_ptr<rclcpp::CallbackGroup>)':
test_spin_action.cpp:(.text._ZN13rclcpp_action13create_serverIN9nav2_msgs6action4SpinEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE[_ZN13rclcpp_action13create_serverIN9nav2_msgs6action4SpinEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE]+0x2b4): undefined reference to `rclcpp_action::ServerBase::ServerBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_server_options_t const&)'
/usr/bin/ld: ../../../libnav2_spin_action_bt_node.so: undefined reference to `rclcpp_action::ClientBase::ClientBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_client_options_t const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/plugins/action/CMakeFiles/test_action_spin_action.dir/build.make:265: test/plugins/action/test_action_spin_action] Error 1
make[1]: *** [CMakeFiles/Makefile2:1876: test/plugins/action/CMakeFiles/test_action_spin_action.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/test_bt_action_node.dir/test_bt_action_node.cpp.o: in function `rclcpp_action::Server<test_msgs::action::Fibonacci>::SharedPtr rclcpp_action::create_server<test_msgs::action::Fibonacci>(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeWaitablesInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp_action::Server<test_msgs::action::Fibonacci>::GoalCallback, rclcpp_action::Server<test_msgs::action::Fibonacci>::CancelCallback, rclcpp_action::Server<test_msgs::action::Fibonacci>::AcceptedCallback, rcl_action_server_options_t const&, std::shared_ptr<rclcpp::CallbackGroup>)':
test_bt_action_node.cpp:(.text._ZN13rclcpp_action13create_serverIN9test_msgs6action9FibonacciEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE[_ZN13rclcpp_action13create_serverIN9test_msgs6action9FibonacciEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE]+0x2b4): undefined reference to `rclcpp_action::ServerBase::ServerBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_server_options_t const&)'
/usr/bin/ld: CMakeFiles/test_bt_action_node.dir/test_bt_action_node.cpp.o: in function `rclcpp_action::Client<test_msgs::action::Fibonacci>::SharedPtr rclcpp_action::create_client<test_msgs::action::Fibonacci>(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeWaitablesInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<rclcpp::CallbackGroup>, rcl_action_client_options_t const&)':
test_bt_action_node.cpp:(.text._ZN13rclcpp_action13create_clientIN9test_msgs6action9FibonacciEEENS_6ClientIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeGraphInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_INS9_13CallbackGroupEERK27rcl_action_client_options_t[_ZN13rclcpp_action13create_clientIN9test_msgs6action9FibonacciEEENS_6ClientIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeGraphInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES8_INS9_13CallbackGroupEERK27rcl_action_client_options_t]+0x218): undefined reference to `rclcpp_action::ClientBase::ClientBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_client_options_t const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/plugins/action/CMakeFiles/test_bt_action_node.dir/build.make:264: test/plugins/action/test_bt_action_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:1904: test/plugins/action/CMakeFiles/test_bt_action_node.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_behavior_tree [4.36s, exited with code 2]
Aborted  <<< nav2_costmap_2d [1.38s]
```

#### Expected behavior

#### Actual behavior

#### Additional information

SteveMacenski, 2021-11-30:
I just updated rolling binaries and built from scratch locally and seems to be working fine, as well as our CI which is using the same setup is turning over OK, so I don't think the issue is the codebase, but perhaps something with your setup.

Can you try to update and upgrade everything, clean your workspace (e.g. remove build/install/log directories) and rebuild without anything cached? My guess is you updated your rolling binaries that had an ABI breaking change and you need to recompile your workspace again without cached results due to changes.

klewisBE, 2021-11-30:
I re did it and it did not work.  After running this command 
`rosdep install -y -r -q --from-paths src --ignore-src --rosdistro rolling` 
I get an error
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: Cannot locate rosdep definition for [slam_toolbox]
Continuing to install resolvable dependencies...`
and the ROS1 bridge failed building ros_rolling with this error message
 `Failed to find ROS 1 roscpp, skipping...`

SteveMacenski, 2021-11-30:
That's not related. SLAM Toolbox is not in rolling as a binary package (nor is nav2 for that matter). Does the building of the package actually fail now? What you posted above really seems to me like a rolling update + caching issue to me.

klewisBE, 2021-12-01:
I followed [these](https://docs.ros.org/en/rolling/Installation/Maintaining-a-Source-Checkout.html) directions for updating rolling. Then rebuilt nav2 as the directions say and got a very similar error. A note is that if I source just ~/ros2_rolling/install/setup.bash the build cannot find bondcpp.cmake until /opt/ros/rolling/setup.bash is also sourced. 

The similar error is:
```
--- stderr: nav2_behavior_tree                                                                                                
/usr/bin/ld: CMakeFiles/test_action_spin_action.dir/test_spin_action.cpp.o: in function `rclcpp_action::Server<nav2_msgs::action::Spin>::SharedPtr rclcpp_action::create_server<nav2_msgs::action::Spin>(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeWaitablesInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp_action::Server<nav2_msgs::action::Spin>::GoalCallback, rclcpp_action::Server<nav2_msgs::action::Spin>::CancelCallback, rclcpp_action::Server<nav2_msgs::action::Spin>::AcceptedCallback, rcl_action_server_options_t const&, std::shared_ptr<rclcpp::CallbackGroup>)':
test_spin_action.cpp:(.text._ZN13rclcpp_action13create_serverIN9nav2_msgs6action4SpinEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE[_ZN13rclcpp_action13create_serverIN9nav2_msgs6action4SpinEEENS_6ServerIT_E9SharedPtrESt10shared_ptrIN6rclcpp15node_interfaces17NodeBaseInterfaceEES8_INSA_18NodeClockInterfaceEES8_INSA_20NodeLoggingInterfaceEES8_INSA_22NodeWaitablesInterfaceEERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS6_12GoalCallbackENS6_14CancelCallbackENS6_16AcceptedCallbackERK27rcl_action_server_options_tS8_INS9_13CallbackGroupEE]+0x2b4): undefined reference to `rclcpp_action::ServerBase::ServerBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_server_options_t const&)'
/usr/bin/ld: ../../../libnav2_spin_action_bt_node.so: undefined reference to `rclcpp_action::ClientBase::ClientBase(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rosidl_action_type_support_t const*, rcl_action_client_options_t const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [test/plugins/action/CMakeFiles/test_action_spin_action.dir/build.make:265: test/plugins/action/test_action_spin_action] Error 1
make[1]: *** [CMakeFiles/Makefile2:1876: test/plugins/action/CMakeFiles/test_action_spin_action.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_behavior_tree [4min 37s, exited with code 2]
Aborted  <<< nav2_costmap_2d [3min 5s]  
```

Are there any other ways I should try to get nav2 main to build?

SteveMacenski, 2021-12-01:
I cannot reproduce and no other external resource currently validates that this is an issue, so I can't offer much in the way of help (not to mention that Mac isn't a Tier 1 supported OS). 

You definitely shouldn't be sourcing both a source build and binaries of rolling, my guess is you've really messed up all of your paths and you really need to start over, slowly, and carefully to make sure you're doing that right thing, busting caches, and that your binaries are up to date. That's the best advice I can give you without significantly more technical detail.

rclcpp_action is definitely a dependency https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/CMakeLists.txt#L31 and is definitely being linked by the test https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/test/plugins/action/CMakeLists.txt#L6, so that screams to me either its related to Mac (compiler flags, compiler version, etc) or your paths are out of wack / not updated.

klewisBE, 2021-12-02:
I ended up getting it to build by creating a folder with the missing dependencies and then sourcing that folder. Im still not sure what I did to be missing these dependencies but it ended up being angles, BehaviorTree.CPP, bond_core, gazebo_ros_pkgs omple and vision_opencv. Im assuming something is still somewhat broken with my setup, but this I think is a working fix.

closed 2021-12-02 by klewisBE