ros-navigation/navigation2 #2719: [nav2_bringup] Simulation with 2 TB3 isn't working at all (multi_tb3_simulation_launch.py) - every Nav2 goal is rejected by server

ljaniec, 2021-12-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
  - ROS2 Galactic (binaries)

#### Steps to reproduce issue
- Source ROS, your workspace and start the script:

```
ros2 launch nav2_bringup multi_tb3_simulation_launch.py
```

- Click 'Startup' in both RViz for Nav2 stack and set initial poses for both TB3s.

- Try to give any sensible Nav2 goal to `robot1` or `robot2`

#### Expected behavior

Similar to these from single TB3 demo:
```
[rviz2-4] Start navigation
[rviz2-4] [INFO] [1638358971.369842831] [_]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[bt_navigator-11] [INFO] [1638358971.371246080] [bt_navigator]: Begin navigating from current location to (-1.15, -0.68)
[controller_server-8] [INFO] [1638358971.392980651] [controller_server]: Received a goal, begin computing control effort.
[controller_server-8] [WARN] [1638358971.393121450] [controller_server]: No goal checker was specified in parameter 'current_goal_checker'. Server will use only plugin loaded general_goal_checker . This warning will appear once.
[controller_server-8] [INFO] [1638358972.443702721] [controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1638358973.493587536] [controller_server]: Passing new path to controller.
[controller_server-8] [INFO] [1638358973.849973317] [controller_server]: Reached the goal!
[bt_navigator-11] [INFO] [1638358973.881616953] [bt_navigator]: Goal succeeded
```
Chosen TB3 navigates to the goal.

#### Actual behavior

Every goal is rejected by server, on both robots, every time.

```
(...)
[rviz2-4] Start navigation
[rviz2-4] [INFO] [1638358155.906285384] [robot1._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-4] [ERROR] [1638358155.908410790] [robot1._]: Goal was rejected by server
(...)
```

### Right now Navigation2 minimal demo of the simulation with multiple mobile robots with Nav2 stacks doesn't seem to work at all.

SteveMacenski, 2021-12-01:
I don't see that on my side,  things seem to be working correctly, except a quirk where sometimes the TF of one or the other robots isn't publishing so I can't get the lifecycle transitions to start up anyway to even reach that point. But once they're up, they're working fine.

SteveMacenski, 2021-12-03:
A good start would be to start debugging the issue and indicating what might be the problem

SteveMacenski, 2021-12-08:
Any more info / testing results?

KarnMatas, 2021-12-11:
until now I have got message "goal was rejected by server" too. my robot can navigate once but every  next goal was rejected.

ljaniec, 2022-01-19:
I have checked it after recent update and it isn't working still (both binaries for Ubuntu 20.04 and compiled from source from `galactic` branch). 
From rqt_graph, phase by phase:

- start without Nav2 startups:
[rosgraph_after_start_gazebo_no_nav2_stacks.pdf](https://github.com/ros-planning/navigation2/files/7897381/rosgraph_after_start_gazebo_no_nav2_stacks.pdf)

- only `robot2` with Nav2 stack with initPose given launched:
[rosgraph_after_start_robot2.pdf](https://github.com/ros-planning/navigation2/files/7897365/rosgraph_after_start_robot2.pdf)

- both robots have Nav2 stacks started with initPoses given:
[rosgraph_after_start_1_and_2_with_initPose.pdf](https://github.com/ros-planning/navigation2/files/7897305/rosgraph_after_start_1_and_2_with_initPose.pdf)

Saved terminal (cleared a bit) for actions:
 - `ros2 launch nav2_bringup multi_tb3_simulation_launch.py`
 - startup of `robot2` Nav2 stack
 - `initPose` for `robot2`
 - Nav2 goal given:
  ```
[rviz2-15] [INFO] [1642597504.499973379] [robot2._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-15] [ERROR] [1642597504.500514584] [robot2._]: Goal was rejected by server
```
-  startup of `robot1` Nav2 stack
- `initPose` for `robot1`
- Nav2 goal given to both robots:
```
[rviz2-15] [INFO] [1642597562.923893887] [robot2._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-15] [ERROR] [1642597562.924373425] [robot2._]: Goal was rejected by server
[rviz2-4] Start navigation
[rviz2-4] [INFO] [1642597567.179889265] [robot1._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-4] [ERROR] [1642597567.180410933] [robot1._]: Goal was rejected by server
```

When I have tried to find the error source, I have found these lines in terminal log:
```
[waypoint_follower-24] [FATAL] [1642597452.744755801] [robot2.waypoint_follower]: Can not get 'plugin' param value for wait_at_waypoint
[ERROR] [waypoint_follower-24]: process has died [pid 45587, exit code 255, cmd '/opt/ros/galactic/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower -r __ns:=/robot2 --params-file /tmp/tmpvn927bmw -r /tf:=tf -r /tf_static:=tf_static'].
...
[rviz2-15] Start navigation
[rviz2-15] [INFO] [1642597504.499973379] [robot2._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-15] [ERROR] [1642597504.500514584] [robot2._]: Goal was rejected by server
...
[waypoint_follower-13] [INFO] [1642597516.141306448] [robot1.waypoint_follower]: Configuring
[waypoint_follower-13] [FATAL] [1642597516.153151747] [robot1.waypoint_follower]: Can not get 'plugin' param value for wait_at_waypoint
[ERROR] [waypoint_follower-13]: process has died [pid 45413, exit code 255, cmd '/opt/ros/galactic/lib/nav2_waypoint_follower/waypoint_follower --ros-args -r __node:=waypoint_follower -r __ns:=/robot1 --params-file /tmp/tmpj1a15_ie -r /tf:=tf -r /tf_static:=tf_static'].
...
rviz2-15] Start navigation
[rviz2-15] [INFO] [1642597562.923893887] [robot2._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-15] [ERROR] [1642597562.924373425] [robot2._]: Goal was rejected by server
[rviz2-4] Start navigation
[rviz2-4] [INFO] [1642597567.179889265] [robot1._]: NavigateToPose will be called using the BT Navigator's default behavior tree.
[rviz2-4] [ERROR] [1642597567.180410933] [robot1._]: Goal was rejected by server
```
It seems like something in the `waypoint_follower` isn't configured correctly (plugin param value for `wait_at_waypoint`?).

Log of terminal:
[save_terminal.txt](https://github.com/ros-planning/navigation2/files/7897720/save_terminal.txt)

ljaniec, 2022-02-08:
The simulation started working after updating in the `nav2_bringup` package on the branch `galactic` the YAML files (`nav2_multirobot_params_1.yaml` and `nav2_multirobot_params_2.yaml` in the folder `params`) with:
```
waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
```
at the end (similar to the `nav2_params.yaml`).

I guess the binary version installed through `sudo apt-get install ros-galactic-nav2-bringup` has the same incorrect parameters for multirobot simulation there.

closed 2022-02-08 by SteveMacenski