ros-navigation/navigation2 #2723: `NavigateThroughPoses` stops early when incidentally passing over final pose

cadmus-to, 2021-12-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - `ros-galactic-navigation2	1.0.7-1focal.20211012.051615`
- DDS implementation:
  - (I'm not sure how to find this, it's whatever the default is for `galactic`)

#### Steps to reproduce issue
1. Follow [the example in the Getting Started page](https://navigation.ros.org/getting_started/index.html#running-the-example):
```bash
source /opt/ros/galactic/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/galactic/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py
```
2. Set a decent initial pose estimate
3. Change to navigate through poses mode
4. Set intermediate pose facing north at the other end of the space (straight ahead from start)
5. Set goal pose in the middle between the starting pose and the intermediate pose (facing the same way as the other two)

#### Expected behavior
Robot should travel all the way to the intermediate pose first (at the other end of the room), then turn around and travel back to the final pose, then turn back around.

#### Actual behavior
Robot travels towards intermediate pose, but upon passing over goal pose, it stops.

#### Additional information
Nothing yet.

SteveMacenski, 2021-12-03:
An image/video would be more illustrative, but sounds like a re-report of https://github.com/ros-planning/navigation2/issues/2622. 

Currently intersections between requests are going to have trouble. Consider using a different mode or happy to discuss a potential contribution to resolve.

cadmus-to, 2021-12-04:
Oh yes, sorry I couldn't find that issue before, I didn't try looking for "NavThroughPoses". I have updated the report with some screenshots. I think you're right, it's probably the same issue. (Happy to move the discussion over to that issue if you think that's more appropriate)

I'm not familiar with how Nav2 works yet (nor the controllers it uses) so I'm not sure how feasible this is, but maybe it's possible parameterising the path with something like the arc length, then track the progress of the parametrisation.

I _believe_ it's using `DWBLocalPlanner` based on the default params file [here](https://github.com/ros-planning/navigation2/blob/galactic/nav2_bringup/bringup/params/nav2_params.yaml).

SteveMacenski, 2021-12-04:
> I didn't try looking for "NavThroughPoses"

Probably my fault, that's the short-hand I use (as well as NHP as opposed to NTP for "To" vs "Through"). Moving over to the existing ticket would be great just to keep it all in one place. I already linked it so that there would be visibility that these were related in the future.

I'll close this one, but happy to continue in the existing ticket.

closed 2021-12-04 by SteveMacenski

mradams-bastian, 2021-12-07:
If I understand everything correctly, I would argue that this is not only a problem in how controllers handle overlapping paths, but it is also a consequence of the default goal checker used by the controller server being the Simple Goal Checker which determines that a goal has been reached solely based on whether the current pose of the robot matches the final pose of the path. If the final pose of the path happens to overlap the trajectory of the robot, the loop in controller_server.cpp ends at that moment and publishZeroVelocity() is called.

Maybe it would make more sense to have the Stopped Goal Checker be the default? Or even create a more sophisticated goal checker?

SteveMacenski, 2021-12-07:
Lets continue in the other ticket