ros-navigation/navigation2 #2728: How to improve the accuracy of AMCL?

iLanceS, 2021-12-06:
Is there someone could tell me how to improve the accuracy of AMCL?
Can i use Cartographer instead of AMCL and how?

SteveMacenski, 2021-12-06:
You'd have to tune it, see our configuration documentation https://navigation.ros.org/configuration/packages/configuring-amcl.html. If you're not super familiar with what all those parameters mean, you should consult Probabilistic Robotics to understand the formulation of AMCL. Even I do that when I have to tune AMCL to re-jog my memory.

In terms of using another localization system, **absolutely**! In your launch file, just swap out the AMCL node with whatever other node you'd prefer to use.

SteveMacenski, 2021-12-09:
Anything else we can answer here or does that meet you intent?

SteveMacenski, 2021-12-15:
Closing - this may be asked on ROS Answers for followup

closed 2021-12-15 by SteveMacenski