ros-navigation/navigation2 #2729: Costmap2D Static Layer is not suitable for outdoor traversability maps

roncapat, 2021-12-06:
This function inside the CostMap2D Static Layer discards all non-lethal obstacle values from input OccupancyGrid message. If Occupancy is used as traversability score (common scenario in outdoor usage) all the info carried by non-lethal obstacle cells is flattened/discarded as "free space".  

https://github.com/ros-planning/navigation2/blob/ec6f70719f302ce0bc267f25d18302b53bacbd8e/nav2_costmap_2d/plugins/static_layer.cpp#L252-L267

I propose to let the user decide if this cut-off is what he needs. I can add a boolean parameter to enable/disable this behaviour if developers agree.

SteveMacenski, 2021-12-06:
> This function inside the CostMap2D Static Layer discards all non-lethal obstacle values from input OccupancyGrid message

That's not true, look at this function again: https://github.com/ros-planning/navigation2/blob/ec6f70719f302ce0bc267f25d18302b53bacbd8e/nav2_costmap_2d/plugins/static_layer.cpp#L265-L266

That only happens if you're in the trinary mode or the cost value is denoted as unknown. You may use the other costmap value types and set the lethal threshold at max in order to use any arbitrary values (its a common thing to do). 

For followup, please ask on ROS Answers, but I think you should be able to take it from there.

closed 2021-12-06 by SteveMacenski

roncapat, 2021-12-06:
Thank you, now it's more clear to me. I will try to tune parameters.