ros-navigation/navigation2 #2730: Can't open gazebo11,in tb3_simulation_launch.py

fantasyplus, 2021-12-07:
**operating system:** Ubuntu20.04
**ros disto:** rolling
**nav version:** main branch

[image]

[image]

I tried this demo with the virtual machine yesterday and encountered the same problem. I thought it's the virtual machine reason.
Today, I install the 20.04 physically. The result is the same,Gazebo can't be opened. And error is same,**[composed_bringup-9] [INFO] [1638861125.504970819] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist**

Can someone tell me why?

SteveMacenski, 2021-12-07:
I don't see this on my side. Search up your log files and see what's missing, my guess there's something else going on there, like Gazebo had other errors or wasn't able to load the robot into simulation. Do you see the TB3 and data being produced by it in Gazebo, irregardless of Nav2? You could very well just be missing the TB3 package / gazebo model paths.

SteveMacenski, 2021-12-09:
Have you figured out your issue?

SteveMacenski, 2021-12-15:
Closing from lack of response and not able to reproduce locally or in CI. Happy to discuss if the user comes back or on ROS Answers. Sounds like its a gazebo issue that is hasn't totally spawned or the something is crashing. Logging would indicate the second, seeing teh gazebo world would indicate the first

closed 2021-12-15 by SteveMacenski

Nathan24zz, 2021-12-21:
I also got the same error (the gazebo can't be opened when running **ros2 launch nav2_bringup tb3_simulation_launch.py**).
Error that show in terminal like **[local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist**.

**OS**: Ubuntu 20.04
**ros distro**: foxy

vaibhavkad-peppermint, 2022-01-22:
**I got the same error** 
Gazebo does not open when 
`run ros2 launch nav2_bringup tb3_simulation_launch.py`
**Error**: [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist

When opened separately Gazebo starts quickly

OS: Ubuntu 20.04
ros distro: foxy

[image]

irfanbadshah, 2022-04-22:
I have the same issue.. 

Any idea on how to fix it?
ros distro: galactic

Ryujiyasu, 2022-06-01:
Hi @miku54 you should check GAZEBO_MODEL_PATH, 
I checked below
GAZEBO_MODEL_PATH ☓ -> no working
GAZEBO_MODEL_PATH ○ -> working 

Ex.My Env
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/ubuntu/data/nav2_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models

Thank you!!

Ryujiyasu, 2022-06-01:
OS: Ubuntu 20.04
ros distribution: galactic

Ryujiyasu, 2022-06-02:
https://github.com/ros-planning/navigation2/issues/2757

miku54, 2022-06-02:
> #2757

I tried doing this, but still can't start gazebo from the command line in the terminal.       ><

miku54, 2022-06-02:
> #2757

Solved! I added an extra '--verbose' thanks! ありがとう！

KennySoh, 2022-07-20:
I had the same issue. I was trying to run the turtlebot3 simualtion with foxy. 
Seems like there something wrong with gazebo 

I managed to fix it by following the instruction below.

https://classic.gazebosim.org/tutorials?tut=ros2_installing&cat=connect_ros
```
source /usr/share/gazebo/setup.sh.  #this seems to fix gazebo
killall gzserver 
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py. #Able to start gazebo!
```

cliffe10086, 2024-11-22:
may you can check if the gzserver is already running,
$ ps aux | grep gzserver
if there are 2 msg(indicating there is already running)
$ pkill -f gzserver
then all right ,try run nav2 again