ros-navigation/navigation2 #2732: Obstacle and voxel layers does not update map at range more than 3.0 meters from sensor

KrasnovPavel, 2021-12-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Foxy (from apt)
- Version or commit hash:
  - ros-foxy-navigation2    0.4.7-1focal.20211014.203448
- DDS implementation:
  - Unknown

#### Steps to reproduce issue

I want to build costmap from my point cloud. 
But it does not update points if they are more than 3.0 meters from `robot_base_frame`.
I tried to set `obstacle_max_range` and `raytrace_max_range` to 10.0 but it didn't help.

My costmap params:
```
global_costmap:
  global_costmap:
    ros__parameters:
      update_frequency: 5.0
      publish_frequency: 5.0
      global_frame: map
      robot_base_frame: base_link
      use_sim_time: True
      robot_radius: 0.22
      width: 55
      height: 55
      origin_x: -15.0
      origin_y: -15.0
      resolution: 0.05
      track_unknown_space: true
      plugins: ["voxel_layer", "inflation_layer"]
      voxel_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
        enabled: True
        observation_sources: points points_clearing
        points:
          topic: /zed2/zed_node/point_cloud/cloud_registered
          sensor_frame: zed2_left_camera_frame
          min_obstacle_height: 0.3
          max_obstacle_height: 5.0
          obstacle_max_range: 10.0
          raytrace_max_range: 10.0
          clearing: False
          marking: True
          data_type: "PointCloud2"
          publish_voxel_map: True
        points_clearing:
          topic: /zed2/zed_node/point_cloud/cloud_registered
          sensor_frame: zed2_left_camera_frame
          min_obstacle_height: -2.0
          max_obstacle_height: 0.2
          obstacle_max_range: 10.0
          raytrace_max_range: 10.0
          clearing: True
          marking: False
          data_type: "PointCloud2"
          publish_voxel_map: True
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
        cost_scaling_factor: 3.0
        inflation_radius: 3.0
      always_send_full_costmap: True
  global_costmap_client:
    ros__parameters:
      use_sim_time: True
  global_costmap_rclcpp_node:
    ros__parameters:
      use_sim_time: True
```

[image]

I'm launchng nav2 using this command:
```ros2 launch nav2_bringup navigation_launch.py params_file:=./src/my_package/config/nav2_params.yaml```
[Full nav2_params.yaml](https://github.com/ros-planning/navigation2/files/7690710/nav2_params.txt)

#### Expected behavior

Detection of obstacles works at specified range.

#### Actual behavior

Obstacles are detected only at 3.0m range.

SteveMacenski, 2021-12-13:
https://github.com/ros-planning/navigation2/blob/foxy-devel/nav2_costmap_2d/plugins/obstacle_layer.cpp#L129

I assume what you think is 3.0 m is actually 2.5m as the default value.

You're using the wrong parameter name, in Foxy its `obstacle_range` and `raytrace_range`. The change to max / min was added in Galactic and newer. Please see the nav2_params.yaml file in the nav2_bringup package for your appropriate branch.

KrasnovPavel, 2021-12-14:
Oh, now I see it, thanks!

I using the wrong parameter name because it was listed here https://navigation.ros.org/configuration/packages/configuring-costmaps.html. It would be nice for this docs to say that some parameters has different names in some supported branches.

closed 2021-12-14 by KrasnovPavel

SteveMacenski, 2021-12-14:
Yes, that is an on going challenge. I hope in the short term future APIs will stabilize so that won’t be an issue, but in the interim, the docs are pointed towards the latest branch.  

I can definitely appreciate the frustration. With the skeleton crew of maintainers that exist today, we’re a bit limited to the degree of bespoke support we can offer for each distribution. I try to balance resources along new development and existing package support, but its never perfect.

YANGuanTing, 2021-12-21:
Sorry I have some like BUG!

I use cartographer load map and use navigation2-1.0.8, but static_layer can't inflation
[image]