ros-navigation/navigation2 #2739: Ackerman car can't reverse while navigating

miku54, 2021-12-15:
## Bug report
Hello, when I used The Ackermann model for navigation, THE target point was released behind the Ackermann car, which could not reverse, so I could not reach the target point. Which file should I change to make Ackerman back up?

**Required Info:**

- Operating System:
ubuntu18.04
- ROS2 Version:
eloquent
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior
Ackerman car reverses normally when navigating
#### Actual behavior

#### Additional information
Ackerman car can't reverse while navigating

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2021-12-15:
Please fill in the template properly and also provide more configuration information (e.g. what algorithms are you using? What are your settings for them? etc) Please provide at bare minimum your planner and controller algorithms & their configurations.

Have you read our documentation? https://navigation.ros.org/  Particularly the [Algorithm Selection](https://navigation.ros.org/plugins/index.html) guide which may provide more information about the algorithms that pair with the Ackermann model, in case you hadn't.

SteveMacenski, 2021-12-15:
Oh, note that there is no support for Ackermann in Eloquent. Eloquent is end of life, meaning it receives no maintenance time. The algorithms for Ackermann were added in Foxy and newer only.

miku54, 2021-12-16:
I'm using a DWB controller and my controller algorithm configuration file is：

miku54, 2021-12-16:
`amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.9
    alpha2: 0.1
    alpha3: 0.05
    alpha4: 0.01
    alpha5: 0.04
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.02
    pf_z: 0.85
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 2
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.02
    tf_broadcast: true
    transform_tolerance: 0.5
    update_min_a: 0.06
    update_min_d: 0.025
    z_hit: 0.7
    z_max: 0.001
    z_rand: 0.059
    z_short: 0.24

    # Initial Pose
    set_initial_pose: True
    initial_pose.x: 0.0
    initial_pose.y: 0.0
    initial_pose.z: 0.0
    initial_pose.yaw: 0.0

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_footprint
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    enable_groot_monitoring: True
    groot_zmq_publisher_port: 1666
    groot_zmq_server_port: 1667
    # 'default_nav_through_poses_bt_xml' and 'default_nav_to_pose_bt_xml' are use defaults:
    # nav2_bt_navigator/navigate_to_pose_w_replanning_and_recovery.xml
    # nav2_bt_navigator/navigate_through_poses_w_replanning_and_recovery.xml
    # They can be set here or via a RewrittenYaml remap from a parent launch file to Nav2.
[... 230 lines cut ...]
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_footprint
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 0.3
    min_rotational_vel: 0.05
    rotational_acc_lim: 0.3

robot_state_publisher:
  ros__parameters:
    use_sim_time: False

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200`

miku54, 2021-12-16:
Is there any way to make the Akeman's car drive backwards in the eloquent?

SteveMacenski, 2021-12-16:
No, see my comments & doc links above.

SteveMacenski, 2021-12-16:
Closing - please update to a modern, supported distribution to make use of Ackermann algorithms outlined in the documentation.

closed 2021-12-16 by SteveMacenski