ros-navigation/navigation2 #2740: no velocity topic data

AndreV84, 2021-12-15:
```
ros2 launch slamcore_slam slam_publisher.launch.py session_file:=/home/a/_slamcore/session21_out/2021-12-09T14\:04\:29+0200.session
ros2 launch slamcore_nav2 static_tf.launch.py
ros2 launch sick_scan2 sick_tim_5xx.launch.py
ros2 launch slamcore_nav2 navigation.launch.py
rviz2 slamcore_nav.rviz
```
as well as running a robot node which takes /cmd_vel as an input
from devs - no data on velocity topic

SteveMacenski, 2021-12-15:
Are those open source packages? If not, we can’t help debug your proprietary system. I have absolutely no idea what’s in any of those files.

closed 2021-12-15 by SteveMacenski

AndreV84, 2021-12-15:
@SteveMacenski from dev:
```
the main problem that cmd_vel is not published when goal_pose is set from rviz
all needed navigation nodes are launched
```
so the question is if the cmd_vel gets published if to set goal pose in rviz if using your default nav2 package ?

SteveMacenski, 2021-12-15:
This is not nearly enough information. That's basically saying that you added gasoline to a car, why isn't it going 60 miles per hour? There's alot of stuff going on between a command and an output velocity.

AndreV84, 2021-12-15:
@SteveMacenski Thank you for your reply. It is a good point. However, does the default non customized nav2 package supports publishing velocity upon setting up goal in rviz? devs told they got teleop mode working publishing the velocity, but when they try from rviz it won't work.
So is it supported on default stock nav2 package that it publishes the velocity topic ?

SteveMacenski, 2021-12-15:
The default package should work if you don't make any change to it, using the documented getting started guide https://navigation.ros.org/getting_started/index.html.

This is also extremely similar to the setup I use on a daily basis and know that things work - but you don't have to take my word for it, we have an extensive set of system tests that run nightly and on every PR that test every navigation function in simulation so we are confident at any given time that things are working properly. 

>  does the default non customized nav2 package support publishing velocity upon setting up goal in rviz

Yes, if you use the Nav2 pose setting tool, again, see our default configs in `nav2_bringup/rviz`