ros-navigation/navigation2 #2742: docs: Tag container image as something other than "main"

Aposhian, 2021-12-15:
## Feature request

#### Feature description
Currently the image used for CI and development comes from the `builder` target in the Dockerfile. It is designed as a base that has all of the dependencies and setup ready to build Nav2 source. However, I personally was confused by the tag `main`, thinking that I could use the published container images to run the latest and greatest versions of Nav2 from a container. This is not the case, and I had to look at the CI config to figure out why the image I downloaded had no executable artifacts inside it.

#### Implementation considerations
This sort of confusion could be resolved by:
- adding to the docs, clarifying that he container images are designed for CI and development, not runtime
- using more descriptive image names like `navigation2:builder` or `nav2-builder:main`

SteveMacenski, 2021-12-15:
@ruffsl is the best to give you a detailed explanation of that system

ruffsl, 2021-12-16:
The tags for the CI images where meant to denote the branch name that the image was derived from, the main tags originating from main branch of the GitHub repo. Presently our CI only runs for the main branch, but if we ever needed to extend it for other branches of previous releases, we would simply add the branch name as the tag for the respective CI image.

The CI images also do not include compiled workspaces to save on image size, to reduces the time it takes to pull images for each CI job run.

This is documented in my PR for adding CI docs, but I need to update most of it, as a bit has changed since I've last had the time to add to it.

Is there a noticable demand to host a docker image with compiled version of the main branch? I'd figure most eager users would settle with the latest binaries from Rolling, or developers would prefer to clone and build the source repo, using the Dockerfile if feeling lazy.

SteveMacenski, 2021-12-16:
I suppose we could make part of nightly to rebuild a user-focused docker image

Aposhian, 2021-12-16:
If there doesn't seem to be significant demand for a runnable container image, then it is probably not worth the trouble. My thought process went more like "Oh there is a hosted container image? I'll go ahead and run that!" without realizing that it was not designed for that use case, since most hosted container images are designed to run applications out of the box.

It makes a lot of sense to just keep the image for CI, but an addition to the docs could be helpful, even with an additional sentence on the README where it links to "Container Images".

SteveMacenski, 2021-12-16:
I believe they should work for development, but they're not what we're recommending use for deployments, but I don't think it would be that much effort (would it @ruffsl ) to run another Actions job that is specifically targeting external users?

> I'd figure most eager users would settle with the latest binaries from Rolling, or developers would prefer to clone and build the source repo, using the Dockerfile if feeling lazy.

Some yes, but alot of robotics companies are big Docker users so it might feel more natural for them to actually work with the image versus binaries.

ruffsl, 2021-12-16:
We could have a `latest` tag the points to an image building FROM `main` that includes nightly binary builds of Nav2. That is if you think there would be demand for such a bleeding edge image to regularly excitement with. Shouldn't be too hard if needed.

As for binary installs with released Ros distros, I think it would be better for end users to host and manage their own Dockerfile/images by building FROM the `ros:<distro>` hosted by the official docker hub library for production use cases. As installing Nav2 from apt doesn't take any extraordinary setup that would helpful to front load with hosted images here.

SteveMacenski, 2021-12-16:
Fair enough. I’d be fine closing this but I do want to hear @Aposhian ’s thoughts as to what would be most useful if they wanted to use a provided Docker container.

Aposhian, 2021-12-17:
As far as whether a experimental docker image or the rolling binaries are preferable, probably the rolling binaries are preferable. If I really want to play with the bleeding edge of `main` then I think it would be perfectly normal to expect to build it myself. So if the hosted image existed I might use it sometimes, yes, but I'm not requesting that it be added either. I would love to just make a small PR to the README clarifying that the container images are for CI and nav2 development primarily, and you shouldn't expect them to run out of the box. Unless a lot of other people want that use-case, then it probably isn't worth the overhead to support it.

SteveMacenski, 2021-12-20:
That sounds good to me!

SteveMacenski, 2022-01-11:
@Aposhian can you submit that PR?

Aposhian, 2022-01-11:
Yeah I can get that.

closed 2022-01-12 by SteveMacenski