ros-navigation/navigation2 #2749: BT-Navigator transition error

HarunTeper, 2021-12-17:
Hello,

I am currently working on a multi-robot simulation, you can find the package here:

https://github.com/HarunTeper/AuNa

When I try to launch multiple robots with the navigation stack (only localization and navigation, no slam), I get the following transition error:

[bt_navigator-26] [ERROR] [1639740019.328779724] []: Caught exception in callback for transition 10
[bt_navigator-26] [ERROR] [1639740019.328788944] []: Original error: Address already in use
[bt_navigator-26] [WARN] [1639740019.328804553] []: Error occurred while doing error handling.
[bt_navigator-26] [FATAL] [1639740019.328810123] [robot1.bt_navigator]: Lifecycle node bt_navigator does not have error state implemented

This error can be replicated by building the package and running the following command:

ros2 launch car_simulator scenario_multi_robot_racetrack.launch.py

What changes do I have to apply to my launch setup, so that the navigation components are properly setup?

SteveMacenski, 2021-12-17:
Please ask on ROS Answers and I’ll follow up there, that’s the appropriate place for config help. Please also include all of the logs. My guess is something just above the snippet you included is critical to understanding the error.

closed 2021-12-17 by SteveMacenski

HarunTeper, 2021-12-18:
Thanks, https://answers.ros.org/question/393029/bt-navigator-transition-error/