ros-navigation/navigation2 #2751: Static_layer can't inflation

YANGuanTing, 2021-12-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - galactic
- Version or commit hash:
  - navigation2-1.0.8
- DDS implementation:
  - Unknown

#### Steps to reproduce issue

I use cartographer load map and use navigation2-1.0.8, but static_layer can't inflation

My costmap params:
```
amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
[... 265 lines cut ...]
    use_sim_time: True

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
```

[image]

I'm launchng nav2 using this command:
```
ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
```

#### Expected behavior
map will inflation
#### Actual behavior
map inflation Incomplete 
#### Additional information

SteveMacenski, 2021-12-21:
Make sure your cartographer map has those cells marked as the occupied values. My guess is that they’re below that threshold and that’s what’s causing the issue. They appear occupied in rviz but the actual value is below the occupied threshold so they aren’t picked up using your current map server and map.yaml file settings. 

This sounds like a configuration issue of cartographer and/or map server.

YANGuanTing, 2021-12-22:
> Make sure your cartographer map has those cells marked as the occupied values. My guess is that they’re below that threshold and that’s what’s causing the issue. They appear occupied in rviz but the [...]

Thanks!! I try it.

closed 2022-01-10 by SteveMacenski

YANGuanTing, 2022-03-16:
HI @SteveMacenski 

I found this wrong solution, do u want modify ?

YANGuanTing, 2022-03-17:
Hi @SteveMacenski 

I modify static_layer.cpp function, setting lethal_threshold_ = 50, because  map topic publish Occupancy value 0 to 100 (0 free, 100 occupancy).
I rember map server save pgm and yaml file, use the same setting.

[image]
[image]