ros-navigation/navigation2 #2752: Make Lifecycle manager robust to the crash of a managed node

tuxnese, 2021-12-21:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04
- ROS2 Version:
Galactic binaries
- Version or commit hash:
ros-galactic-navigation2	1.0.7-1focal.20211012.051615

- DDS implementation:
Cyclone

#### Steps to reproduce issue

1) Run a set of managed nodes passing the "respawn=True" parameter

2) Run 'nav2_lifecycle_manager', in my scenario the managed nodes are:
lifecycle_nodes = ['controller_server',
                       'planner_server',
                       'recoveries_server',
                       'bt_navigator',
                       'waypoint_follower']

3) Get the pid of the process of one of the managed nodes

ps -A | grep planner_server

4) Kill the process

#### Expected behavior

The lifecycle manager gets a heartbeat timeout and restarts all the nodes, making the system working again. 

#### Actual behavior

The reset fails and the navigation does not work.

Here is my log

[ERROR] [planner_server-3]: process has died [pid 13642, exit code -15, cmd '/opt/ros/galactic/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server -r __ns:=/virtual_robot_5 --params-file /tmp/tmp0h27fliy --params-file /tmp/tmp6pjfl3ha --params-file /tmp/tmp72cf6i6t --params-file /tmp/tmpbsloxdxh --params-file /tmp/tmp7vpq0t0u --params-file /tmp/tmpl1zy647i --params-file /tmp/tmp33_23par -r cmd_vel:=input/navigation'].
[INFO] [planner_server-3]: process started with pid [14154]
[planner_server-3] [INFO] [1640075883.643358596] [virtual_robot_5.planner_server]: 
[planner_server-3] 	planner_server lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1640075883.644622698] [virtual_robot_5.planner_server]: Creating
[planner_server-3] [INFO] [1640075883.651607220] [virtual_robot_5.global_costmap.global_costmap]: 
[planner_server-3] 	global_costmap lifecycle node launched. 
[planner_server-3] 	Waiting on external lifecycle transitions to activate
[planner_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [INFO] [1640075883.652172825] [virtual_robot_5.global_costmap.global_costmap]: Creating Costmap
[lifecycle_manager-10] [INFO] [1640075893.631898096] [virtual_robot_5.lifecycle_manager_navigation]: Have not received a heartbeat from planner_server.
[lifecycle_manager-10] [ERROR] [1640075893.631941133] [virtual_robot_5.lifecycle_manager_navigation]: CRITICAL FAILURE: SERVER planner_server IS DOWN after not receiving a heartbeat for 10000 ms. Shutting down related nodes.
[lifecycle_manager-10] [INFO] [1640075893.631954617] [virtual_robot_5.lifecycle_manager_navigation]: Terminating bond timer...
[lifecycle_manager-10] [INFO] [1640075893.631964208] [virtual_robot_5.lifecycle_manager_navigation]: Resetting managed nodes...
[lifecycle_manager-10] [INFO] [1640075893.631972079] [virtual_robot_5.lifecycle_manager_navigation]: Deactivating waypoint_follower
[waypoint_follower-6] [INFO] [1640075893.632185050] [virtual_robot_5.waypoint_follower]: Deactivating
[waypoint_follower-6] [INFO] [1640075893.632224300] [virtual_robot_5.waypoint_follower]: Destroying bond (waypoint_follower) to lifecycle manager.
[lifecycle_manager-10] [INFO] [1640075893.834885499] [virtual_robot_5.lifecycle_manager_navigation]: Deactivating bt_navigator
[bt_navigator-5] [INFO] [1640075893.835374570] [virtual_robot_5.bt_navigator]: Deactivating
[bt_navigator-5] [INFO] [1640075893.835475187] [virtual_robot_5.bt_navigator]: Destroying bond (bt_navigator) to lifecycle manager.
[lifecycle_manager-10] [INFO] [1640075894.050208248] [virtual_robot_5.lifecycle_manager_navigation]: Deactivating recoveries_server
[recoveries_server-4] [INFO] [1640075894.050638321] [virtual_robot_5.recoveries_server]: Deactivating
[recoveries_server-4] [INFO] [1640075894.050734276] [virtual_robot_5.recoveries_server]: Destroying bond (recoveries_server) to lifecycle manager.
[lifecycle_manager-10] [INFO] [1640075894.698043561] [virtual_robot_5.lifecycle_manager_navigation]: Deactivating planner_server
[planner_server-3] [WARN] [1640075894.698354848] [rcl_lifecycle]: No transition matching 4 found for current state unconfigured
[planner_server-3] [ERROR] [1640075894.698383492] []: Unable to start transition 4 from current state unconfigured: Transition is not registered., at /tmp/binarydeb/ros-galactic-rcl-lifecycle-3.1.2/src/rcl_lifecycle.c:355
[lifecycle_manager-10] [ERROR] [1640075894.698855961] [virtual_robot_5.lifecycle_manager_navigation]: Failed to change state for node: planner_server
[lifecycle_manager-10] [ERROR] [1640075894.698893522] [virtual_robot_5.lifecycle_manager_navigation]: Failed to reset nodes: aborting reset

#### Additional information

I do not know if this is a bug report or a feature request. Anyway, I think that the lifecycle manager should be able to face the crash of a managed node.

tuxnese, 2021-12-21:
I have been able to make nav2 robust to the crash or kill of a controller in the following way:

1. Setting the parameter of the Node respawn=false in the launch file
2. Setting the parameter of the lifecycle_manager bond_timeout = 0.0
3. Using a RegisterEventHandler in a similar way to what is done in https://answers.ros.org/question/376101/ros2foxylaunch-restart-node-every-time-it-exits-launch-file/ to execute the following actions on exit
- start again the lifecycle node
- execute the "configure" transition of the lifecycle node
- execute the "activate" transition of the lifecycle node

SteveMacenski, 2021-12-21:
A contribution would be appreciated, but historically there hasn't been a respawn option in launch so this hasn’t come up before. I’m not sure its the role of the manager to bring things back up once they fail for safety concerns, but I’d be open to talking about it! 

If it shuts everything down due to a failure, it would make sense for an application to recycle things back up once the issue has been resolved so that your robot application system has control over knowing when its safe to move about. 

However, adding the auto-restart on failure sounds useful for those more “simple” systems and could be easily parameterized to enable / disable, so I’d be very open to that addition to the Lifecycle Manager.

SteveMacenski, 2022-04-11:
We run into a few issues here, see my PR for current status: https://github.com/ros-planning/navigation2/pull/2910

The main issue that during the deactivation phrase, we kill the bond connection on the server side, so we have no knowledge if any of the servers, let alone the server that died, is back up without manually trying to transition them which would be cludgy. 

I can get things to work on the lifecycle manager side of things in my PR but, I'm not sure how to get around this issue with the currently available interfaces.

I think I need to add a service vector for all of the `get_state`'s of the bond connected nodes. From that, I can call that service and it would only respond if the system was back online and use that as a proxy for if the servers are all alive again. I don't really care about the actual result they send me, more that I actually get a ping back.

SteveMacenski, 2022-04-12:
Got it working -- now working on the launch configurations for it which are giving me a little trouble. Should be able to handle that this afternoon. Take a look over at the PR now, should be good for reviews

SteveMacenski, 2022-04-18:
PR to be merged imminently. Thanks @tuxnese for bringing this to our attention!

closed 2022-04-18 by SteveMacenski

tuxnese, 2022-04-22:
Thank you for having taken care of it. I would have liked to contribute, but I did not have the time.