ros-navigation/navigation2 #2757: Can't open gazebo11 when run tb3_simulation_launch.py

Nathan24zz, 2021-12-28:
**OS**: Ubuntu 20.04
**distro**: foxy

I got the error (the gazebo can't be opened when I run **ros2 launch nav2_bringup tb3_simulation_launch.py**).
The error that shows in terminal like **[local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist.**

[image]

Can someone give me a solution?

SteveMacenski, 2021-12-28:
Please provide full logs. Does Gazebo come up eventually? If not, you may not have properly sourcing turtlebot gazebo files and if you’re not loading gazebo,  clearly there would be no odometry frame as there’s no simulation data being produced. 

See the getting started guide in our documentation.

Nathan24zz, 2021-12-30:
Here is the full logs
```
[INFO] [launch]: All log files can be found below /home/nathan/.ros/log/2021-12-30-12-23-43-167584-nathan-5809
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gzserver-1]: process started with pid [5811]
[INFO] [gzclient-2]: process started with pid [5813]
[INFO] [robot_state_publisher-3]: process started with pid [5815]
[INFO] [rviz2-4]: process started with pid [5817]
[INFO] [map_server-5]: process started with pid [5819]
[INFO] [amcl-6]: process started with pid [5821]
[INFO] [lifecycle_manager-7]: process started with pid [5823]
[INFO] [controller_server-8]: process started with pid [5825]
[INFO] [planner_server-9]: process started with pid [5828]
[INFO] [recoveries_server-10]: process started with pid [5831]
[INFO] [bt_navigator-11]: process started with pid [5833]
[INFO] [waypoint_follower-12]: process started with pid [5836]
[INFO] [lifecycle_manager-13]: process started with pid [5838]
[lifecycle_manager-7] [INFO] [1640841827.561925053] [lifecycle_manager_localization]: Creating
[lifecycle_manager-13] [INFO] [1640841827.562544869] [lifecycle_manager_navigation]: Creating
[robot_state_publisher-3] [WARN] [1640841827.563235234] [robot_state_publisher]: No robot_description parameter, but command-line argument available.  Assuming argument is name of URDF file.  This backwards compatibility fallback will be removed in the future.
[map_server-5] [INFO] [1640841827.784920775] [map_server]: 
[map_server-5] 	map_server lifecycle node launched. 
[map_server-5] 	Waiting on external lifecycle transitions to activate
[map_server-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-5] [INFO] [1640841827.785297130] [map_server]: Creating
[bt_navigator-11] [INFO] [1640841827.789274675] [bt_navigator]: 
[bt_navigator-11] 	bt_navigator lifecycle node launched. 
[bt_navigator-11] 	Waiting on external lifecycle transitions to activate
[bt_navigator-11] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-11] [INFO] [1640841827.789605138] [bt_navigator]: Creating
[waypoint_follower-12] [INFO] [1640841827.789642151] [waypoint_follower]: 
[waypoint_follower-12] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-12] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-12] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-12] [INFO] [1640841827.789793320] [waypoint_follower]: Creating
[recoveries_server-10] [INFO] [1640841827.796889303] [recoveries_server]: 
[recoveries_server-10] 	recoveries_server lifecycle node launched. 
[recoveries_server-10] 	Waiting on external lifecycle transitions to activate
[recoveries_server-10] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-6] [INFO] [1640841827.797786467] [amcl]: 
[amcl-6] 	amcl lifecycle node launched. 
[amcl-6] 	Waiting on external lifecycle transitions to activate
[amcl-6] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[amcl-6] [INFO] [1640841827.798032033] [amcl]: Creating
[planner_server-9] [INFO] [1640841827.801184927] [planner_server]: 
[planner_server-9] 	planner_server lifecycle node launched. 
[planner_server-9] 	Waiting on external lifecycle transitions to activate
[planner_server-9] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-8] [INFO] [1640841827.801810594] [controller_server]: 
[controller_server-8] 	controller_server lifecycle node launched. 
[controller_server-8] 	Waiting on external lifecycle transitions to activate
[controller_server-8] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-7] [INFO] [1640841827.833048307] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[lifecycle_manager-13] [INFO] [1640841827.834908976] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-7] [INFO] [1640841827.895971517] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-7] [INFO] [1640841827.896155240] [lifecycle_manager_localization]: Configuring map_server
[map_server-5] [INFO] [1640841827.896719626] [map_server]: Configuring
[map_server-5] [INFO] [map_io]: Loading yaml file: /opt/ros/foxy/share/nav2_bringup/maps/turtlebot3_world.yaml
[robot_state_publisher-3] Parsing robot urdf xml string.
[lifecycle_manager-13] [INFO] [1640841827.903072599] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-13] [INFO] [1640841827.903241594] [lifecycle_manager_navigation]: Configuring controller_server
[robot_state_publisher-3] Link base_link had 7 children
[robot_state_publisher-3] Link camera_link had 2 children
[robot_state_publisher-3] Link camera_depth_frame had 1 children
[robot_state_publisher-3] Link camera_depth_optical_frame had 0 children
[robot_state_publisher-3] Link camera_rgb_frame had 1 children
[robot_state_publisher-3] Link camera_rgb_optical_frame had 0 children
[robot_state_publisher-3] Link caster_back_left_link had 0 children
[robot_state_publisher-3] Link caster_back_right_link had 0 children
[robot_state_publisher-3] Link imu_link had 0 children
[... 131 lines cut ...]
[... cut ...]
[controller_server-8] [INFO] [1640841835.662288614] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
```
After that it keep repeating print `local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist`

Nathan24zz, 2021-12-30:
the gazebo does not come up, just stuck like this:
[image]

And for sourcing turtlebot gazebo, i just need `export TURTLEBOT3_MODEL` and `export GAZEBO_MODEL_PATH` isn't it? Or there is anything else that i forgot?

SteveMacenski, 2022-01-03:
Please refer to TB3 docs if using their resources, I can't give specific advice on that platform, but I have the following in the bashrc to get the models

```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/path/to/turtlebot3_simulations/turtlebot3_gazebo/models
```

If gazebo isn't coming up, that's the issue, its not in Nav2.

Nathan24zz, 2022-01-05:
I already follow your instruction and if I run the gazebo independently, it works. For example, here I run `worlds/pioneer2dx.world` like a tutorial in gazebo.
[image]
But when running `ros2 launch nav2_bringup tb3_simulation_launch.py`, it still produces the same error and the gazebo doesn't show up just like before.

SteveMacenski, 2022-01-10:
Please ask on ROS Answers or review documentation, we cannot reproduce this issue and it is well documented how to work with gazebo / ROS 2. This issue isn't related to Nav2.

closed 2022-01-10 by SteveMacenski

derkaputte, 2022-04-26:
@Nathan24zz Did you manage it? I've the same issue.
I think I have to reinstall my machine from beginning.

ljjhome, 2022-04-28:
I run into the same issue probably, here is how i solve it, though it may not be the same problem as you, but let us have a try
1. look at the pic below, it is from tb3_simulation_launch.py
[image]
it runs a gzserver, in my case it is this cmd block the whole process.
so open a new terminal, and run "gzserver --verbose -s libgazebo_ros_init.so -s libgazebo_ros_factory.so <waffle.modle file>"
then if you see it tells you to wait for gazebo models to be downloaded, then it is the problem.

just download the gazebo models from github, then install them, and export the path to environment variables. then you can successfully run gzserver, and the gzclient will be ok as well.

but there is still a problem in the waffle model file, it is not properly checked. just modify it like you want it to be.

jingle, 2022-06-18:
RUN
```
. /usr/share/gazebo/setup.bash
```
**BEFORE**
```
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models
```
will init gazebo successfully, just FYI

rageorgi, 2022-06-23:
Just sharing my experience, but waiting for around 8-10 minutes after following the instructions (https://navigation.ros.org/getting_started/index.html#running-the-example) works for me.

tsadarsh, 2022-10-27:
Waiting for 5-10 minutes loads up the models and then the Gazebo starts up as intended.

gnap114039, 2022-11-03:
> RUN
> 
> ```
> . /usr/share/gazebo/setup.bash
> ```
> 
> **BEFORE**
> 
> ```
> export TURTLEBOT3_MODEL=waffle
> export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models
> ```
> 
> will init gazebo successfully, just FYI

This solve my problem on ros2 humble. 
Before source "/usr/share/gazebo/setup.bash", running the following command which from the nav2 instruction https://navigation.ros.org/getting_started/index.html#running-the-example

```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False 
```
will always getting the following error:
```
[component_container_isolated-5] [INFO] [1667483986.201120372] [local_costmap.local_costmap]: Timed out waiting for transform from base_link to odom to become available, tf error: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
```

FYI.