ros-navigation/navigation2 #2758: Offer integration into a mission control system for application autonomy

SteveMacenski, 2022-01-03:
Examples:
- A Behavior Tree based system using the `NavigateToPose` higher level primitive to use the Nav2 stack in an autonomy application (e.g. BT.CPP)
- A state machine based system using wrapped clients to Nav2 (e.g. smach, smacc, etc)
- A sequential based system (e.g. @allenh1 )

SteveMacenski, 2022-04-04:
Yeah, the higher-level autonomy software, which could be one of a number of things depending on the application. Some examples of what I'm thinking are above including a BT-based solution (as you describe), a sequential task autonomy system (get a bunch of primitives of things to do and do them in order), or a scheduler (given a bunch of options and weights / objective functions, pick what to do next)

padhupradheep, 2022-06-09:
@SteveMacenski I understand this will be a bigger project by itself, are we targeting to start with this anytime this year? Or is it for the next year?

jwallace42, 2022-06-09:
Next year thing :)

mikeferguson, 2022-06-25:
Possibly consider integration with RMF instead of building another fleet planner? I've not used it, but https://github.com/open-rmf has some interesting stuff - such as traffic editor GUIs: https://github.com/open-rmf/rmf_traffic_editor