ros-navigation/navigation2 #2764: aws_robomaker_small_warehouse_world not found

tonynajjar, 2022-01-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Rolling Binaries
- Version or commit hash:
14051a6971efca0aaf51a5ef3df01b24e83b1098

- DDS implementation:
  - 

#### Steps to reproduce issue

```
ros2 launch nav2_simple_commander nav_to_pose_example_launch.py
```

#### Expected behavior
That it doesn't crash

#### Actual behavior
`LookupError: Could not find the resource 'aws_robomaker_small_warehouse_world' of type 'packages'
`
#### Additional information
Shouldn't aws_robomaker_small_warehouse_world be in [package.xml](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/package.xml)?

However when I tried to add it and run rosdep I got:

```
root@XMG-CORE-REN-M20:/opt/overlay_ws# rosdep install -y -r --from-paths src --ignore-src --rosdistro rolling --skip-keys "slam_toolbox"
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_simple_commander: Cannot locate rosdep definition for [aws_robomaker_small_warehouse_world]
```

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-01-13:
Its not released in binary form, you have to build yourself

SteveMacenski, 2022-01-14:
Note that https://github.com/ros-planning/navigation2/pull/2649 should fix this largely, but also blocked by some issues with the AWS warehouse releasers

SteveMacenski, 2022-01-18:
Can I close this? Its imminently being released and largely out of our direct control

tonynajjar, 2022-01-18:
Sorry for the late answer. Thanks for the clarification

closed 2022-01-18 by tonynajjar