ros-navigation/navigation2 #2766: RegulatedPurePursuitController Footprint Collision Bug

TipluJacob, 2022-01-13:
Since the last Galactic sync 5 (#2748) I have massive problems with the Regulated Pure Pursuit Controller. While the controller managed to pass through doors (or in general areas near obstacles) perfectly fine before the update, it's now getting stuck most of the time while spaming "RegulatedPurePursuitController detected collision ahead!". We observed this behaviour with a bigger robot than the turtlebot3, that now isn't able to pass through doors since the update. But we could reproduce the problem with the turtlebot3 too.
The following gif shows this behaviour:
[image]

## Bug report

**Required Info:**

- Operating System:
  Docker Ubuntu 20.04
- ROS2 Version:
  Galactic (installed via apt-get)
- Version or commit hash:
  ros-galactic-navigation2        1.0.8-1focal.20211223.041721
- DDS implementation:
  default

#### Steps to reproduce issue
There are 2 ways to reproduce this problem:
1) Use your favorite default setup for the turtlebot3 with the default nav params except that you use the RegulatedPurePursuitController instead of the DWB controller for the navigation and try to navigate close to obstacles (like shown in the gif). It's even better to observe if you try to navigate to narrow doors. For comparison, these are the parameters I used:
```
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False

bt_navigator:
[... 264 lines cut ...]
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: False

waypoint_follower:
  ros__parameters:
    loop_rate: 200
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

```
2) I created a repository to generate a default turtlebot3 setup. You could follow the steps described in the readme to reproduce the problem:
https://github.com/Robast-GmbH/ros_nav2_issue_git/tree/regulated_pure_pursuit_controller_issue

#### Expected behavior
If the robot starts close to an obstacle, it should be able to navigate past the obstacle. In other words: Just cleanly follow the global path.

#### Actual behavior
The robot get's stuck and fails to navigate the navigation goal.

SteveMacenski, 2022-01-13:
Is your footprint properly set? The only change in that sync was changing from point costs at the center to using the robot footprint given to the controller server as the basis of collision checking. Also make sure that your look ahead time is reasonable.

Try using the current state of RPP in the `main` branch, if you grab it from the repo it should build fine (just RPP). We added a change to it so that the time to look ahead for collision is bounded by the lookahead point distance and that may also largely resolve your problem if you still see it. The issue is we cannot backport that particular improvement because we change parameter names and defaults, so it would break other peoples' configurations.

TipluJacob, 2022-01-14:
Thank you for the quick response. The footprint for the turtlebot is properly set to the standard robot_radius 0.22. In our project it's a rectangle, but I double checked if it's properly set. Same for the look ahead time.

I tried using the main branch in our project, which worked fine. I tried as well using the old galactic version, which worked fine too. Only the updated galactic branch causes problems in our project. Honestly, it's kind of hard to reproduce the problems with the turtlebot reliable, which makes me kind of uncertain whether it's a bug in the current galactic version or I just didn't manage to use it properly. This is just my experience I can report so far.

SteveMacenski, 2022-01-14:
Can you test on the rectangular robot you reported the issue see on?

TipluJacob, 2022-01-17:
Yes I tested the issue on the rectangular robot. As I reported before, we have massive problems in passing through doors with the Galactic sync 5 version. The older galactic version on the other side works fine, the main branch also.

SteveMacenski, 2022-01-18:
OK, just to verify, you tested `main` on the rectangular robot and it worked? If so, then that's good the state in main is the most safe way. 

I'll revert the changes to Galactic from the most recent sync so that they're both up and running fine. I would though recomment moving to the `main` version as soon as possible with `Rolling` or `Humble`. It includes some more safe aspects, especially if you're working with a rectangular robot.

closed 2022-01-18 by SteveMacenski

TipluJacob, 2022-01-19:
Yes you stated everything correctly. We are definitely looking forward to move to the main version. Thanks for the advice and your  attention!