ros-navigation/navigation2 #2767: ComputePathToPose not working after update from foxy to galactic

KrasnovPavel, 2022-01-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic from apt
- Version or commit hash:
  - ros-galactic-navigation2 1.0.8-1focal.20211223.041721
- DDS implementation:
  - default

#### Steps to reproduce issue
I am launching 
`ros2 launch nav2_bringup navigation_launch.py params_file:=./src/planner/config/nav2_params_1.yaml` 
in foxy and in galactic (not simultaneously) with almost identical configs and then trigger action 
`ros2 action send_goal navigate_to_pose nav2_msgs/NavigateToPose "pose: { pose: { position: { x: 1.0 } } }"`. 
Foxy works correctly and robot is moving while galatcic logs 
`[bt_navigator-4] [INFO] [1642088149.265828484] [bt_navigator]: Begin navigating from current location to (1.00, 0.00)` 
and does nothing more.

In the same time at other terminal I am providing transforms: map->odom and odom->base_link.

My params file
```
bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: /odom
    bt_loop_duration: 10
    default_server_timeout: 20
    enable_groot_monitoring: True
    groot_zmq_publisher_port: 1666
    groot_zmq_server_port: 1667
    plugin_lib_names:
    - nav2_compute_path_to_pose_action_bt_node
    - nav2_compute_path_through_poses_action_bt_node
    - nav2_follow_path_action_bt_node
    - nav2_back_up_action_bt_node
    - nav2_spin_action_bt_node
    - nav2_wait_action_bt_node
    - nav2_clear_costmap_service_bt_node
    - nav2_is_stuck_condition_bt_node
    - nav2_goal_reached_condition_bt_node
    - nav2_goal_updated_condition_bt_node
    - nav2_initial_pose_received_condition_bt_node
    - nav2_reinitialize_global_localization_service_bt_node
    - nav2_rate_controller_bt_node
    - nav2_distance_controller_bt_node
    - nav2_speed_controller_bt_node
    - nav2_truncate_path_action_bt_node
    - nav2_goal_updater_node_bt_node
    - nav2_recovery_node_bt_node
    - nav2_pipeline_sequence_bt_node
    - nav2_round_robin_node_bt_node
    - nav2_transform_available_condition_bt_node
    - nav2_time_expired_condition_bt_node
    - nav2_distance_traveled_condition_bt_node
    - nav2_single_trigger_bt_node
    - nav2_is_battery_low_condition_bt_node
    - nav2_navigate_through_poses_action_bt_node
    - nav2_navigate_to_pose_action_bt_node
    - nav2_remove_passed_goals_action_bt_node
    - nav2_planner_selector_bt_node
    - nav2_controller_selector_bt_node
    - nav2_goal_checker_selector_bt_node

[... 661 lines cut ...]
[bt_navigator-4] [INFO] [1642089447.363154410] [bt_navigator]: Begin navigating from current location to (1.00, 0.00)
```

I am sure that is exactly `ComputePathToPose` BT-Node because I've added `bt_navigator.default_nav_to_pose_bt_xml` parameter pointing to my custom very simple tree:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
          <RecoveryNode number_of_retries="1" name="rec">
      		<PipelineSequence name="pipe">
          		<Spin spin_dist="-0.3" server_name="spin" server_timeout="10"/>
          		<ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
          		<Spin spin_dist="0.3" server_name="spin" server_timeout="10"/>
      		</PipelineSequence>
          	<Spin spin_dist="-0.3" server_name="spin" server_timeout="10"/>
        </RecoveryNode>
  </BehaviorTree>
</root>
```
Now I can see that robot only did first spin action.

#### Expected behavior
Computing path to pose.

#### Actual behavior
 ComputePathToPose silently does noting.

KrasnovPavel, 2022-01-14:
Thanks for your response!

> From your logs for Foxy and Galactic I can observe that the BT you are using for galactic and foxy are completely different.
> In the foxy log, you can note that different BT nodes that are not part of your BT are configured by the BT Navigator, such as FollowPath and ClearGlobalCostmap-Context. This is indeed not configured by the BT navigator for Galactic. Also, these behaviors are not configured in the BT attached by you.

I reexamined logs and it looks like galactic does not log bt_navigation configuration process at all. 
In my yaml file I tried to:
- not specify `bt_navigator.default_nav_to_pose_bt_xml`
- set it to "/opt/ros/galactic/share/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml"
- set it to my simple BT: "/home/notebook/work/ROS2-planner/src/planner/config/nav_to_pose.xml"
and in all cases I get in log only
```
[lifecycle_manager-6] [INFO] [1642144534.123876069] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-4] [INFO] [1642144534.124013902] [bt_navigator]: Configuring
```
while in foxy it is much more detailed:
```
[lifecycle_manager-6] [INFO] [1642086578.216939354] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-4] [INFO] [1642086578.217099191] [bt_navigator]: Configuring
[bt_navigator-4] [INFO] [1642086578.249554992] [bt_navigator_rclcpp_node]: Waiting for "compute_path_to_pose" action server
[bt_navigator-4] [INFO] [1642086578.249726056] [bt_navigator_rclcpp_node]: "ComputePathToPose" BtActionNode initialized
[bt_navigator-4] [INFO] [1642086578.250442107] [bt_navigator_rclcpp_node]: Waiting for "global_costmap/clear_entirely_global_costmap" service
[bt_navigator-4] [INFO] [1642086578.250518040] [bt_navigator_rclcpp_node]: "ClearGlobalCostmap-Context" BtServiceNode initialized
[bt_navigator-4] [INFO] [1642086578.253388677] [bt_navigator_rclcpp_node]: Waiting for "follow_path" action server
[bt_navigator-4] [INFO] [1642086578.253525401] [bt_navigator_rclcpp_node]: "FollowPath" BtActionNode initialized
[bt_navigator-4] [INFO] [1642086578.254252399] [bt_navigator_rclcpp_node]: Waiting for "local_costmap/clear_entirely_local_costmap" service
[bt_navigator-4] [INFO] [1642086578.254422363] [bt_navigator_rclcpp_node]: "ClearLocalCostmap-Context" BtServiceNode initialized
[bt_navigator-4] [INFO] [1642086578.255300009] [bt_navigator_rclcpp_node]: Waiting for "local_costmap/clear_entirely_local_costmap" service
[bt_navigator-4] [INFO] [1642086578.255396095] [bt_navigator_rclcpp_node]: "ClearLocalCostmap-Subtree" BtServiceNode initialized
[bt_navigator-4] [INFO] [1642086578.256146041] [bt_navigator_rclcpp_node]: Waiting for "global_costmap/clear_entirely_global_costmap" service
[bt_navigator-4] [INFO] [1642086578.256241174] [bt_navigator_rclcpp_node]: "ClearGlobalCostmap-Subtree" BtServiceNode initialized
[bt_navigator-4] [INFO] [1642086578.260113468] [bt_navigator_rclcpp_node]: Waiting for "spin" action server
[bt_navigator-4] [INFO] [1642086578.260256003] [bt_navigator_rclcpp_node]: "Spin" BtActionNode initialized
[bt_navigator-4] [INFO] [1642086578.263619358] [bt_navigator_rclcpp_node]: Waiting for "wait" action server
[bt_navigator-4] [INFO] [1642086578.263770860] [bt_navigator_rclcpp_node]: "Wait" BtActionNode initialized
```

> Also ComputePathToPose on a simpler terms is just for generating the path by the planner. 

I know it. I am using only `ComputePathToPose` to isolate the problem. I am trying do debug BT using my simple xml:
```
<root main_tree_to_execute="MainTree">
  <BehaviorTree ID="MainTree">
          <RecoveryNode number_of_retries="1" name="rec">
      		<Sequence name="pipe">
          		<Spin spin_dist="0.3" server_name="spin" server_timeout="10"/>
          		<ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>
          		<Spin spin_dist="-0.6" server_name="spin" server_timeout="10"/>
      		</Sequence>
          	<Spin spin_dist="-0.3" server_name="spin" server_timeout="10"/>
        </RecoveryNode>
  </BehaviorTree>
</root>
```
as far as I understand BTs it should do spin, compute path and do big counterspin or if something goes wrong do small counterspin.
I can see it does first spin but then does nothing at all.
```
[INFO] [launch]: All log files can be found below /home/notebook/.ros/log/2022-01-14-10-28-09-888672-notebook-Zephyrus-GX501GI-6304
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [6306]
[INFO] [planner_server-2]: process started with pid [6308]
[INFO] [recoveries_server-3]: process started with pid [6310]
[INFO] [bt_navigator-4]: process started with pid [6312]
[INFO] [waypoint_follower-5]: process started with pid [6314]
[INFO] [lifecycle_manager-6]: process started with pid [6316]
[waypoint_follower-5] [INFO] [1642145290.340269210] [waypoint_follower]: 
[waypoint_follower-5] 	waypoint_follower lifecycle node launched. 
[waypoint_follower-5] 	Waiting on external lifecycle transitions to activate
[waypoint_follower-5] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-5] [INFO] [1642145290.340913946] [waypoint_follower]: Creating
[... 109 lines cut ...]
[recoveries_server-3] [INFO] [1642145290.850595921] [recoveries_server]: Creating bond (recoveries_server) to lifecycle manager.
[lifecycle_manager-6] [INFO] [1642145290.952567829] [lifecycle_manager_navigation]: Server recoveries_server connected with bond.
[... cut ...]
[recoveries_server-3] [INFO] [1642145296.790504820] [recoveries_server]: Turning 0.30 for spin recovery.
[recoveries_server-3] [INFO] [1642145297.690690989] [recoveries_server]: spin completed successfully
```

### off-topic

Btw, do you know how to debug BTs properly? I tried to use groot in monitor mode but it crushes when any message is received. And it looks like it is not supported now anyway https://navigation.ros.org/migration/Galactic.html#dropping-support-for-live-groot-monitoring-of-nav2

padhupradheep, 2022-01-14:
> not specify bt_navigator.default_nav_to_pose_bt_xml
set it to "/opt/ros/galactic/share/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml"
set it to my simple BT: "/home/notebook/work/ROS2-planner/src/planner/config/nav_to_pose.xml"

I tried to reproduce your issue, but all the BT works as intended.

> and in all cases I get in log only

You are right, galactic does not show all the BT node configurations. But, they work as intended. 

> as far as I understand BTs it should do spin then compute path and then do big counterspin or if something goes wrong do small counterspin.

It does an initial spin, plans a path and finally does another spin. Are you sure, that you are not give a goal location in the position of the obstacle?

>  I've added bt_navigator.default_nav_to_pose_bt_xml parameter pointing to my custom very simple tree:

Make sure, you've done this correctly. Also, if you are using nav2_bringup I highly doubt there is an issue right now with it. As most of the features that I develop are tested with the launch from nav2_bringup. 

> Btw, do you know how to debug BTs properly? 

At the moment, I do not have an actual suggestion for the debugging. For me it is more based on the behavior that the robot exhibits and it's just on the fly. 

BTW, it would have been nice if you had asked  this question in ROS Answers. As, I think there is no actual bug with any of the BT nodes or BT itself.

SteveMacenski, 2022-01-18:
Please ask on ROS Answers, I'm pretty active on there. Lets keep the issue tracker for issues / features

closed 2022-01-18 by SteveMacenski