ros-navigation/navigation2 #2768: LIDAR timestamp on the message is earlier than all the data in the transform cache

billymccafferty, 2022-01-14:
I am trying to bringup Nav2 on a "server" Ubuntu machine hosting ROS2 Rolling, using messages published from a physical robot "client" via Raspberry Pi hosting ROS2 Foxy. The URDF description is at https://github.com/billymccafferty/xerbo .

When I bring up Nav2 via `ros2 launch nav2_bringup navigation_launch.py` on the server, I'm seeing the following error:

    [controller_server-1] [INFO] [1640913301.944847057] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'laser_frame' at time 1640913301.637 for reason 'the timestamp on the message is earlier than all the data in the transform cache'

I have chrony running on both the server and client with successful clock sync on the client. E.g., if I run ntpdate from the client to the server, I see the following, showing that they're almost perfectly synced:

    server 192.168.x.x, stratum 8, offset -0.000500, delay 0.03116
    30 Dec 18:26:02 ntpdate[7778]: adjust time server 192.168.x.x offset -0.000500 sec

Furthermore, if I echo /scan coming from the client and compare that to echo /tf on the server,  I see that the header timestamps are very close to one another; i.e., they're only milliseconds apart; e.g.,:

*/tf*

    transforms:
    - header:
        stamp:
          sec: 1640913147
          nanosec: 522616833
        frame_id: odom
    
*/scan*

    header:
      stamp:
        sec: 1640913147
        nanosec: 60040883
      frame_id: laser_frame

I can get everything working fine in the demos and such, but am running into this issue with Nav2 using my physical robot.

Thank you!

Billy McCafferty

SteveMacenski, 2022-01-14:
Please ask on ROS Answers, this is the wrong forum for configuration help

SteveMacenski, 2022-01-14:
I'd start with it all on the same CPU first and then branch out to make sure things work properly outright (e.g. your issue isn't straight up misconfiguration) then tackle the remote server problem.

closed 2022-01-14 by SteveMacenski

billymccafferty, 2022-01-14:
I appreciate this feedback Steve; I'll try that today. (I posted this a couple weeks ago on ROS Answers, but alas, no response. So I appreciate yours! :) Thx for all you're doing!

robvoi, 2022-03-15:
Did you have any luck finding the source of the problem?
Have the same issue. All search and posts didn‘t bring up any anything useful.

billymccafferty, 2022-03-15:
Hi @robvoi , unfortunately, I stalled out on this problem and my project is collecting a bit of dust at the moment. Hoping to muster the motivation to dig back in, in the coming month.

billymccafferty, 2023-06-29:
> Have you solved your problem @billymccafferty

No, I gave up after hitting this particular block. :)

Wuhu12351, 2025-08-11:
你应该尝试去检查一下自己的雷达时间戳和tf的时间戳，如果有问题那么需要做雷达和tf坐标变换的时间同步，如果这个完成了，还有问题，可以尝试查看一下自己发布tf坐标的地方的话题，是否有话题的频率 一直在降低，如果一直在降低说明你的程序存在bug,需要做修正
You should try to check your radar timestamp and TF timestamp, if there is a problem, then you need to do the time synchronization of radar and tf coordinate transformation, if this is done, there is still a problem, you can try to check the topic where you posted the tf coordinates, whether the frequency of the topic has been decreasing, if it has been decreasing, it means that there is a bug in your program, and it needs to be corrected