ros-navigation/navigation2 #2771: Recovery Backup Not Triggering

philmurp, 2022-01-14:
Hello, I am running Foxy install of ROS2/Nav2 stack with Mecanum, NVIDIA Jetson computer. I used Linorobot2 as install setup, [Navigation.yaml](github.com/linorobot/linorobot2/blob/1ebfae81ea741ac4d0ee6024eebe26c2f1e329f3/linorobot2_navigation/config/navigation.yaml): 

When my robot is stuck in a corner, I see it enters the recovery tree and does the spin and wait actions; however it has never performed the backup function and as a result it usually ends up aborting after a few tries and I have to restart the entire navigation stack.

Based on the Nav2 documentation it looks all pretty default for the parameters for recovery server so not sure why it wont try backup when it tries spin and wait pretty well

If it does abort is there any reason that I need to completely restart the entire Nav stack, rather than it being able to accept a new goal?

## Bug report

**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2022-01-14:
Your hyperlink doesn't render. 

Do you see logging associated with the backup, perhaps with an error saying that is was unable to due to a collision? 

I know in early days there was an issue where the default configuration file incorrectly in foxy said `back_up` instead of `backup` which messed with the topic names. Try `ros2 topic list` and see if you have both. If so, there was a misconfiguration

philmurp, 2022-01-14:
Updated the LINK so hopefully it shows up now. 

Looking at the file:

One line for plugins says:
`recovery_plugins: ["spin", "back_up", "wait"]`

Another line says:
```
    back_up:
      plugin: "nav2_recoveries/BackUp"
```

So should it be "BackUp" in both places instead of "back_up" in the recovery_plugins line?

To your other question: Have never seen backup even come up in the loggin, it shows spin and wait and then aborts, never seen backup in any logs.

SteveMacenski, 2022-01-14:
Does your behavior tree include a backup call? Many don't have it even. The backup would only be triggered if your behavior tree uses it. 

You should check the topics like I requested - I don't remember off hand which caps/underscores we used in Foxy. It's changed a few times.

philmurp, 2022-01-14:
Bottom of post is my full topic list, recoveries_server is the only relevant topic I believe, and in the .yaml file this is what is within that server:

As I mentioned above I see mismatch of "back_up" in the plugins vs below "BackUp"
```
recoveries_server:
  ros__parameters:
    costmap_topic: local_costmap/costmap_raw
    footprint_topic: local_costmap/published_footprint
    cycle_frequency: 10.0
    recovery_plugins: ["spin", "back_up", "wait"]
    spin:
      plugin: "nav2_recoveries/Spin"
    back_up:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2 
```

Full topic list:
```
ros2 topic list
/amcl/transition_event
/amcl_pose
/bt_navigator/transition_event
/camera/depth/color/points
/cmd_vel
/controller_server/transition_event
/cost_cloud
/diagnostics
/evaluation
/global_costmap/clearing_endpoints
/global_costmap/costmap
/global_costmap/costmap_raw
/global_costmap/costmap_updates
/global_costmap/footprint
/global_costmap/global_costmap/transition_event
/global_costmap/published_footprint
/goal_pose
/imu/data
/initialpose
/joint_states
/local_costmap/clearing_endpoints
/local_costmap/costmap
/local_costmap/costmap_raw
/local_costmap/costmap_updates
/local_costmap/footprint
/local_costmap/local_costmap/transition_event
/local_costmap/published_footprint
/local_plan
/map
/map_server/transition_event
/marker
/odom
/odom/unfiltered
/parameter_events
/particle_cloud
/particlecloud
/plan
/planner_server/transition_event
/received_global_plan
/recoveries_server/transition_event
/robot_description
/rosout
/scan
/set_pose
/tf
/tf_static
/transformed_global_plan
/waypoint_follower/transition_event
/zed/point_cloud/cloud_registered
```

SteveMacenski, 2022-01-14:
Does your behavior tree include a backup call? Many don't have it even. The backup would only be triggered if your behavior tree uses it.

Also, apologies, list the **actions** not the topics, these are action servers.

philmurp, 2022-01-14:
I believe the tree does have backup call.

--

 ros2 action list
/FollowWaypoints
/back_up
/compute_path_to_pose
/follow_path
/navigate_to_pose
/spin
/wait

SteveMacenski, 2022-01-14:
> I believe the tree does have backup call.

Check

SteveMacenski, 2022-01-14:
https://github.com/ros-planning/navigation2/tree/foxy-devel/nav2_bt_navigator/behavior_trees none of these in Foxy do it, so unless you've modified a file, it wouldn't in Foxy.

philmurp, 2022-01-14:
Based on this line in my code, and it being Foxy + the default file doesnt have backup, looks like you are correct that its not in the .xml

`default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"`

So would I have to copy the default .xml and modify locally and call to that to add backup? Any idea why backup wasn't added to these default files?

SteveMacenski, 2022-01-14:
We didn't think it appropriate for defaults at the time :shrug: It also could have been that I added backup after Foxy was initially released. I'm not exactly sure. 

> So would I have to copy the default .xml and modify locally 

Yup!

closed 2022-01-14 by SteveMacenski

philmurp, 2022-01-15:
Any example of .xml files with backup included (perhaps Galactic version?)

SteveMacenski, 2022-01-18:
Not sure, that could be a litany of configuration or other issues that I cannot debug from afar. I'd recommend reviewing BT.CPP's documentation on writing XML files to make sure you've done it properly and you can always refer to the navigation.ros.org configuration guide for Nav2 specific content.