ros-navigation/navigation2 #2777: Remove nav2-rviz-plugins from navigation2 dependencies?

RiZom-91, 2022-01-19:
Hi,

I've noticed navigation2 package has nav2-rviz-plugins as a dependency which can be unnecessary and/or unbuildable for embedded targets (Qt5 dependency mostly) like mobile robots.

Of course, it is possible to build/install each sub-package individually but I've created this issue to ask if this should be an optional dependency somehow.

Best Regards,
Romain G.

SteveMacenski, 2022-01-19:
No, since most users want the full metapackage including rviz plugins, we include it in the nav2 dependency list so its a 1 stop shop for this project. But you’re always welcome to pick and choose which you would like! That’s alot of the value of independent packages! 🙂

RiZom-91, 2022-01-19:
Thanks for the answer. I'll close the ticket.

closed 2022-01-19 by RiZom-91