ros-navigation/navigation2 #2785: Failed to change the odom_topic of bt_navigator

bot-lin, 2022-01-26:
## Bug report

**Required Info:**

- Operating System:
  - 
  - Ubuntu 20.04
- ROS2 Version:
  - 
  - Galactic binaries

#### Steps to reproduce issue

```
In the nav2_params.yaml file, I changed the odom_topic of bt_navigator from /odom to /odom1, but the bt_navigator node was still subscribed to /odom when launch 
```

#### Expected behavior
bt_navigator node subscribes to /odom1
#### Actual behavior
bt_navigator node subscribes to /odom
#### Additional information

SteveMacenski, 2022-01-26:
Please check that you changed odom topic for any BT node’s port that may use it in your behavior tree. You may be using BT nodes which have their own independent parameters that are using it. 

All parameters and ports are specified in the configuration guide at navigation.ros.org

SteveMacenski, 2022-01-26:
Actually, I think this may be the result of a bug. 

https://github.com/ros-planning/navigation2/blob/main/nav2_bt_navigator/src/bt_navigator.cpp#L75

We declare it here but do not get the parameter and pass it onto the odometry smoother later down in the file. The smoother has a default odom topic constructor argument we aren’t overriding. 

A PR would be appreciated to resolve. You’d only need to get the parameter & pass it as the 3rd argument to the odom smoother constructor

SteveMacenski, 2022-02-04:
Merging when CI finishes

closed 2022-02-04 by SteveMacenski