ros-navigation/navigation2 #2796: Couldn't load critics when launching in namespace

Fellfalla, 2022-02-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- ROS2 Version: Galactic and installed via `apt`
- Version or commit hash: 1.0.8-1focal.20211223.041721
- DDS implementation: default

#### Steps to reproduce issue

Run `ros2 launch bringup navigation.launch.py namespace:=ns` with following launch file and the default `nav2_params.yaml`

```
# Source: https://github.com/ros-planning/navigation2 @ d1ed5e466286898abebbc1bf80dd51e009670fe5

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from nav2_common.launch import RewrittenYaml

def generate_launch_description():
    # Get the launch directory
    bringup_dir = get_package_share_directory('bringup')

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')
    autostart = LaunchConfiguration('autostart')
    params_file = LaunchConfiguration('params_file')
    default_bt_xml_filename = LaunchConfiguration('default_bt_xml_filename')
    map_subscribe_transient_local = LaunchConfiguration('map_subscribe_transient_local')

    lifecycle_nodes = ['controller_server',
                       'planner_server',
                       'recoveries_server',
                       'bt_navigator',
                       'waypoint_follower']

    # Map fully qualified names to relative ones so the node's namespace can be prepended.
    # In case of the transforms (tf), currently, there doesn't seem to be a better alternative
    # https://github.com/ros/geometry2/issues/32
    # https://github.com/ros/robot_state_publisher/pull/30
    # TODO(orduno) Substitute with `PushNodeRemapping`
    #              https://github.com/ros2/launch_ros/issues/56
    # remappings = [('/tf', 'tf'),
    #               ('/tf_static', 'tf_static')]
    remappings = []

    # Create our own temporary YAML files that include substitutions
    param_substitutions = {
        'robot_base_frame': [namespace, '_base_link'],
        'odom_frame_id': [namespace, '_odom'],
        'local_costmap.local_costmap.ros__parameters.global_frame': [namespace, '_odom'],
        'recoveries_server.ros__parameters.global_frame': [namespace, '_odom'],
        'use_sim_time': use_sim_time,
        'default_bt_xml_filename': default_bt_xml_filename,
        'autostart': autostart,
        'map_subscribe_transient_local': map_subscribe_transient_local}

    configured_params = RewrittenYaml(
            source_file=params_file,
            root_key=namespace,
            param_rewrites=param_substitutions,
            convert_types=True)

    return LaunchDescription([
        # Set env var to print messages to stdout immediately
        SetEnvironmentVariable('RCUTILS_LOGGING_BUFFERED_STREAM', '1'),

        DeclareLaunchArgument(
            'namespace', 
            default_value='',
            description='Top-level namespace'),

        DeclareLaunchArgument(
            'use_sim_time', 
            default_value='True',
            description='Use simulation (Gazebo) clock if true'),

        DeclareLaunchArgument(
            'autostart', 
            default_value='True',
            description='Automatically startup the nav2 stack'),

        DeclareLaunchArgument(
            'params_file',
            default_value=os.path.join(bringup_dir, 'param', 'nav.param.yaml'),
            description='Full path to the ROS2 parameters file to use'),

        DeclareLaunchArgument(
            'default_bt_xml_filename',
            default_value=os.path.join(
                get_package_share_directory('nav2_bt_navigator'),
                'behavior_trees', 'navigate_w_replanning_and_recovery.xml'),
            description='Full path to the behavior tree xml file to use'),

        DeclareLaunchArgument(
            'map_subscribe_transient_local', 
            default_value='false',
            description='Whether to set the map subscriber QoS to transient local'),

        Node(
            package='nav2_controller',
            executable='controller_server',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),

        Node(
            package='nav2_planner',
            executable='planner_server',
            name='planner_server',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),

        Node(
            package='nav2_recoveries',
            executable='recoveries_server',
            name='recoveries_server',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),

        Node(
            package='nav2_bt_navigator',
            executable='bt_navigator',
            name='bt_navigator',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),

        Node(
            package='nav2_waypoint_follower',
            executable='waypoint_follower',
            name='waypoint_follower',
            output='screen',
            parameters=[configured_params],
            remappings=remappings),

        Node(
            package='nav2_lifecycle_manager',
            executable='lifecycle_manager',
            name='lifecycle_manager_navigation',
            output='screen',
            parameters=[{'use_sim_time': use_sim_time},
                        {'autostart': autostart},
                        {'node_names': lifecycle_nodes}]),

    ])

```

#### Expected behavior
Booting controller_server works as if the node was **not** launched from withing a namespace.

#### Actual behavior
Configuring controller_server fails.

#### Additional information
I get following error:
> [controller_server-1] [ERROR] [1643806193.910975259] [controller_server]: Couldn't load critics! Caught exception: No critics defined for FollowPath
> [controller_server-1] [ERROR] [1643806193.928792793] []: Caught exception in callback for transition 10
> [controller_server-1] [ERROR] [1643806193.928813740] []: Original error: No critics defined for FollowPath

padhupradheep, 2022-02-02:
Tried to reproduce the error, seems that the `LaunchConfiguration` is not reading the namespace at all, consequently not reading the parameters from the nav2_params.yaml. 

I checked  `ros2 node list` to see if the namespaces are read. But, unfortunately the namespace is not read at all.  Cannot figure out what is the issue. 

Leaving it to Steve!

SteveMacenski, 2022-02-02:
PRs are always appreciated, I don't have the cycles in the foreseeable future to address this myself

padhupradheep, 2022-02-02:
Sure, I will look into this issue in few days, if this is not too urgent for @Fellfalla. I need to finish the other issues that I’m working on first. I will add it to my list of Todo’s.

SteveMacenski, 2022-02-03:
By the way, we push the namespaces of the nodes in the parent file to the navigation.launch.py using `PushRosNamespace` https://github.com/ros-planning/navigation2/blob/galactic/nav2_bringup/bringup/launch/bringup_launch.py#L80-L82. 

The `bringup_launch.py` file is the entry point I'd recommend starting modifications at for that reason and that it passes through the parameters to the navigation launch file + offers direct control over localization as well. 

If you're not using the default nav2 launch files, then you could go the route @padhupradheep PR does in adding it to the nodes, but if you want to use the nav2's provided navigation launch file, you should instead just modify the bringup file instead and passthrough any parameter changes there (they all propagate). 

@padhupradheep does this namespace work when doing `bringup` instead of `navigation`? I suppose the issue here could just be as simple as "this is the wrong launch file to be declaring namespace for".

It might be worth at some point going through these launch files and simplifying them. Admittedly, its a little non-obvious at first glance especially since all of the launch files have near mirrored set of launch configurations.

padhupradheep, 2022-02-03:
> @padhupradheep does this namespace work when doing bringup instead of navigation?

I can confirm that this works. 

> Is there a way to detect if that's happened and take the appropriate action?

Yes, we can check if the namespace given in the `LaunchConfiguration` has received a value or not. With the help of the `LaunchContext` Api, we can see if the particular argument is empty or not.

SteveMacenski, 2022-02-03:
> I can confirm that this works.

Got it, so this works, we're just in the wrong place for the current behavior (which may not be ideal)

> we can check if the namespace given in the LaunchConfiguration has received a value or not. 

I don't think that necessarily gives us the full picture.
- namespace is `ns` that was defined in bringup so nodes are already pushed into it and navigation launch still gets `namespace`
- namespace is set as `ns` on CLI for navigation without bringup and navigation file gets `namespace`

Both situations get `namespace` value but we would want to have different behavior if one was already "pushed" and the other was not.

padhupradheep, 2022-02-03:
> launch file is just for the Nav2 stack and is expecting to be orchestrated by another bringup file which has other important things like localization attached that will push its children nodes into the namespace. Assuming you have a parent launch file (which you would need to have if you wanted localization), you could push there or if you're maintaining a completely separate launch file from navigation, doing the simple change the linked PR contains.

I think I will agree with this point. As you said `navigation_launch.py` is used for launching from another parent file. The idea that I have put inside my PR linked with this issue will obviously help. Moreover, I believe `navigation_launch` was not built with an intention to be launched standalone.  Hope @Fellfalla was helped with our conversations.

SteveMacenski, 2022-02-04:
Agreed, I'm not adamant that it has to be that way at all, so don't take the status quo as something immovable. I do like @padhupradheep's PR, but I do want to understand exactly what is going on for why its not pushing into the namespaces twice. I want to make sure we're relying on an intentional feature and not a bug that we've stumbled onto. 

Assuming its intentional on the Launch side, I'd 100% support it (and maybe a good excuse to just remove the `Push` action, except for the multirobot case.

Fellfalla, 2022-02-04:
I was able to solve my problem by setting the `default_value` of the `DeclareLaunchArgument` to the `LaunchConfiguration` value according to following modification:

```python

namespace = LaunchConfiguration('namespace')

# ... Other code goes here....

DeclareLaunchArgument(
    'namespace', default_value=namespace,
    description='Top-level namespace'),
```

Fellfalla, 2022-02-04:
I was able to solve my problem by doing 2 things.

1. setting the `default_value` of the `DeclareLaunchArgument` to the `LaunchConfiguration` value according to following modification:

```python

namespace = LaunchConfiguration('namespace')

# ... Other code goes here....

DeclareLaunchArgument(
    'namespace', default_value=namespace,
    description='Top-level namespace'),
```

2. Adding `PushRosNamespace(namespace=namespace)` to the `LaunchDescription` in navigation.launch.py

closed 2022-02-04 by SteveMacenski