ros-navigation/navigation2 #2799: Ability to pass array of planner_ids to ComputePathThroughPoses.action

AlexKaravaev, 2022-02-03:
## Feature request

Ability to pass array of planner_ids to ComputePathThroughPoses.action.

#### Feature description

I think it will be quite handy and not so hard to implement this. I have some following scenario, I have a set-up where robot is navigating with nav2 plannet plugin that plans global plan on a lanelet2 map(I hope I will open source this in the future, but this it's not up to me). I want to be able to switch planner when I am navigating in the free space outside the graph, and use for example smac planner for a free space navigation.

#### Implementation considerations
In a straightforward way just to change action definition and this line and below 
https://github.com/ros-planning/navigation2/blob/459b74d5fa53e1588b01e2c7bbd865acc0dc6cbb/nav2_planner/src/planner_server.cpp#L394

Also maybe it's possible to include this in rviz gui too.

SteveMacenski, 2022-02-03:
Sounds easy enough! Would you be willing to submit a PR?

I think the only missing important detail is that we need to check that the array of planner id’s are either 0 (use default), 1 (use for all), or the same length as the number of waypoints requested. Else, reject

AlexKaravaev, 2022-02-03:
@SteveMacenski Check sounds great! 

I will be working on this in the future and will open PR for that in a little while

AlexKaravaev, 2022-02-09:
I've started working on this and I have implemented corresponding changes to the planner_server.cpp, it now can accept and compute paths for arbitrary different planners. However, I am little bit stuck on the conceptual and design thing, how we should pass these different planner_ids to the planner_server?

One thing is that with array of planner_ids the behaviour tree node ComputePathTroughPoses becoming a little bit obsolete imo.

https://github.com/ros-planning/navigation2/blob/5a17db5a07e1573d81be33f0b737bcac8c43d2d1/nav2_bt_navigator/behavior_trees/navigate_through_poses_w_replanning_and_recovery.xml#L14

Because in behaviour tree we set this planner_ids as a static variable. One way to overcome this is to set planner_ids as a blackboard variable and then create another behaviour tree node, that will set this planner_ids based on some specific rule. Or this way if planner_ids is zero, then default planner for all the path segments will be used. 

I just wanted to check, if logic here is correct and won't break any api or smth else.

SteveMacenski, 2022-02-09:
> Because in behaviour tree we set this planner_ids as a static variable

It doesn't have to be, anything with the `{}` are blackboard variables who's value is used. Hence how the planners and controllers pass the `path` variable value via `{path}`. So you could define a set of `{planner_ids}` to use, but setting those would, as you mention, be up to you as the end user to figure out. 

Though I think the easiest solution would be to have a BT node that changes the `planner_ids` based on a service call or something. That could be generalized and merged into Nav2 so that you could leverage it. Have a service with fields `blackboard_variable_name` and `value` and call to set them. You'd have several `value` fields for each data type, but just populate the 1 that you're using. The dynamic parameter interface would be an analog and that's what that does https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterValue.msg

SteveMacenski, 2022-03-08:
See https://github.com/ros-planning/navigation2/pull/2817#issuecomment-1062381538

closed 2022-03-08 by SteveMacenski