ros-navigation/navigation2 #2801: Feature request: BehaviorTree Loggers

benjinne, 2022-02-04:
## Feature request

#### Feature description
https://www.behaviortree.dev/tutorial_05_subtrees/ describes how a logger object can log state changes. Is this possible with nav2? 
I've searched this nav2 repo and I don't see any instances of `FileLogger` mentioned anywhere.

SteveMacenski, 2022-02-04:
This is already done with the https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/include/nav2_behavior_tree/ros_topic_logger.hpp. We publish state changes over a topic so they can be collected in a rosbag along with other debug information so you can track the flow of information. It doesn't make sense in this project to have a separate file logger given ROS 2 is largely an event based async framework where we need to track multiple time-series operations at once.

But you can do this now if you subscribe to the `behavior_tree_log` topic!

closed 2022-02-04 by SteveMacenski

benjinne, 2022-02-08:
Is there any way to visualize this topic other than text format? I've had to manually format it for it to be easier to look at, but it's a bit tedious.

benjinne, 2022-02-09:
Another problem I found with the `behavior_tree_log` topic is it only tracks changes in the tree node states. It doesn't provide a path of which nodes are ticked.