ros-navigation/navigation2 #2808: [Smack] Crash using footprint instead of radius

matijazigic, 2022-02-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20
- ROS2 Version:
  - Rolling
- Version or commit hash:
  - 5a17db5a07e1573d81be33f0b737bcac8c43d2d1
- DDS implementation:
  - default

#### Steps to reproduce issue

```

1. Use smack planner
2. Use footprint instead of radius
3. Send goal

```

#### Expected behavior

#### Actual behavior

#### Additional information
    #robot_radius: 0.22
    footprint: "[[0.2, 0.1], [0.2, -0.1], [-0.2, -0.1], [-0.2, 0.1]]"

    planner_server:
      ros__parameters:
        planner_plugins: ["GridBased"]
        use_sim_time: True
        GridBased:
          plugin: "nav2_smac_planner/SmacPlannerLattice"
          tolerance: 0.5                      # tolerance for planning if unable to reach exact pose, in meters, for 2D node
          downsample_costmap: false           # whether or not to downsample the map
          downsampling_factor: 1              # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
          allow_unknown: false                # allow traveling in unknown space
          max_iterations: 1000000             # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
          max_on_approach_iterations: 1000    # maximum number of iterations to attempt to reach goal once in tolerance, 2D only
          max_planning_time: 3.5              # max time in s for planner to plan, smooth, and upsample. Will scale maximum smoothing and upsampling times based on remaining time after planning.
          motion_model_for_search: "DUBINS"    # 2D Moore, Von Neumann; Hybrid Dubin, Redds-Shepp; State Lattice set internally
          cost_travel_multiplier: 2.0         # For 2D: Cost multiplier to apply to search to steer away from high cost areas. Larger values will place in the center of aisles more exactly (if non-`FREE` cost potential field exists) but take slightly longer to compute. To optimize for speed, a value of 1.0 is reasonable. A reasonable tradeoff value is 2.0. A value of 0.0 effective disables steering away from obstacles and acts like a naive binary search A*.
          angle_quantization_bins: 64         # For Hybrid nodes: Number of angle bins for search, must be 1 for 2D node (no angle search)
          analytic_expansion_ratio: 3.5 #3.5       # For Hybrid/Lattice nodes: The ratio to attempt analytic expansions during search for final approach.
          analytic_expansion_max_length: 3.0    # For Hybrid/Lattice nodes: The maximum length of the analytic expansion to be considered valid to prevent unsafe shortcutting (in meters). This should be scaled with minimum turning radius and be no less than 4-5x the minimum radius
          minimum_turning_radius: 0.40        # For Hybrid/Lattice nodes: minimum turning radius in m of path / vehicle
          reverse_penalty: 2.1                # For Reeds-Shepp model: penalty to apply if motion is reversing, must be => 1
          change_penalty: 0.0                 # For Hybrid nodes: penalty to apply if motion is changing directions, must be >= 0
          non_straight_penalty: 1.20          # For Hybrid nodes: penalty to apply if motion is non-straight, must be => 1
          cost_penalty: 2.0                   # For Hybrid nodes: penalty to apply to higher cost areas when adding into the obstacle map dynamic programming distance expansion heuristic. This drives the robot more towards the center of passages. A value between 1.3 - 3.5 is reasonable.
          retrospective_penalty: 0.025        # For Hybrid/Lattice nodes: penalty to prefer later maneuvers before earlier along the path. Saves search time since earlier nodes are not expanded until it is necessary. Must be >= 0.0 and <= 1.0
          rotation_penalty: 5.0               # For Lattice node: Penalty to apply only to pure rotate in place commands when using minimum control sets containing rotate in place primitives. This should always be set sufficiently high to weight against this action unless strictly necessary for obstacle avoidance or there may be frequent discontinuities in the plan where it requests the robot to rotate in place to short-cut an otherwise smooth path for marginal path distance savings.
          lattice_filepath: "/home/matija/Desktop/nav2_ws/src/navigation2/nav2_smac_planner/lattice_primitives/sample_primitives/5cm_resolution/0.5m_turning_radius/ackermann/output.json" # The filepath to the state lattice graph
          lookup_table_size: 20.0              # For Hybrid nodes: Size of the dubin/reeds-sheep distance window to cache, in meters.
          cache_obstacle_heuristic: True      # For Hybrid nodes: Cache the obstacle map dynamic programming distance expansion heuristic between subsiquent replannings of the same goal location. Dramatically speeds up replanning performance (40x) if costmap is largely static.  
          allow_reverse_expansion: True #False      # For Lattice nodes: Whether to expand state lattice graph in forward primitives or reverse as well, will double the branching factor at each step.   
          smooth_path: True                   # For Lattice/Hybrid nodes: Whether or not to smooth the path, always true for 2D nodes.
          smoother:
            max_iterations: 1000
            w_smooth: 0.3
            w_data: 0.2
            tolerance: 0.0000000001
            do_refinement: true               # Whether to recursively run the smoother 3 times on the results from prior runs to refine the results further

----
## Feature request

#### Feature description

#### Implementation considerations

matijazigic, 2022-02-08:
Stacktrace:

```
Log line format: [IWEF]mmdd hh:mm:ss.uuuuuu threadid file:line] msg
E0208 17:27:37.680630 383299 main.cpp:29] *** Aborted at 1644337657 (unix time) try "date -d @1644337657" if you are using GNU date ***
E0208 17:27:37.683888 383299 main.cpp:29] PC: @                0x0 (unknown)
E0208 17:27:37.684031 383299 main.cpp:29] *** SIGSEGV (@0x8) received by PID 383029 (TID 0x7f3390b0b700) from PID 8; stack trace: ***
E0208 17:27:37.685885 383299 main.cpp:29]     @     0x7f33c82693c0 (unknown)
E0208 17:27:37.689031 383299 main.cpp:29]     @     0x7f33c7e03975 nav2_costmap_2d::FootprintCollisionChecker<>::footprintCost()
E0208 17:27:37.689402 383299 main.cpp:29]     @     0x7f33c00f0e5f nav2_smac_planner::GridCollisionChecker::inCollision()
E0208 17:27:37.689700 383299 main.cpp:29]     @     0x7f33c00fae9a nav2_smac_planner::NodeLattice::isNodeValid()
E0208 17:27:37.689934 383299 main.cpp:29]     @     0x7f33c00fc0fc nav2_smac_planner::NodeLattice::getNeighbors()
E0208 17:27:37.690172 383299 main.cpp:29]     @     0x7f33c00e63f6 nav2_smac_planner::AStarAlgorithm<>::createPath()
E0208 17:27:37.690488 383299 main.cpp:29]     @     0x7f33c00ceee2 nav2_smac_planner::SmacPlannerLattice::createPlan()
E0208 17:27:37.692167 383299 main.cpp:29]     @     0x7f33c862169f nav2_planner::PlannerServer::getPlan()
E0208 17:27:37.693833 383299 main.cpp:29]     @     0x7f33c862710e nav2_planner::PlannerServer::computePlan()
E0208 17:27:37.695358 383299 main.cpp:29]     @     0x7f33c864bdeb nav2_util::SimpleActionServer<>::work()
```

SteveMacenski, 2022-02-08:
Can you compile with -g so we can see exact lines / memory locations? That would narrow it down. I haven’t seen this before but looks like it should be a quick fix if I know exactly what variable to analyze

matijazigic, 2022-02-08:
I tried compiling with the -g flag, and everything that is suggested [here (https://navigation.ros.org/tutorials/docs/get_backtrace.html) with no luck. I added -g flag inside nav2_planner
 and nav2_smac_planner. Am I doing something wrong?

Only thing that I can get is: 
> [component_container_isolated-6] [WARN] [1644321675.406004450] [planner_server]: GridBased plugin failed to plan calculation to (0.43, 1.77): "vector::reserve"

So I removed [try/catch block ](https://github.com/ros-planning/navigation2/blob/main/nav2_planner/src/planner_server.cpp#L354) from planner_server and implemented glog crash dump and got [this](https://github.com/ros-planning/navigation2/issues/2808#issuecomment-1032818655). Here you can find an exact line where SIGSEGV happened.

SteveMacenski, 2022-02-08:
Wait, is this a crash or just an exception is thrown and caught?

SteveMacenski, 2022-02-08:
A couple of follow up questions then.

- If you use Hybrid instead of Lattice, do you see this?
- If you use another footprint definition, does it work?
- If you use Lattice with radius, does it work?

matijazigic, 2022-02-08:
It is an exception caught. It would not plan because of that.

- I'll try
- No, tried different footprints
- Lattice with radius working normally

matijazigic, 2022-02-08:
Hybrid also working. I used parameters from [example](https://navigation.ros.org/configuration/packages/smac/configuring-smac-hybrid.html?highlight=smac#example).

SteveMacenski, 2022-02-08:
Hybrid with the same footprint as you used in Lattice?

SteveMacenski, 2022-02-08:
From that, I think this is the issue

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_lattice.cpp#L214

Does changing 

```
const double & angle = motion_table.getAngleFromBin(this->pose.theta) / bin_size;
```

to 

```
const double & angle = floor(motion_table.getAngleFromBin(this->pose.theta) / bin_size);
```

fix it? If so, please submit a PR with that and I'll merge. I think what's happening is that it rounds up when its in the N-1'th bin size to N which doesn't exist (that would be 0 turning around the circle)

matijazigic, 2022-02-08:
Hello @SteveMacenski, 
Maybe the problem lies here: 
https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/collision_checker.cpp#L117
oriented_footprints_ is a vector, and angle_bin is float?

SteveMacenski, 2022-02-08:
Hybrid-A* does the same thing, but computes the float angle differently, that's why I'm suspect of the line I linked to avoid that would floor the angle used in the collision checker rather than allowing it to round up when it is cast into an `int` for that line you linked to.

The error `vector::reserve` and that your trace shows `footprintCost` makes me suspicious that that's not actually the issue, its actually in your footprint itself, but its a good first thing to try.

matijazigic, 2022-02-08:
I tried changing node_lattice.cpp, as you suggested, but the same problem occurs.

SteveMacenski, 2022-02-08:
OK, I'll actually need a traceback with line numbers and memory locations then. Or you can add some prints and see if anything looks suspicious or determine what line it crashes at. That may be faster given we know roughly where its happening.

matijazigic, 2022-02-09:
It fails when the bin has a negative value, I put one std::cout inside GridCollisionChecker::inCollision to print the angle_bin value. I'll invest more time later, to debug it further and try to fix it.

> [planner_server-11] angle_bin: -39.381
> [planner_server-11] [ERROR] [1644427350.836848361] [planner_server]: Action server failed while executing action callback: "vector::reserve"

SteveMacenski, 2022-02-09:
That is odd. The angle should already be normalized from the control set generated. I don't see any reason that I can identify why that could be negative. 

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_lattice.cpp#L504 

angle is set to the motion projection which is fresh off of the control set generator, so unless you see a negative value in your json file for the control set primitives, its not that. 

If we reflect it for a backwards maneuver, we also normalize it again to be safe

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_lattice.cpp#L507-L513

Then those values are set for use and calls the `isNodeValid` function where this all goes down. Plus, if it were negative, I don't think motion_table.getAngleFromBin(this->pose.theta) would return a valid solution. 

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/src/node_lattice.cpp#L516-L525

matijazigic, 2022-02-09:
> angle is set to the motion projection which is fresh off of the control set generator, so unless you see a negative value in your json file for the control set primitives, its not that.

But I'm using default json, and the backward flag is false, I need to dive deeper in code, I'm still not familiar with it.

https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/lattice_primitives/sample_primitives/5cm_resolution/0.5m%20_turning_radius/ackermann/output.json

SteveMacenski, 2022-02-09:
If you do not specify a file, this is the default https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/test/default_model.json. Your configuration file shows that you're overriding the default. 

Either way, I scanned them both and they both look good to me. Might be worth adding some prints along that trail I laid and see where the first negative value appears. They should always be between 0-15 for a 16 discretization control set.

mattbooker, 2022-02-25:
> Also, I figure out that there is a problem with generated motion primitives: https://github.com/ros-planning/navigation2/blob/main/nav2_smac_planner/lattice_primitives/sample_primitives/5cm_resolution/0.5m%20_turning_radius/ackermann/output.json#L2096
> 
> Adjacent poses have a large diff in the theta. From 4.185 to -2.034, is that something that is supported by the code?

Took a look at this and fixed the bug causing it. It doesn't affect any of the downstream planning stuff as the values were representing the correct direction, they just were off by 2 * pi. 

In any case, it may affect other peoples code so I have fixed it by normalizing all the angles in every trajectory. I've also added some test cases for the output to make sure there aren't other small issues like this.

SteveMacenski, 2022-02-25:
I think it could cause an issue within the planner since we assume all of the angles are normalized from 0-2PI so that could be causing this issue as well (albeit, its also just as likely that we have a floating point round up issue).

SteveMacenski, 2022-03-10:
@matijazigic how are things going?

matijazigic, 2022-03-15:
@SteveMacenski I'll try it. Hopefully today, I was really busy in the last couple of days. Thanks.

SteveMacenski, 2022-03-21:
The PR should fix it, if there are continued issues let me know and we can reopen

closed 2022-03-21 by SteveMacenski