ros-navigation/navigation2 #2809: rate_controller resets start time too often

benjinne, 2022-02-08:
https://github.com/ros-planning/navigation2/blob/main/nav2_behavior_tree/plugins/decorator/rate_controller.cpp

Is a useful tool for slowing down things like printing to the console, however it resets it's start time whenever its state reaches `idle`. This causes issues with a simple tree like this:

[image]

The rate controller doesn't slow down since after each pass of the recovery node it calls `halt()` on all the children nodes and sets the states to idle.

Does it make sense to for the rate controller to check if it's idle to reset?
https://github.com/ros-planning/navigation2/blob/5a17db5a07e1573d81be33f0b737bcac8c43d2d1/nav2_behavior_tree/plugins/decorator/rate_controller.cpp#L36

SteveMacenski, 2022-02-08:
No, that is expected behavior. A BT node should fully reset its state when not ticked over and is out of scope. You can just move the decorator to a higher level since you really want to be throttling your check, not the publication, if the only thing that’s being done is publishing (eg why do the work if you’re not going to publish yet). 

Please read the docs on BehaviorTree.CPP and read other guides for BT design

benjinne, 2022-02-08:
What if I don't want to move the rate_controller decorator to a higher level since I want my condition checked at the full rate? It doesn't appear theres a good solution for this and is very un-intuitive unless you dive into how the rate_controller works.

SteveMacenski, 2022-02-08:
You are always able to create custom BT nodes if you like to create the desirable behavior, but this is intended behavior following best practices of Behavior Tree design. 

Michelle’s book does a good job explaining the basics https://arxiv.org/abs/1709.00084 and you have a few BTs in Nav2 you can use as additional reference for use. But in general unless marked as global, all BT nodes states will reset when its parent is to avoid cross pollution

benjinne, 2022-02-08:
I tried moving `rate_control` up the tree, but now It's not being used since the recovery node keeps the loop inside. This causes the 300 retries to execute immediately.

[image]

Here's a bit of the `behavior_tree_log` topic
```
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
node_name: PublishMessage
  previous_status: RUNNING
  current_status: SUCCESS
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
node_name: PublishMessage
  previous_status: RUNNING
  current_status: SUCCESS
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
node_name: PublishMessage
  previous_status: RUNNING
  current_status: SUCCESS
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
node_name: PublishMessage
  previous_status: RUNNING
  current_status: SUCCESS
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
node_name: PublishMessage
  previous_status: RUNNING
  current_status: SUCCESS
node_name: PublishMessage
  previous_status: SUCCESS
  current_status: IDLE
node_name: IsArmed
  previous_status: IDLE
  current_status: FAILURE
node_name: IsArmed
  previous_status: FAILURE
  current_status: IDLE
node_name: PublishMessage
  previous_status: IDLE
  current_status: RUNNING
  ```

benjinne, 2022-02-09:
I found a solution, although it's kind of a hack. I found that to keep the rate_controller's state, the subtree must always return `running` that way its start time doesn't reset and bypass the purpose of the rate_controller. After looking at other trees using the rate_controller it's clear that's how it's always used since it's typically used to limit the rate of planning while a controller is reporting running since it takes some time for controllers to finish. 

Here's my working tree

[image]

It's not ideal since `retry_count` never increments 
[image]

benjinne, 2022-02-09:
Actually, the recovery_node won't work since it'll only tick the running side and never check `is_armed`. I had to change it to `reactive_fallback` and now it works as intended

[image]