ros-navigation/navigation2 #2810: Not able to execute navigation2 launch file

Gianstiff, 2022-02-08:
I'm following [this tutorial](https://navigation.ros.org/tutorials/docs/navigation2_with_slam.html) and I can confirm that with a similar setup (on a turtlebot3) such as [this](https://github.com/ros-planning/navigation2/issues/2651) closed issue and I'm having the same problem when executing:

`ros2 launch nav2_bringup navigation_launch.py`

[image]

SteveMacenski, 2022-02-08:
Ask on ROS Answers

closed 2022-02-08 by SteveMacenski

Gianstiff, 2022-02-09:
> Ask on ROS Answers

Ok

> but looks like gazebo isn't up / happy with the tree you're using. That means the nothing is publishing the odom->base_link which comes from gazebo / RL

@SteveMacenski thanks for the feedback, but actually I'm working with a physical robot, not the simulated one