ros-navigation/navigation2 #2813: [BUG] Unable to resolve rosdep rolling keys in CI build

Aposhian, 2022-02-11:
## Bug report

**Required Info:**

- Operating System:
  - Nav2 build environment
- ROS2 Version:
  - rolling
- Version or commit hash:
  - main (`5a17db5a07e1573d81be33f0b737bcac8c43d2d1` right now)
- DDS implementation:
  - N/A

#### Steps to reproduce issue
Open a PR to `main`

#### Expected behavior
Build succeeds, and tests pass.

#### Actual behavior
Build fails on trying to install dependencies:
```ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_system_tests: No definition of [launch_testing] for OS version [focal]
nav2_costmap_2d: No definition of [launch_testing] for OS version [focal]
nav2_smac_planner: No definition of [ament_cmake_pytest] for OS version [focal]
nav2_util: No definition of [ament_cmake_pytest] for OS version [focal]
costmap_queue: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_waypoint_follower: No definition of [ament_cmake_pytest] for OS version [focal]
nav2_simple_commander: No definition of [ament_pep257] for OS version [focal]
nav2_rotation_shim_controller: No definition of [ament_lint_auto] for OS version [focal]
nav2_recoveries: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_navfn_planner: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_behavior_tree: No definition of [ament_cmake_gtest] for OS version [focal]
nav_2d_msgs: No definition of [rosidl_default_generators] for OS version [focal]
nav2_map_server: No definition of [launch_testing] for OS version [focal]
nav2_smoother: No definition of [ament_cmake_pytest] for OS version [focal]
nav2_controller: No definition of [ament_cmake_pytest] for OS version [focal]
nav_2d_utils: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_amcl: No definition of [ament_lint_auto] for OS version [focal]
dwb_msgs: No definition of [rosidl_default_runtime] for OS version [focal]
nav2_voxel_grid: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_bringup: No definition of [launch_testing] for OS version [focal]
nav2_dwb_controller: No definition of [ament_cmake] for OS version [focal]
nav2_lifecycle_manager: No definition of [ament_cmake_pytest] for OS version [focal]
nav2_core: No definition of [launch_testing] for OS version [focal]
nav2_regulated_pure_pursuit_controller: No definition of [ament_lint_auto] for OS version [focal]
navigation2: No definition of [ament_cmake] for OS version [focal]
dwb_core: No definition of [ament_cmake_gtest] for OS version [focal]
dwb_plugins: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_theta_star_planner: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_common: No definition of [ament_cmake_core] for OS version [focal]
nav2_rviz_plugins: No definition of [ament_lint_auto] for OS version [focal]
dwb_critics: No definition of [ament_cmake_gtest] for OS version [focal]
nav2_msgs: No definition of [nav_msgs] for OS version [focal]
nav2_planner: No definition of [ament_lint_auto] for OS version [focal]
nav2_bt_navigator: No definition of [ament_lint_auto] for OS version [focal]
```

#### Additional information
In my local install of rolling, `rosdep` seems to be able to resolve the keys. Maybe it is a caching issue in CI? I'm not sure if people are already aware of this issue, but I wanted to create a ticket to track it.

vinnnyr, 2022-02-11:
Is this related to rolling being Ubuntu Jammy soon ?  https://discourse.ros.org/t/holding-releases-to-perform-rolling-transition-to-new-platforms/24191

SteveMacenski, 2022-02-11:
I believe @vinnnyr is correct. Rolling is in the process of moving to 22.04 for Humble

SteveMacenski, 2022-05-02:
Closing this ticket since we moved our CI back to 20.04 but we're still having trouble updating for 22.04 / Humble at the moment covered in #2852 and #2851 which has threads on that transition.

closed 2022-05-02 by SteveMacenski