ros-navigation/navigation2 #2826: Feature request: Waypoint Publisher

girvenavery2022, 2022-02-18:
#### Feature description
The Waypoint Publisher feature would allow developers to give nav2 a set of waypoints taken in from a file and then allow Waypoint Follower to follow them in order. This will solve the issue of only being able to send a set of waypoints through the Rviz2 plugin

#### Implementation considerations
Right now, I have a working proof of theory package that needs some refactoring/clean up but has been tested in simulation on 2 separate robots in foxy/galactic as well as tested on a physcial robot in foxy. The way I implemented it was, the node takes in a CSV file of waypoints described in x, y, z, roll, pitch, and yaw. After it formats the points and pushes them into a vector, it then waits until an initial pose is set in the map frame before it sends the waypoints to the WayPoint follower.

padhupradheep, 2022-02-22:
This is the same exact feature, Me and Steve were speaking during the last weekend. Your idea seems to be good. But the way I had implemented was to directly store the information from the pre-existing navigation plugin in RViz (have some buttons on it "Load" and "save"). Then you can easily load and store the waypoints, wherever you want to. 

On top of this feature, I will also like to see a loop functionality with an count option, where the robot repeats the specified waypoints for the given count. 

All these features on RViz, would enhance the user friendliness and makes it much more easier for non-domain specific people who wants to use Nav2. Also, this really paves a way for semantic navigation as Steve said.

I'm going +1 for this feature.

girvenavery2022, 2022-02-23:
I actually really like the idea of the loop functionality with a counter (why didn't I think of that before?) and I am all for enhancing user friendless, so hooking into the RViz plugin would also be a great feature.

Since you implemented by directly storing the waypoints through the RViz plugin, and then allowing the user to load and store wherever. Could we combine both our ideas and allow the user to choose which one they would like? either through the navigation parameter file or through like the RViz plugin?

Anyways, Please let me know and I will get right onto this feature!

padhupradheep, 2022-02-24:
> Since you implemented by directly storing the waypoints through the RViz plugin, and then allowing the user to load and store wherever. Could we combine both our ideas and allow the user to choose which one they would like? 

Sure, we can do that. 

> either through the navigation parameter file or through like the RViz plugin?

The big picture I see is that, as you said we set a default directory to the waypoints specific file in the parameter file. If and if at all the user wants to add, load or store through the RViz plugin, he/she can also do that. But, if the users decides to do it with the RViz plugin, then the waypoint file should be overriden. 

So the tasks involved to make the feature live are as follows:

- [ ] Add load and save buttons in the RViz Plugin to store and retrieve the Wp's
- [ ] Publish the waypoints, by setting the path to the waypoint file in the parameter file
- [ ] **Bonus:** Add a counter and a loop fields in RViz. 

We just need to find a compromise between the format of the file in which we want to store the waypoints. Currently, I'm storing it in a JSON format. An example of it would be as follows:

```{
    "pose": [
        {
            "id": "0",
            "orient_w": 0.9968881726256568,
            "orient_x": 0,
            "orient_y": 0,
            "orient_z": 0.07882874652738593,
            "pos_x": -1.3630658388137817,
            "pos_y": 0.30783212184906006
        },
        {
            "id": "1",
            "orient_w": 0.9608212861146934,
            "orient_x": 0,
            "orient_y": 0,
            "orient_z": -0.2771686420735696,
            "pos_x": 1.1901600360870361,
            "pos_y": 2.7318668365478516
        },
        {
            "id": "2",
            "orient_w": 0.9983982703770435,
            "orient_x": 0,
            "orient_y": 0,
            "orient_z": 0.056576441281931185,
            "pos_x": -1.05819833278656,
            "pos_y": 0.495088666677475
        }
    ]
}```

I'll just do myself some research if it is possible to easily, store and retrieve through the CSV files using the QT library.

SteveMacenski, 2022-03-01:
> The Waypoint Publisher feature would allow developers to give nav2 a set of waypoints taken in from a file and then allow Waypoint Follower to follow them in order. This will solve the issue of only being able to send a set of waypoints through the Rviz2 plugin

This essentially sounds like the role of an autonomy system, which Nav2 does not right now have available. This is the bring-your-own part of the robot system to implement your use-case specific logic (whether it be waypoint following, waypoint following in a loop, intelligent scheduling, etc). 

I would, though, suppose some kind of caching in the rviz plugin so that you can define a set of waypoints that can be reused over and over again if you wish for demonstration and debugging purposes. But the intention of this would still not be for an autonomy system, that's up to the user to provide and I don't want to go down the rabbit hole of trying to make an autonomy system generalized framework (just yet). 

> Right now, I have a working proof of theory package that needs some refactoring/clean up but has been tested in simulation on 2 separate robots in foxy/galactic as well as tested on a physcial robot [...]

That sounds reasonable for a simple proof of concept demo. 

> either through the navigation parameter file or through like the RViz plugin?

This is where we need to define the bounds of this feature. If its something through a parameter file, then presumably something is reading that in (e.g. a new server or capability within another server) which would be new and a basic autonomy system. I don't want to get into that business in Nav2 for the immediate future since that is a **big** can of worms to open and we need to be able to support more realistic autonomy needs than just following.

If you want some non-demo solution, what you propose above is very reasonable to have some generalized file format (csv) that contains waypoints with a server that reads it in and then publishes the appropriate waypoint following commands. That would be a nice little open source project on your GitHub even! However, I don't think that fits into the 'generalized navigation framework' scope that this project contains. We solve the problems from request -> motor controls, but the autonomy system is a bit out of scope for the moment (until we finish working on some of the more core problems in navigation that we can move up the stack).

I would support having the rviz plugin be able to read some file potentially to execute, however, since the rviz plugin in entirety is for demonstration capabilities. And that would be a demonstration-level autonomy system that I could see having value to integrate for the purposes of testing, benchmarking, and demos. 

As for file format, I'm not overly picky, but I would point out that everything else in Nav2 / ROS 2 uses YAML for the most part. So I would generally prefer to go that route for purely consistency purposes.

girvenavery2022, 2022-03-02:
I guess there was a small gap in my understanding between nav2 and an autonomy system but thank you for clearing that up!

>This is where we need to define the bounds of this feature. If its something through a parameter file, then presumably something is reading that in (e.g. a new server or capability within another server) which would be new and a basic autonomy system. 
>
Okay,  I'll take the addition of allowing the user to choose which one they would like off the table bc its not in the current scope.

So with my current understanding of everything, the tasks @padhupradheep listed to make this feature alive are good? i.e we add save/load widgets to the Rviz plugin, publishing the waypoints by setting the path to the waypoint file, and adding a loop counter to the Rviz plugin.

Coming back to the file format, I'll have to do some research over spring break this week to figure out which file format is going to be the best option.

> That would be a nice little open source project on your GitHub even!  
>
I thought of that same thing while writing my package and even pinned it to my GitHub XD

SteveMacenski, 2022-03-07:
> I guess there was a small gap in my understanding between nav2 and an autonomy system but thank you for clearing that up!

Nav2 does the autonomous navigation from A to B (or A through B, C, D, etc), but the autonomy system or scheduler is what handles the actual definition of what / where A, B, C, D, etc exist at. That can be waypoint following, or scheduling, or multirobot-collaboration. The uses and needs are very broad and if all you want to do is follow some points and loop, using the Nav2 action API you can do that in less than 200 lines of code that wouldn't need to be added to Nav2 specifically, you're just using Nav2 like any other user would for their simple application. 

I don't mind at some point adding a high level autonomy scheduler / task planner application, but that's a much deeper conversation than this ticket implies and a topic for another day and covered by https://github.com/ros-planning/navigation2/issues/2758. 

Yes, what @padhupradheep mentions is good: some widget to save / load waypoints and then execute them using the Nav2 waypoint follower with a looping option. This could be built into the Rviz panel we already have in Nav2, or as a separate one. Separate would probably be easier and would probably be most sensible since this is kind of a stand-alone utility. We can add it to the `nav2_rviz_plugins` and have a plugin you contributed! See the SLAM Toolbox and Nav2 panel plugins for some general advice. The Nav2 is much better structured, but the SLAM Toolbox plugin has path loading fields that you can largely just copy/paste and use.

padhupradheep, 2022-03-08:
> Yes, what @padhupradheep mentions is good: some widget to save / load waypoints and then execute them using the Nav2 waypoint follower with a looping option.

Just continuing this point and for the future readers, as an additional feature, the user can also make this to work much more intelligent by using any of the [waypoint task executors](https://navigation.ros.org/plugins/index.html#waypoint-task-executors).

Maybe a demo documentation would be a nice addition as part of this feature.

girvenavery2022, 2022-03-10:
> I don't mind at some point adding a high level autonomy scheduler / task planner application, but that's a much deeper conversation than this ticket implies and a topic for another day and covered by https://github.com/ros-planning/navigation2/issues/2758.
>
When school slows down here in the next few months, I would love to help contribute to that!

>This could be built into the Rviz panel we already have in Nav2, or as a separate one. Separate would probably be easier and would probably be most sensible since this is kind of a stand-alone utility. We can add it to the nav2_rviz_plugins
>
ok awesome, I already started adding the save/load widgets into their own plugin inside of the nav2_rviz2_plugin 

> Maybe a demo documentation would be a nice addition as part of this feature.
>
This is something I would like to add to it as well. Unfortunately, my Electrical/Mechanical teams have both my robots torn apart right now to do some very needed upgrades so for the time being I can add demo documentation with ignition gazebo but when they are finished I can add demos on the actual robot.

padhupradheep, 2022-04-29:
Hey @girvenavery2022 I could give a hand here, if you need any help here.

girvenavery2022, 2022-04-29:
Hey @padhupradheep, 
sorry for the silence there for a bit. I got caught up in finishing up this semester of school. I did create the rviz plugin UI and will be working on the backend this weekend. I will let you know if I need any help!

SteveMacenski, 2022-05-02:
@girvenavery2022 @padhupradheep any update?

girvenavery2022, 2022-05-03:
hey @SteveMacenski @padhupradheep 
I do have an update. I have it to where it should be accumulating goals by using the goal tool plugin. I have been pushing my work to my fork: [https://github.com/girvenavery2022/navigation2/tree/waypoint_tool](url) 
Tomorrow I am hopefully going to get it to publish those poses. @padhupradheep if you want, would you be able to help me with the loading and saving functionality of this feature?

padhupradheep, 2022-05-03:
>  @padhupradheep if you want, would you be able to help me with the loading and saving functionality of this feature?

Sure, I can give a hand here. I had already developed this feature with JSON file format for another project. Now, it is just that, we have to use the YAML file for loading and saving the waypoints..

girvenavery2022, 2022-05-04:
Okay awesome.

SteveMacenski, 2022-05-04:
yamlcpp would be my recommendation

padhupradheep, 2022-05-04:
@girvenavery2022 will look into this feature by latest on Friday..

girvenavery2022, 2022-05-09:
@padhupradheep any news?

padhupradheep, 2022-05-09:
Sorry for the delay, I guess, it will take another day or two before I can start working on this feature..

padhupradheep, 2022-05-09:
I gave an initial try today, for some reason I get seg-faults while following the yaml-cpp tutorial on parsing yaml-files..! Let me do some further research and get back..

girvenavery2022, 2022-05-11:
> This is going to take some time.. @girvenavery2022 you need to hold on for a bit..
That is totally okay!

in the mean time I'll continue on prettying up the user interface and continue on fixing a bug I found

padhupradheep, 2022-05-13:
seg faults were caused because of the linker issue, I had 2 versions of yaml-cpp installed and that messed up the cmake linkers.. 

Nav2 helps me learn new things everytime :-)

SteveMacenski, 2022-05-13:
That seems reasonable to me

girvenavery2022, 2022-05-15:
> Will you handle the waypoints once the loading has been done? Similarly for the saving as well?
yes of course!
the main branch is targeted for humble correct @SteveMacenski, @padhupradheep?
if so, I'll have to upgrade my computer to ubuntu 22.04 first
but the format looks good!

padhupradheep, 2022-05-16:
>  if so, I'll have to upgrade my computer to ubuntu 22.04 first

not necessarily, the current version of the main branch builds without any problem in ubuntu 20.04

padhupradheep, 2022-05-27:
@girvenavery2022 did you have time to test it ? 

I can open a PR of all the changes that I made on to your fork.. 

For some reason, I'm not able to open a PR on to your fork..

jackykwok2002, 2022-07-29:
Hi, I successfully loaded the plugin into Rviz. However, when I clicked publish point or 2d goal pose, the error "No accumulated Points to save" is shown. What tool should I use to save the save waypoints? Thank you so much for your help in advance.

padhupradheep, 2022-08-17:
So, I just did work on this a bit in the recent days, I made a short video the working. I did not create a separate plugin, mainly because of the fact that, the nav2_plugin already had a nice flexibility to load and save waypoints. Plus sharing memory between the RViz (QT) plugins for the accumulation of waypoints, did not seem to be a direct approach as expected. 

I also planned to add a `pause waypoints` button for pausing the navigation during the waypoint following.

https://user-images.githubusercontent.com/20242192/185057004-08a16f00-61a7-4a31-a1b2-9411d39074ad.mp4

SteveMacenski, 2022-08-18:
Oh that's awesome! How does the saving waypoints bit work? After you mark them in rviz2 but before starting do you hit that and saves to file? For the filesystem prompt, can you have it default to `yaml` instead of `json` so it shows up?

I'd be very happy to review that PR and merge it with only a couple of small tweaks (mostly that `number of loops` should probably be `0` for no looping and `1` for a 'loop' over the first trial, defaulting to `0`)!

padhupradheep, 2022-08-19:
>After you mark them in rviz2 but before starting do you hit that and saves to file?

Exactly, after you start, you cannot use the buttons anymore. The only button that will be available is the cancel button. 

>I'd be very happy to review that PR and merge it with only a couple of small tweaks (mostly that number of loops should probably be 0 for no looping and 1 for a 'loop' over the first trial, defaulting to 0)

Sure, I will do that change and give in a PR by EoD or starting next week. 

Also what do you think on having a pause button?

SteveMacenski, 2022-08-19:
What would the "pause" actually do, technically behind the scenes?  

Also, does this work for both waypoint following and for navigate through poses?

padhupradheep, 2022-08-19:
> What would the "pause" actually do, technically behind the scenes?

Cancel the current goal that the robot is executing, in parallel the pause button transitions to a resume button. The accumulated poses are not cleared when the pause button is pressed. When the resume button is pressed, we send the goals once again. 

> Also, does this work for both waypoint following and for navigate through poses?

Technically, I developed the feature only for the waypoint following. It does not have any limitations for it to be extended to the navigate through poses, but it does not make sense as the intention of the navigate through poses is to reach just the goal location and it does not give as much as importance to the waypoints like how the waypoint following does. Even if you loop with navigate through poses, there is no effect, since the robot is already in the goal location.

SteveMacenski, 2022-08-19:
> Cancel the current goal that the robot is executing, in parallel the pause button transitions to a resume button. The accumulated poses are not cleared when the pause button is pressed. When the resume button is pressed, we send the goals once again.

How would you know which goal to use as the newest one after the pause? It wouldn't still be the first one. Especially for navigate through poses, that information isn't exposed to the client. 

I would like this to work for both WP following / navigate through poses since they are in the same interface. I think the NavThroughPoses makes the pause feature impossible unfortunately :/ Just cancel.

padhupradheep, 2022-08-19:
> I think I expose that in the waypoint follower feedback message though - is that the plan?

Yes, you are right you expose the information about the current waypoint in the feedback. 

> But I suppose a simple greying out of the NavThroughPoses button would suffice here once waypoints are loaded from file so its clear to a user this is only for WP Following.

Makes sense, although the looping feature can be used on the waypoints that is getting selected on RViz. The best way would be to grey out the `nav_through_poses` also when the `number_of_loops > 0`.

padhupradheep, 2022-08-22:
Went a bit further, below you can find the pause and resume functionality. 

https://user-images.githubusercontent.com/20242192/185981299-ae9efb9d-4443-4441-9ffb-525a3fbcd92a.mp4

I will also try to change the `feedback` message to `paused`, now it shows canceled in the status bar of the plugin

SteveMacenski, 2022-08-22:
That looks awesome!

SteveMacenski, 2023-01-04:
Merge imminent!

closed 2023-01-04 by SteveMacenski