ros-navigation/navigation2 #2828: Launch nav2_costmap_2d alone.

TaoYibo1866, 2022-02-21:
In `nav2_bringup/launch/navigation_launch.py` i couldn't locate where nodes related to costmap are launched. I guess they are integrated into `planner_server` and `controller_server`?
I wonder how i can launch` nav2_costmap_2d` alone and pass parameters into it. I fail attempting `ros2 run nav2_costmap_2d nav2_costmap_2d`.

padhupradheep, 2022-02-22:
> i couldn't locate where nodes related to costmap are launched. I guess they are integrated into planner_server and controller_server?

That is true, the costmaps are instantiated in those servers. 

> I wonder how i can launch nav2_costmap_2d alone and pass parameters into it. I fail attempting ros2 run nav2_costmap_2d nav2_costmap_2d.

Did you checkout there is an example.launch in the [nav2_costmap_2d](https://github.com/ros-planning/navigation2/blob/main/nav2_costmap_2d/launch/example.launch) ? Maybe this can help. 

Is there any specific reason you want to do it w/o `controller_server` or the `planner_server` ?

But anyways, this is a bug or feature request tracker. I would recommend you to use [ROS Answers](https://answers.ros.org/questions/) for this question.

TaoYibo1866, 2022-02-22:
Sorry, i shouldn't ask here.

> Is there any specific reason you want to do it w/o controller_server or the planner_server ?
I was trying to write a demo for my own planner, which visualize path once i select start point and goal point. I want to use this tool to tune my planner. All i need is a publisher who publishes costmap according to a static map using only static and inflation layer.

padhupradheep, 2022-02-22:
> I was trying to write a demo for my own planner, which visualize path once i select start point and goal point. I want to use this tool to tune my planner. 

Then maybe you need to follow this tutorial in our beautiful [documentation](https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2planner_plugin.html) on how to write a new planner plugin. By that way, you can easily have all the costmap information (I assume here want the global costmap).

TaoYibo1866, 2022-02-22:
Thank you, i will try it.

padhupradheep, 2022-02-22:
If there are no further queries, feel free to close the issue.

closed 2022-02-22 by TaoYibo1866