ros-navigation/navigation2 #2830: Not able to link map generated by rtab-map with nav2

LuccaMorosiniGioia, 2022-02-22:
Hello, i'm using rtab-map with a turtlebot2i and gazebo to generate a map of an enviroement which is saved on ~/.ros/rtabmap.db, but i'm unable to link this map to nav2 in order to navigate and control the turtlebot2i. 

I'm not sure on how to proceed, should i modify some of the params.yaml or is there a way to link nav2 to the rtab-map topic and feed the map in real time to nav2? 

I'm using nav_bringup to launch the nav2 node, specifically the navigation_launch.py. 

# Sytem
Ubuntu 20.04.03
ROS2 - Galactic

SteveMacenski, 2022-02-22:
Does rtab map not have any documentation about a set with Nav? Nav2 isn’t any different wrt the inputs it would take, so any ROS1 example would work here fine translated with Ros2 parameters.

SteveMacenski, 2022-02-22:
I don’t have experience with rtab map in particular, but if you had their a 2D map that could be generated from the 3D one or a pointcloud of the 3D environment, there are a couple of options available to you like the voxel layer or static layers.

LuccaMorosiniGioia, 2022-02-22:
I actually just figured it out how to do it! 

It actually alredy is integrated with nav2, i was just using the wrong launch file. When launching the rtabmap with ros2 launch rtabmap_ros turtlebot3_rgbd.launch.py and launching the nav2 it worked! I was able to visualize the map on nav-rviz and navigate with goal_pose. 

I just need to figure out now if there's a way to navigate while rtab-map is not running, only with the map saved on memory. 

Anyway, thanks for your time and help.

Sincerely, 

Lucca

closed 2022-02-22 by LuccaMorosiniGioia