ros-navigation/navigation2 #2832: How to bring up your own robot

rjrs23, 2022-02-25:
Hello,

First of all kudos to the entire team on the documentation for nav2, I believe it has the best documentation of any ros package!

However I believe there needs to be a "Run Nav2 on Your Robot Guide" I have followed the First time robot setup guide but am still unclear on how can I have my on robot run nav2.

Forgive me if this is obvious but I am new to ros and am unable to figure this out based on the documentation on the nav2 documentation alone.

What I already have setup
1. Topics and launch file so that my robot can listen to cmd_vel and publish odom
2. Necessary transforms between `/odom,` `/baselink` and `/scan`

What I want to achieve
1. SLAM using nav2
2. Ability to provide a goal on rviz and have the robot automatically move there

I am unable to figure out how to do the above. If someone can help me with this (point me in the right direction) I will happily document the same and publish it so that it can help the community. 

Thanks!

padhupradheep, 2022-02-25:
Hey, I would suggest you to go through the [nav2_bringup](https://github.com/ros-planning/navigation2/tree/main/nav2_bringup) package that gives you an extensive overview on how to set up the packages for your own robot. `nav2_bringup` uses turtlebot3 as an example. I would particularly ask you to re-visit [First time robot setup guide ](https://navigation.ros.org/setup_guides/index.html) (especially the parts [Setting Up Navigation Plugins](https://navigation.ros.org/setup_guides/algorithm/select_algorithm.html) and [Setting Up Lifecycle and Composition Nodes](https://navigation.ros.org/setup_guides/lifecycle_composition/setup_lifecycle_composition.html)) that can help you to quickly grasp on what is there in the bringup and you can co-relate. Hope this helps.

SteveMacenski, 2022-03-01:
I would also add go through the tutorials: https://navigation.ros.org/tutorials/index.html there are tutorials for everything that you are asking for, as well as the https://navigation.ros.org/getting_started/index.html for rviz

closed 2022-03-01 by SteveMacenski