ros-navigation/navigation2 #2835: Underflow in costmaps lead to segmentation fault in planner server

jonipol, 2022-02-28:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  -  Galactic binaries (ros:galactic - docker)
- Version or commit hash:
  - 1.0.8-1focal.20220203.051427
- DDS implementation:
  - Cyclonedds

#### Steps to reproduce issue
```
1. Have obstacle layer in global costmap with laser scan
2. Move robot near the edge of costmap (I have got this error when robot is near the edge of costmap on position with negative y coordinates.)
3. Planner might crash
```

#### Expected behavior
Costmaps are generated, if goal is given the plan is made, execution continues normally
#### Actual behavior
Planner server crashes to segmentation fault
#### Additional information
So far I have managed to reproduce this crash on simulation on two different worlds and on our physical robot.
[image]

This segmentation fault seems to happen when the offset in costmap_2d.hpp, on [bresenham2D](https://github.com/ros-planning/navigation2/blob/0a961df9cbe50356484877681abfdbeae6ba7b33/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp#L486) function, is updated with offset_b being negative. This can lead to underflow of the unsigned int offset, which then leads to segmentation fault when ```at(offset)``` is called.

```
[planner_server-19] Stack trace (most recent call last) in thread 2892:
[planner_server-19] #20   Object "", at 0xffffffffffffffff, in 
[planner_server-19] #19   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 95, in __clone [0x7f81a1b2a292]
[planner_server-19] #18   Source "/build/glibc-eX1tMB/glibc-2.31/nptl/pthread_create.c", line 477, in start_thread [0x7f81a1899608]
[planner_server-19] #17   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.28", at 0x7f81a1cedde3, in 
[planner_server-19] #16   Source "/usr/include/c++/9/thread", line 195, in _M_run [0x7f81a16bf509]
[planner_server-19]         192: 	{ }
[planner_server-19]         193: 
[planner_server-19]         194: 	void
[planner_server-19]       > 195: 	_M_run() { _M_func(); }
[planner_server-19]         196:       };
[planner_server-19]         197: 
[planner_server-19]         198:     void
[controller_server-18] [WARN] [1645688710.606256448] [local_costmap.local_costmap]: /camera_4/slope_filtered_3d buffer updated in 0.56s, it should be updated every 0.50s.
[planner_server-19] #15   Source "/usr/include/c++/9/thread", line 251, in operator() [0x7f81a16c3837]
[planner_server-19]         248: 	{
[planner_server-19]         249: 	  using _Indices
[planner_server-19]         250: 	    = typename _Build_index_tuple<tuple_size<_Tuple>::value>::__type;
[planner_server-19]       > 251: 	  return _M_invoke(_Indices());
[planner_server-19]         252: 	}
[planner_server-19]         253:       };
[planner_server-19] #14   Source "/usr/include/c++/9/thread", line 244, in _M_invoke<0> [0x7f81a16c8cdd]
[planner_server-19]         241: 	template<size_t... _Ind>
[planner_server-19]         242: 	  typename __result<_Tuple>::type
[planner_server-19]         243: 	  _M_invoke(_Index_tuple<_Ind...>)
[planner_server-19]       > 244: 	  { return std::__invoke(std::get<_Ind>(std::move(_M_t))...); }
[planner_server-19]         245: 
[planner_server-19]         246: 	typename __result<_Tuple>::type
[planner_server-19]         247: 	operator()()
[planner_server-19] #13   Source "/usr/include/c++/9/bits/invoke.h", line 95, in __invoke<std::_Bind<void (nav2_costmap_2d::Costmap2DROS::*(nav2_costmap_2d::Costmap2DROS*, double))(double)> > [0x7f81a16cb062]
[planner_server-19]          92:       using __result = __invoke_result<_Callable, _Args...>;
[planner_server-19]          93:       using __type = typename __result::type;
[planner_server-19]          94:       using __tag = typename __result::__invoke_type;
[planner_server-19]       >  95:       return std::__invoke_impl<__type>(__tag{}, std::forward<_Callable>(__fn),
[planner_server-19]          96: 					std::forward<_Args>(__args)...);
[planner_server-19]          97:     }
[planner_server-19] #12   Source "/usr/include/c++/9/bits/invoke.h", line 60, in __invoke_impl<void, std::_Bind<void (nav2_costmap_2d::Costmap2DROS::*(nav2_costmap_2d::Costmap2DROS*, double))(double)> > [0x7f81a16cc81e]
[planner_server-19]          57:   template<typename _Res, typename _Fn, typename... _Args>
[planner_server-19]          58:     constexpr _Res
[... 88 lines cut ...]
[planner_server-19]         503:   }
[planner_server-19]         504: 
[planner_server-19]         505: 
[planner_server-19] #0    Source "/robot_ws/src/navigation2/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp", line 534, in operator() [0x7f819080fb3e]
[planner_server-19]         531:     }
[planner_server-19]         532:     inline void operator()(unsigned int offset)
[planner_server-19]         533:     {
[planner_server-19]       > 534:       costmap_[offset] = value_;
[planner_server-19]         535:     }
[planner_server-19]         536: 
[planner_server-19]         537:   private:
[planner_server-19] Segmentation fault (Address not mapped to object [0x7f828ca16658])
[ERROR] [planner_server-19]: process has died [pid 1629, exit code -11, cmd '/robot_ws/install/nav2_planner/lib/nav2_planner/planner_server --ros-args --log-level info --ros-args -r __node:=planner_server --params-file /tmp/launch_params_up3n6wv8 --params-file /tmp/tmppcmyy2f9 -r /tf:=tf -r /tf_static:=tf_static -r /tf:=tf -r /tf_static:=tf_static'].
```

For this issue I created a quick fix simply by adding a if statement to lines [494](https://github.com/ros-planning/navigation2/blob/0a961df9cbe50356484877681abfdbeae6ba7b33/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp#L494) and [502](https://github.com/ros-planning/navigation2/blob/0a961df9cbe50356484877681abfdbeae6ba7b33/nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp#L502). I did not yet have time to investigate this further and did not have chance to test this on rolling or with turtlebot for example.
```
unsigned int costmap_size = size_x_ * size_y_;
...
    if (costmap_size > offset) {
      at(offset);
    }
``` 

Please let me know if more information, configurations or something is needed.

SteveMacenski, 2022-03-01:
@AlexeyMerzlyakov can you take this one? Looks like another related issue that you're most recently familiar with this part of the codebase

AlexeyMerzlyakov, 2022-03-09:
I've tried to reproduce the problem locally, but so far it's all works. @jonipol, could I ask you to share the results with default Nav2 bringup scripts? This indeed would be greatly helpful for us and allow to reproduce and understand the problem.

jonipol, 2022-03-09:
I'm sorry I have not had time to try this out yet. I will be returning from a work trip next week and will try to allocate some time to investigate this issue as it is really weird one. We just did migration from foxy to galactic, and this issue came up on the galactic branch but did not happen in Foxy one. Will update here once I manage to try this out.

SteveMacenski, 2022-03-17:
@jonipol can you try on `main`? Perhaps this is a bug that we already fixed in Rolling. @AlexeyMerzlyakov did you test with Galactic at tag `1.0.8` https://github.com/ros-planning/navigation2/releases/tag/1.0.8 to see if that's the case?

AlexeyMerzlyakov, 2022-03-18:
> @AlexeyMerzlyakov did you test with Galactic at tag 1.0.8 https://github.com/ros-planning/navigation2/releases/tag/1.0.8 to see if that's the case?

Yes, today I've re-built nav2 stack for this tag (in a VM with Galactic installed from deb-packages) and tried to reproduce the problem moving robot to the edge of both: global and local costmaps. In case if robot inside costmap boundaries, nothing happens. In case if robot is slightly out of costmap, appropriate server (controller or planner) reports this, but still no crashes or segfaults. I've also tried to move robot to some edges of global costmaps having negative X or Y coordinates. Still no effect:
[image]

@jonipol, it is not finally clear what does "Move robot near the edge of costmap" mean? Does it mean robot is moving on the costmap using "2D pose esitmate" RViz command or is it physical move: immediate changing robot coordinates in Gazebo?

jonipol, 2022-03-18:
Sorry for unclear explanation. In this case both work, giving pose estimate and navigating the robot near edge of global costmap.

Today I managed to do some digging into this one. Tried the default config for global costmap. Corrected topic and was not able to get the crash. Did some comparisons and more testing. Found out that setting `raytrace_min_range` > 0.0 (0.2 in my case) makes the crash happen for me.

AlexeyMerzlyakov, 2022-03-18:
@jonipol, thank you for pointing out `raytrace_min_range`! Confirmed the problem for TB3 standard simulation on both: Galactic and Rolling releases. I'll continue investigation/debugging on next week and give more details as they will be appeared.

AlexeyMerzlyakov, 2022-03-21:
There are two versions of `raytraceLine()`/`bresenham()` algorithms. One is placed in `nav2_voxel_grid/include/nav2_voxel_grid/voxel_grid.hpp` and used for 3D space voxel grids. Another - in `nav2_costmap_2d/include/nav2_costmap_2d/costmap_2d.hpp` which is used for 2D costmaps. The root cause of this problem in 2D space and is the same as was fixed in #2460 but for 3D version of Bresenham algorithm: incorrect component shift in `length` calculation inside `raytraceLine()` routine, which leads to negative shifts of `costmap_` array for some cases.

So, all related bugfixes were (partially) back-ported from 3D `raytraceLine()` to 2D version of algorithms in #2857. Local verification shown the problem disappeared on my side. @jonipol, could you please check: does the fix proposed works for you?

jonipol, 2022-03-22:
Yes, the proposed fix does solve the issue on hand for me!

closed 2022-03-22 by SteveMacenski