ros-navigation/navigation2 #2844: Exception thrown while trying to run turtlebot simulation with ThetaStarPlanner

NeoKish, 2022-03-10:
## Bug report

The exception is highlighted in the image below

[image]

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic 
- Version or commit hash:
  - 
After executing the command 

`dpkg-query --show "ros-$ROS_DISTRO-nav2-*", `

I see the navfn-planner and smac-planner but not the theta planner. I am not sure why that is happening
[image]

- DDS implementation:
  - Default
 

#### Steps to reproduce issue

```
ros2 launch nav2_bringup tb3_simulation_launch.py params_file:=/path/to/nav2_params.yaml
```
The default navigation2 bringup params file, nav2_params.yaml has been modified to include the theta star planner config parameters

padhupradheep, 2022-03-10:
Could you please try to do `sudo apt-get update` and retry ? As far as I see, I have the theta-star-planner.

NeoKish, 2022-03-10:
@padhupradheep ,Thank you for the suggestion. I tried `sudo apt-get update` and then `sudo apt-get upgrade` but still facing the same issue. 

Yeah I will put up the question on ROS-Answers. Thanks!

SteveMacenski, 2022-03-10:
Did your docs PR resolve the issue? 

Did you install it? If you just installed Nav2, its not currently on the executable list of packages https://github.com/ros-planning/navigation2/blob/main/navigation2/package.xml#L16 so it wouldn't be autoinstalled, you'd need to specifically ask for it via apt.

SteveMacenski, 2022-03-10:
https://github.com/ros-planning/navigation2/commit/643bb0f4912494389be580763784203ec60aeb39 Just added for future releases

NeoKish, 2022-03-11:
@SteveMacenski , yeah after building with the modified package.xml, it did work. Thanks!
[image]