ros-navigation/navigation2 #2847: BT Navigator node segfaults on cleanup

guilyx, 2022-03-16:
Hi there, found a few things while playing around with transitioning `bt_navigator` down and up. I'll open a PR shortly, it seems like for the first bug, `haltAllActions` is called and an empty pointer `root_node` is passed to the function and accessed unsafely. For my fix I returned the function if `root_node` is empty.
After this fixed my issue when transitioning to cleanup, I could deactivate and cleanup the node without any problem, but on configure would crash as we're redeclaring parameters without checking if they were already declared.

I have only tested this for Galactic as this didn't change from Galactic to Rolling, but I'll test on Rolling in a few days.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04/21.04
- ROS2 Version:
  - Galactic
- Version or commit hash: 28dbc9472e49accd646f2a063367ce82318f6eb4 (galactic branch latest?)
- DDS implementation:
  - Default for Galactic (didn't touch anything) so I think Cyclone ?

### On Cleanup Segmentation Fault

My XML paths are correct and the tree is populated correctly for both navigators.

#### Steps to reproduce issue

```
$ ros2 launch nav2_bringup bringup_launch.py
$ ros2 lifecycle set /bt_navigator deactivate
Transitioning successful
$ ros2 lifecycle set /bt_navigator cleanup
... (no prompt)
```

#### Stack Trace
```
[bt_navigator-20] Stack trace (most recent call last):
[bt_navigator-20] #15   Object "/home/ros/workspace/cmr_ros_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator", at 0x5555557783, in 
[bt_navigator-20] #14   Object "/usr/lib/aarch64-linux-gnu/libc.so.6", at 0x7fb7751d4f, in __libc_start_main
[bt_navigator-20] #13   Object "/home/ros/workspace/cmr_ros_ws/install/nav2_bt_navigator/lib/nav2_bt_navigator/bt_navigator", at 0x55555575f7, in 
[bt_navigator-20] #12   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7dab1c3, in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>)
[bt_navigator-20] #11   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7dac7b3, in rclcpp::executors::SingleThreadedExecutor::spin()
[bt_navigator-20] #10   Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7da4f7b, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[bt_navigator-20] #9    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7da3ebb, in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>)
[bt_navigator-20] #8    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7da3aab, in 
[bt_navigator-20] #7    Object "/opt/ros/galactic/lib/librclcpp.so", at 0x7fb7da2f57, in 
[bt_navigator-20] #6    Object "/opt/ros/galactic/lib/librclcpp_lifecycle.so", at 0x7fb7c930c3, in rclcpp::Service<lifecycle_msgs::srv::ChangeState>::handle_request(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<void>)
[bt_navigator-20] #5    Object "/opt/ros/galactic/lib/librclcpp_lifecycle.so", at 0x7fb7c90a17, in std::_Function_handler<void (std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >), std::_Bind<void (rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::*(rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl*, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>))(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<rmw_request_id_t>&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >&&, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >&&)
[bt_navigator-20] #4    Object "/opt/ros/galactic/lib/librclcpp_lifecycle.so", at 0x7fb7c99073, in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::on_change_state(std::shared_ptr<rmw_request_id_t>, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::ChangeState_Response_<std::allocator<void> > >)
[bt_navigator-20] #3    Object "/opt/ros/galactic/lib/librclcpp_lifecycle.so", at 0x7fb7c98c3b, in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::change_state(unsigned char, rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn&)
[bt_navigator-20] #2    Object "/opt/ros/galactic/lib/librclcpp_lifecycle.so", at 0x7fb7c9876f, in rclcpp_lifecycle::LifecycleNode::LifecycleNodeInterfaceImpl::execute_callback(unsigned int, rclcpp_lifecycle::State const&)
[bt_navigator-20] #1    Object "/home/ros/workspace/cmr_ros_ws/install/nav2_bt_navigator/lib/libbt_navigator_core.so", at 0x7fb7b97b67, in nav2_bt_navigator::BtNavigator::on_cleanup(rclcpp_lifecycle::State const&)
[bt_navigator-20] #0    Object "/home/ros/workspace/cmr_ros_ws/install/nav2_behavior_tree/lib/libnav2_behavior_tree.so", at 0x7fb73e9dc0, in nav2_behavior_tree::BehaviorTreeEngine::haltAllActions(BT::TreeNode*)
[bt_navigator-20] Segmentation fault (Address not mapped to object [(nil)])
```

#### Expected behavior

Successfully transitioning to the unconfigured state

#### Actual behavior

Segmentation fault on cleanup

#### Fix

Nothing to do if root node is emtp

### On Configure after On Cleanup crashes too

```
[bt_navigator-20] [INFO] [1647435648.511477811] [bt_navigator]: Configuring
[bt_navigator-20] [ERROR] [1647435648.522595967] []: Caught exception in callback for transition 10
[bt_navigator-20] [ERROR] [1647435648.522746054] []: Original error: parameter 'default_nav_to_pose_bt_xml' has already been declared
[bt_navigator-20] [WARN] [1647435648.522920367] []: Error occurred while doing error handling.
```

SteveMacenski, 2022-03-16:
> There should be a warning somewhere at least ?

 Agreed. Let me take a look at your PR and we can continue from there

guilyx, 2022-03-16:
I'll go back to the debugger tomorrow with the wrong XML path again. I should've seen:

```cpp
if (!loadBehaviorTree(default_bt_xml_filename_)) {
    RCLCPP_ERROR(logger_, "Error loading XML file: %s", default_bt_xml_filename_.c_str());
    return false;
  }
```

```cpp
if (!xml_file.good()) {
    RCLCPP_ERROR(logger_, "Couldn't open input XML file: %s", filename.c_str());
    return false;
  }
```

Which should've failed the transition to activate.

closed 2022-03-21 by SteveMacenski