ros-navigation/navigation2 #2868: Default parameters do not generate a cost map in ROS2 Galactic

fig-shimooka, 2022-03-25:
ROS2 Galactic (amd64) environment running Navigation2.
The default parameters do not generate a local cost map.
https://github.com/ros-planning/navigation2/blob/galactic/nav2_bringup/bringup/params/nav2_params.yaml#L177
Why?

Also, if you run it with the sample parameters from the wiki, a cost map is generated, but it is displayed incorrectly.
```

## wikipedia source
local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 3.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: False
      robot_radius: 0.3
      footprint: "[[0.12, 0.3], [0.12, -0.3],[-0.63, -0.3],[-0.63, 0.3]]"
      rolling_window: true
      width: 4
      height: 4
      resolution: 0.05

```
## default parameter 
[image]

## wikipedia example
[image]

padhupradheep, 2022-03-25:
If you are using Turtlebot3, then just look into this example from the [nav2_bringup](https://github.com/ros-planning/navigation2/blob/eab2aa6df7063d42491da1da9d836d576a22eabb/nav2_bringup/params/nav2_params.yaml#L181)

padhupradheep, 2022-03-25:
> The default parameters do not generate a local cost map.
https://github.com/ros-planning/navigation2/blob/galactic/nav2_bringup/bringup/params/nav2_params.yaml#L177

Atleast for me, I cannot reproduce your problem. Are you sure you are giving the right laser scanner topic ? And are you using Turtlebot3?

SteveMacenski, 2022-03-25:
The default params are only for the default setup, obviously if you have another system, you'll need to change the parameters and whatnot. We have extensive descriptions of configurations on our website https://navigation.ros.org/configuration/index.html

fig-shimooka, 2022-03-27:
I am trying to move it with a robot I built myself, which is not a Turtlebot.

Apparently the origin of the cost map is wrong.
      global_frame: odom
      robot_base_frame: base_link
Is this setting wrong?

```
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 100
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
![Uploading Screenshot from 2022-03-27 19-59-14.png…]()

    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.07
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: laser_1/scan

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False

bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: odom
    bt_loop_duration: 10
    default_server_timeout: 20
    enable_groot_monitoring: True
    groot_zmq_publisher_port: 1666
    groot_zmq_server_port: 1667
[... 244 lines cut ...]
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: False
    simulate_ahead_time: 2.0
    max_rotational_vel: 1.0
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: False

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"   
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200

```

fig-shimooka, 2022-03-27:
The costmap origin improved when rolling_window was set to True.
However, the costmap is still not generate
[image]
d according to the scan result

fig-shimooka, 2022-03-27:
Resolved.
The cause was that the coordinates of the laser_1 frame were negative.
The urdf file was modified and set to the correct height, which improved the situation.

closed 2022-03-27 by SteveMacenski