ros-navigation/navigation2 #2869: The simple action of loading a map with map_server is way to problematic in this version

luisCFsantos, 2022-03-25:
I am using Ros foxy on Ubuntu 20.04 LTS and I cannot load an occupancy grid map using map_server in ROS2.  

First I had to add the folders with the map files to the install directory in Cmake:

```
install(DIRECTORY
    config
    launch
    DESTINATION share/${PROJECT_NAME})
```

Then, I created the "map_server_params.yaml":

```
 # map_server_params.yaml
map_server:
    ros__parameters:
        yaml_filename: "map.yaml" 

```

with the corresponding "map.yaml":

```
image: map.pgm
resolution: 0.050000
origin: [-1.000000, -17.000000, 0.000000]
negate: 0
occupied_thresh: 0.65
free_thresh: 0.196
```

Then, after trying to run it with the conventional way:

`$ ros2 run nav2_map_server map_server __params:="map_server_params.yaml"
`

It appears only this without publishing a map:

```
[INFO] [1648228989.053565939] [map_server]: 
	map_server lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1648228989.053671243] [map_server]: Creating

```

Until I found out this, I do not know the reason,but for sure it would be something useful in the main documentation of this repository.

`$ros2 run nav2_util lifecycle_bringup map_server
`

Then, this happens:

```
$ ros2 run nav2_map_server map_server __params:="map_server_params.yaml"
[INFO] [1648229286.576805688] [map_server]: 
	map_server lifecycle node launched. 
	Waiting on external lifecycle transitions to activate
	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[INFO] [1648229286.576892582] [map_server]: Creating
[INFO] [1648229286.726197837] [map_server]: Configuring
[ERROR] [1648229286.726335883] []: Caught exception in callback for transition 10
[ERROR] [1648229286.726360080] []: Original error: yaml_filename
[WARN] [1648229286.726392818] []: Error occurred while doing error handling.
[FATAL] [1648229286.726416146] [map_server]: Lifecycle node map_server does not have error state implemented
[WARN] [1648229286.727152143] [rcl_lifecycle]: No transition matching 3 found for current state unconfigured
[ERROR] [1648229286.727188760] []: Unable to start transition 3 from current state unconfigured: Transition is not registered., at /tmp/binarydeb/ros-foxy-rcl-lifecycle-1.1.13/src/rcl_lifecycle.c:350
```

Eventually, with some unfriendly approaches I manager to load a map once, and I was never able to repeat it again! In ROS1 version, this operation is performed with one simple command. 

Then I moved on to a launch file. It would be really useful to have a functional launch file example in the repository. I constructed the following launch file. Be aware that I had to install the "nav2_lifecycle_manager" package. 

```
import os
import yaml
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import ExecuteProcess
from launch.substitutions import LaunchConfiguration
import launch_ros.actions

    
def generate_launch_description():

    lifecycle_nodes = ['map_server']
    use_sim_time = True
    autostart = True
    save_map_timeout = 2000
    free_thresh_default = 0.25
    occupied_thresh_default = 0.65

    ld = LaunchDescription()
    
    # Map server
    map_server_config_path = os.path.join(
        get_package_share_directory('agrob_path'),
        'launch',
        'map_server_params.yaml'
    )
    map_server_cmd = Node(
        package='nav2_map_server',
        executable='map_server',
        output='screen',
        parameters=[map_server_config_path])
        
        
    life_cycle_cmd = launch_ros.actions.Node(
            package='nav2_lifecycle_manager',
            executable='lifecycle_manager',
            name='lifecycle_manager',
            output='screen',
            emulate_tty=True,  # https://github.com/ros2/launch/issues/188
            parameters=[{'use_sim_time': use_sim_time},
                                 {'autostart': autostart},
                                 {'node_names': lifecycle_nodes}])
    

    ld.add_action(map_server_cmd)
    ld.add_action(life_cycle_cmd)
    

    return ld
```

I was only able to run it by modifying the YAML file in the INSTALL folder, and give it the absolute path for the "map.yaml" file:

`$ vim ~/dev_ws/install/agrob_path/share/agrob_path/launch/map_server_params.yaml`

```
# map_server_params.yaml
map_server:
    ros__parameters:
        yaml_filename: "/home/agrob/dev_ws/install/agrob_path/share/agrob_path/launch/map.yaml"
```

This was the only way I managed to load an occupancy grid map. I cannot work with the absolute path in YAML files, and for sure I cannot directly change a parameter in the install folder. Even with this way too complex method, I had to restart the node 2/3 times to successfully visualize the map on Rviz.

So, how can i correctly load a map in ROS2?

luisCFsantos, 2022-03-25:
One I discover this, I am going to add the documentation to the repository and and do a push request

SteveMacenski, 2022-03-25:
> This was the only way I managed to load an occupancy grid map. I cannot work with the absolute path in YAML files, and for sure I cannot directly change a parameter in the install folder. 

Set this through your launch file and set it as a launch configuration / pass to the map server so that it is used -- don't set via yaml for that same reason you mention. The launch API can let you find package's shared directory / do anything in the filesystem you might want to do with Python3. 

In general though, please ask on ROS Answers, I'm happy to follow up there but this is more of an assistance item than a bug/feature request.

closed 2022-03-25 by SteveMacenski

luisCFsantos, 2022-03-25:
> > This was the only way I managed to load an occupancy grid map. I cannot work with the absolute path in YAML files, and for sure I cannot directly change a parameter in the install folder.
> 
> Set this through your launch file and set it as a launch configuration / pass to the map server so that it is used -- don't set via yaml for that same reason you mention. The launch API can let you find package's shared directory / do anything in the filesystem you might want to do with Python3.
> 
> e.x. https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/tb3_simulation_launch.py#L85-L89
> 
> In general though, please ask on ROS Answers, I'm happy to follow up there but this is more of an assistance item than a bug/feature request.

Thank for your help. I was using an .yaml file because it was referenced in the the documentation. I will try your suggestion.

I do not fully agree with your final sentence as this is clearly caused by faulty documentation. Even without using the launch file, the process is way too complex when compared to the ROS1 version. The authors specifically indicate the create an YAML file pointing to he traditional "map.yaml".

SteveMacenski, 2022-03-25:
ROS 2 is more complex than ROS 1, our goal isn't to make something "simple" but something reliable and cohesive in architecture. This is the tip of the iceberg in changes to Navigation.

luisCFsantos, 2022-03-25:
> ROS 2 is more complex than ROS 1, our goal isn't to make something "simple" but something reliable and cohesive in architecture. This is the tip of the iceberg in changes to Navigation.

I understand. But is a fact that your documentation is this specific package is poor. I am going to suggest improvements, once I figure this out. map_server is an important package in my ROS1 software, and I need to solve this before moving everything to ros2