ros-navigation/navigation2 #2877: Behavior tree: pause if 'different path solution' found before major local diversion

SteveMacenski, 2022-03-30:
As an extension of https://github.com/ros-planning/navigation2/pull/2802 to be more general to pause not only near the goal, but also pause in other situations that might be adventitious for a robot to wait and see if a situation resolves itself before moving on down a significantly more time consuming solution. Like if a forklift if blocking the path when 80% down an aisleway to wait vs going up and around to another aisle, taking significantly more time. 

https://discourse.ros.org/t/nav2-feedback-required-potential-change-to-default-behavior/24687/5 A discussion on this topic was had here as well, where we want a solution to pausing if there is something in the way of the robot's path and we may want to pause before simply trying to replan around it if that replan involves going down another aisle / "path solution space".

I think there are a number of potential creative solutions to this and I don't want to muddle the waters too much with my gut instinct, but it would be great to get a number of users' experiences and requirements so we can find a solution that meets the most of them.

jginesclavero, 2022-07-12:
Hi @SteveMacenski! I have found this possible feature so interesting. I would like to present our scenario in case it would be helpful. I'm working on a project to guide and walk-together blind people in bigger public spaces, like shopping malls or train stations. The blind user latches our robot, and the robot navigates to the user's desired goal. 
Users tell us that is uncomfortable that the robot tries to change its direction (the robot is replanning) when there are people in its path. This is especially appreciable when the robot has 2 ways to reach its goal, one with less cost than the other. The robot tries to navigate by the less cost way but people are blocking the path, the robot replans and changes its direction (90 degrees) to take another way, but when the robot replans again, the robot tries to navigate to the "original" way. It is produced so quickly and generally, the blockers go away in a few seconds, but the user suffers that its service robot is moving erratically.

Maybe this [video](https://star-robotics.com/videos/video_wb_as.mp4) could show you better the use case.

What is your instinct to solve these situations?

We are thinking of a BT Node to compare the new path length versus the path length of some iterations and if it is bigger than a threshold, wait some seconds... I don't know if this solution is general enough...

SteveMacenski, 2022-07-12:
The BT XML changes proposed here could be part of the solution, but that feels like an optimization to further aid the good behavior you want more than a solution to your desired behavior outright. This feels like something  you may prefer to have a more sparse general route to the goal rather than a strict path to follow such that you don't really replan at all. Then use or develop a predictive local trajectory planner that can handle crowds effectively. I don't know all the techniques that would apply there, but I know there's a subcommunity of researchers working on crowd navigation (and would be curious to what you found).

SteveMacenski, 2025-03-24:
Note: consider using the BT.CPP parallel nodes to compute a bunch of stuff in parallel in another thread

SteveMacenski, 2026-01-29:
Perhaps: compare the next 3-5 meters when doing a path update. If the projection of them is similar + both are valid and reasoanble, do the update. Else, reject the update as an oscillation. We should probably check the integrated / max costs of both paths to compare the "valid and reasonable" element. I figure if it goes down a totally different direction, that should be clear to identify wih geometry. I don't think we're nitpicking small angles, more like total opposite direction (~135 deg+) apart