ros-navigation/navigation2 #2878: Controller server crashes, references Eigen asserts

peredwardsson, 2022-03-31:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 18.04 on Jetson AGX Xavier
- ROS2 Version:
  Foxy, built from source, docker image based on `nvcr.io/nvidia/l4t-base:r32.5.0`
- Version or commit hash:
Git hash: 3e31f5ad7f1e570c1d751c8944c9b61a7a1578f2
- DDS implementation:
  CycloneDDS

#### Steps to reproduce issue
Run the full nav2 stack. Provide all transforms necessary for autonomous movement. Use the following config for controller server.
```
  controller_server:
    ros__parameters:
      use_sim_time: True
      controller_frequency: 10.0
      min_x_velocity_threshold: 0.001
      min_y_velocity_threshold: 0.5
      min_theta_velocity_threshold: 0.001
      progress_checker_plugin: "progress_checker"
      goal_checker_plugin: "goal_checker"
      controller_plugins: ["FollowPath"]
      progress_checker:
        plugin: "nav2_controller::SimpleProgressChecker"
        required_movement_radius: 0.2
        movement_time_allowance: 10.0
      goal_checker:
        plugin: "nav2_controller::SimpleGoalChecker"
        xy_goal_tolerance: 0.25
        yaw_goal_tolerance: 0.25
        stateful: True
      FollowPath:
        plugin: "nav2_regulated_pure_pursuit_controller::RegulatedPurePursuitController"
        desired_linear_vel: 0.8
        max_linear_accel: 2.5
        max_linear_decel: 2.5
        lookahead_dist: 2.0
        min_lookahead_dist: 0.3
        max_lookahead_dist: 3.0
        lookahead_time: 1.5
        rotate_to_heading_angular_vel: 3.0
        transform_tolerance: 0.1
        use_velocity_scaled_lookahead_dist: false
        min_approach_linear_velocity: 0.2
        use_approach_linear_velocity_scaling: false
        max_allowed_time_to_collision_up_to_carrot: 1.0
        use_regulated_linear_velocity_scaling: true
        use_cost_regulated_linear_velocity_scaling: false
        regulated_linear_scaling_min_radius: 0.9
        regulated_linear_scaling_min_speed: 0.5
        use_rotate_to_heading: true
        rotate_to_heading_min_angle: 0.785
        max_angular_accel: 3.2
        cost_scaling_dist: 0.3
        cost_scaling_gain: 1.0
        inflation_cost_scaling_factor: 3.0
```

Use the following local costmap settings:
```
local_costmap:
    local_costmap:
      ros__parameters:
        update_frequency: 5.0
        publish_frequency: 5.0
        use_sim_time: True
        global_frame: map
        robot_base_frame: base_link
        plugins: ["obstacle_layer", "inflation_layer", "static_layer"]
        rolling_window: true
        width: 5
        height: 5
        resolution: 0.05
        footprint: "[ [0.25, 0.5], [0.25, -0.5], [-0.75, -0.38], [-0.75, 0.38] ]"
        use_maximum: True
        inflation_layer:
          cost_scaling_factor: 3.0
          plugin: "nav2_costmap_2d::InflationLayer"
          inflation_radius: 1.0
          inflate_unknown: True
        static_layer:
          plugin: "nav2_costmap_2d::StaticLayer"
          map_subscribe_transient_local: False
          subscribe_to_updates: True
          map_topic: /map
        obstacle_layer:
          plugin: "nav2_costmap_2d::ObstacleLayer"
          enabled: True
          observation_sources: scan_front scan_back
          footprint_clearing_enabled: True
          scan_front:
            topic: /sensors/scan
            clearing: True
            marking: True
            inf_is_valid: True
            expected_update_rate: 0.15
            min_obstacle_height: 0.1
            max_obstacle_height: 2.0
          scan_back:
            topic: /sensors/scan_back
            clearing: True
            marking: True
            inf_is_valid: True
            expected_update_rate: 0.15
            min_obstacle_height: 0.1
            max_obstacle_height: 2.0
        always_send_full_costmap: True
```
Finally, idle, and have some luck.

#### Expected behavior
Nav2 is stable and nice.
#### Actual behavior
Controller crashes due to some eigen issue:
```
[controller_server-13] controller_server: /usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:110: Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::CwiseBinaryOp(const Lhs&, const Rhs&, const BinaryOp&) [with BinaryOp = Eigen::internal::scalar_product_op<double, double>; LhsType = const Eigen::Array<double, -1, -1>; RhsType = const Eigen::Array<double, -1, -1>; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Lhs = Eigen::Array<double, -1, -1>; Eigen::CwiseBinaryOp<BinaryOp, Lhs, Rhs>::Rhs = Eigen::Array<double, -1, -1>]: Assertion `aLhs.rows() == aRhs.rows() && aLhs.cols() == aRhs.cols()' failed.
```

#### Additional information
Eigen is installed via `apt-get install --no-install-recommends libeigen3-dev`. Nav2, like suggested above, is built from source since Jetson and Foxy has a weird relationship. 

The crash is very intermittent, robot is completely idle at time of crashing with nothing particular happening.

Happy to provide more information if necessary.

Aposhian, 2022-03-31:
Could you provide specifically what version of `libeigen3-dev` is installed by running `apt policy libeigen3-dev`?

Aposhian, 2022-03-31:
Also it looks like you are building from the `main` branch (3e31f5ad7f1e570c1d751c8944c9b61a7a1578f2), which shouldn't be compatible with foxy. Maybe try building off the `foxy-devel` branch?

Aposhian, 2022-03-31:
Also, if you are building and running in docker, as I believe you implied, why build and run on `nvcr.io/nvidia/l4t-base:r32.5.0` rather than using `ros:foxy` as a base docker image? There are definitely use-cases for the L4T images, but you likely don't need them to run Nav2. Food for thought. :slightly_smiling_face:

SteveMacenski, 2022-03-31:
There isn't a use of Eigen in Nav2 anywhere in the controller server https://github.com/ros-planning/navigation2/search?q=Eigen unless its a crash deep inside of another library, like TF. In that case, you'd need to provide a traceback. https://navigation.ros.org/tutorials/docs/get_backtrace.html

In general though, from @Aposhian comments, you're doing something very unsupported so my guess is you're having some serious intergration issues that we're not going to be able to solve. In fact, Nav2 shouldn't even build on Foxy at that commit so you've clearly had to make a bunch of custom modifications that we're not going to be able to sift through.

Its possible that you have the wrong version of Eigen installed for REP 2000's requirements for Foxy so its getting the wrong version of the Eigen dynamically linked library and causing this runtime crash. Given you're starting with some Nvidia image, if it comes with Eigen pre-installed, that's probably it and there's no solution that we can offer.

Please follow the dependencies of the ROS distribution you're using via REP 2000 and use branches of packages built for that distribution.

closed 2022-03-31 by SteveMacenski