ros-navigation/navigation2 #2888: Additional Behavior Server Primitives

SteveMacenski, 2022-04-04:
Perhaps provide some basic behaviors like

- [ ] rotate to sp. heading
- [ ] Drive on heading for X distance (recommission backup plugin)
- [ ] assisted teleop
- [ ] "dance" or some example HRI behavior
- [ ] Back out of recent buffer of commands bc bad situation as recovery, then retry with another algo

sathak93, 2022-04-17:
@SteveMacenski  i like to work on rotate to specific heading behaviour. could you elaborate on it.?

SteveMacenski, 2022-04-18:
It would be 
- Making a behavior to have the robot rotate to a desired relative heading (e.g. +90 or -90 from current orientation)

sathak93, 2022-04-19:
@SteveMacenski Why cant we use spin behaviour for this? Spin can rotate the robot to particular degree from current pose.

SteveMacenski, 2022-04-19:
Those are global angles, not relative angles

jwallace42, 2022-04-20:
I think adding a BT node for spin without increasing the recovery counter would be useful.

SteveMacenski, 2022-04-21:
Maybe a similar analog to your backup --> drive on heading conversion?

jwallace42, 2022-04-21:
The changes for the drive on heading added the ability to drive forward and retained the backup action. 

I guess there is no new functionality that we would need to support(I might be missing something though!).

SteveMacenski, 2022-06-13:
The wiggle behavior can be made by concatenating the spins in the BT with a sequence node. The backout of buffer of commands should be worked on in response to a specific issue. Else, we may prefer to instead have a simple trajectory planner that tries to move a robot into free space instead as a recovery with potential field methods. 

Either way, its something to be readdressed when the need arises. Assisted teleop is covered by another ticket