ros-navigation/navigation2 #2889: hello guys,i am trying to use map_server to laod my map in rviz but i have this error

Tamimi121, 2022-04-05:
[INFO] [launch]: All log files can be found below /home/mh_tamimi/.ros/log/2022-04-05-21-23-56-000140-DESKTOP-O6OJ67J-11637
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [map_server-1]: process started with pid [11639]
[INFO] [lifecycle_manager-2]: process started with pid [11641]
[lifecycle_manager-2] [INFO] [1649183036.200238000] [lifecycle_manager]: Creating
[lifecycle_manager-2] [INFO] [1649183036.211876900] [lifecycle_manager]: Creating and initializing lifecycle service clients
[lifecycle_manager-2] [INFO] [1649183036.214020900] [lifecycle_manager]: Starting managed nodes bringup...
[lifecycle_manager-2] [INFO] [1649183036.214138800] [lifecycle_manager]: Configuring map_server
[map_server-1] [INFO] [1649183036.227771800] [map_server]: 
[map_server-1] 	map_server lifecycle node launched. 
[map_server-1] 	Waiting on external lifecycle transitions to activate
[map_server-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-1] [INFO] [1649183036.227950200] [map_server]: Creating
[map_server-1] [INFO] [1649183036.228188600] [map_server]: Configuring
[map_server-1] [ERROR] [map_io]: Failed processing YAML file turtlebot3_world.yaml at position (-1:-1) for reason: bad file
[map_server-1] [ERROR] [1649183036.228374800] []: Caught exception in callback for transition 10
[map_server-1] [ERROR] [1649183036.228391600] []: Original error: Failed to load map yaml file: turtlebot3_world.yaml
[map_server-1] [WARN] [1649183036.228414800] []: Error occurred while doing error handling.
[map_server-1] [FATAL] [1649183036.228428300] [map_server]: Lifecycle node map_server does not have error state implemented
[map_server-1] [INFO] [map_io]: Loading yaml file: turtlebot3_world.yaml
[lifecycle_manager-2] [ERROR] [1649183036.229795100] [lifecycle_manager]: Failed to change state for node: map_server
[lifecycle_manager-2] [ERROR] [1649183036.229836300] [lifecycle_manager]: Failed to bring up all requested nodes. Aborting bringup.

###

Tamimi121, 2022-04-05:
my launch file 

import os
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.substitutions import LaunchConfiguration
from ament_index_python.packages import get_package_share_directory
from launch_ros.actions import Node

def generate_launch_description():

    # .................. Configurable Arguments .....................

    use_sim_time = True
    map_saver_params_file = 'turtlebot3_world.yaml'
    # ...............................................................

    pkg_dir = get_package_share_directory('nav2_bringup')
    map_save_config = os.path.join(pkg_dir, '/home/mh_tamimi/nav2_ws/src/navigation2/nav2_bringup/bringup/maps', map_saver_params_file)

    return LaunchDescription([

        DeclareLaunchArgument("use_sim_time", default_value=str(use_sim_time), description="Use simulation/Gazebo clock"),
        DeclareLaunchArgument("map_saver_params_file", default_value=map_save_config, description="Map Saver Configuration File"),

        Node(
            package='nav2_map_server',
            executable='map_server',
            name='map_server',
            output='screen',
            emulate_tty=True,  
            parameters=[{'use_sim_time':True},
                        {'yaml_filename': map_saver_params_file}
                        ]),

        Node(
            package='nav2_lifecycle_manager',
            executable='lifecycle_manager',
            name='lifecycle_manager',
            output='screen',
            emulate_tty=True,  
            parameters=[
                {'use_sim_time': LaunchConfiguration('use_sim_time')},
                {'autostart': True},
                {'node_names': ['map_server']}]
        )

    ])

SteveMacenski, 2022-04-05:
> Failed processing YAML file turtlebot3_world.yaml at position (-1:-1) for reason: bad file

That's pretty straight forward, either it doesn't exist (wrong directory) or it exists but its not a properly formed file. That's without a directory path so its hard to tell if that would be the right place, where you launch from matters as well.

Tamimi121, 2022-04-05:
Thank you brother for your quick reply,yes i gave the (map_saver_params_file) full bath and every thing is good <3

Tamimi121, 2022-04-05:
@SteveMacenski 
Thank you brother for your quick reply,yes i gave the (map_saver_params_file) full bath and every thing is good <3

closed 2022-04-05 by SteveMacenski