ros-navigation/navigation2 #2897: Costmaps header stamps are set to 0

charlielito, 2022-04-07:
## Bug report
`/local_costmap/costmap` or `/global_costmap/costmap` topics header stamps are set to 0. Is this somehow intentional?

- Operating System:
  - Ubuntu 20.04
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - Galactic binaries 1.0.8-1focal.20220211
- DDS implementation:
  - Cyclone

#### Steps to reproduce issue
Launch the nav2 stack as usual then see `/local_costmap/costmap` or `/global_costmap/costmap` topics header and note that they are set to 0.

#### Expected behavior
Header stamps consistent with system time

#### Actual behavior
Header stamps set to 0

#### Additional information
I see that the header stamp is set [here](https://github.com/ros-planning/navigation2/blob/30b405c58e6d53ba8c96381416bc4679d35a1483/nav2_costmap_2d/src/costmap_2d_publisher.cpp#L129) to 

```c++
  grid_->header.stamp = rclcpp::Time();
```
which AFAIK is the reason it is set to 0.

SteveMacenski, 2022-04-07:
I don't think it is intentional to be `0`, please submit a PR to implement with the real time!

charlielito, 2022-04-07:
Done https://github.com/ros-planning/navigation2/pull/2899

closed 2022-04-07 by SteveMacenski

Jos-Meran, 2024-12-17:
Hi there, I have the same problem in ROS2 Foxy. Saw what you did to fix it  (from grid_->header.stamp = rclcpp::Time(); to grid_->header.stamp = clock_->now();), but I can't seem to find the file (costmap_2d_publisher.cpp) where the header stamp is defined. I don't know much about this stuff, but is this something that I can't do locally and has to be fixed directly in the foxy-devel branch of the nav2 package? If so, how could I push this change?

SteveMacenski, 2024-12-17:
Foxy is EOL and has not receive updates for some time

Jos-Meran, 2024-12-17:
Oh, so should I install the package from source and use the main branch?

PR #2899 (2022-04-07): Fix: real clock time to costmaps
## Basic Info
Fixes #2897 

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | #2897  |
| Primary OS tested on | Ubuntu20.04 |
| Robotic platform tested on | Gazebo simulation|

---

#### For Maintainers: 
- [ ] Check that any new parameters added are updated in navigation.ros.org
- [ ] Check that any significant change is added to the migration guide
- [ ] Check that any new features **OR** changes to existing behaviors are reflected in the tuning guide
- [ ] Check that any new functions have Doxygen added
- [ ] Chec [... cut]