ros-navigation/navigation2 #2900: Documentation needed for commander api -  Feedback object

rjrs23, 2022-04-08:
Hi,

Where can we find documentation of what objects /  data can we get from the 

In the commander api [nav_to_pose](https://github.com/ros-planning/navigation2/blob/main/nav2_simple_commander/nav2_simple_commander/example_nav_to_pose.py) example we can get the eta to reach the point using the line below

```Duration.from_msg(feedback.estimated_time_remaining).nanoseconds / 1e9```

Where can we find documentation on what other data can we get from the feedback object?

padhupradheep, 2022-04-08:
The feedback that are available can be found here:

https://github.com/ros-planning/navigation2/blob/main/nav2_msgs/action/NavigateToPose.action

Like the last 5 elements.

You can easily look for the list of available feedback by using the command `ros2 interface show nav2_msgs/action/NavigateToPose ` . Of course, there is still the feedback definition missing. But as said before, the last 5 elements are the feedback and the names of the corresponding elements by itself defines what it does.

rjrs23, 2022-04-08:
Thanks @padhupradheep  this helps.