ros-navigation/navigation2 #2913: [SmacPlanner2D] Unintuitive plan when away from obstacles

doisyg, 2022-04-13:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- ROS2 Version:
  - source
- Version or commit hash:
  - latest
- DDS implementation:
  - Cyclone

#### Steps to reproduce issue

Plan with the following parameters
```
    GridBased:
      plugin: "nav2_smac_planner/SmacPlanner2D"
      tolerance: 0.5                      # tolerance for planning if unable to reach exact pose, in meters
      downsample_costmap: false             # whether or not to downsample the map
      downsampling_factor: 1                # multiplier for the resolution of the costmap layer (e.g. 2 on a 5cm costmap would be 10cm)
      allow_unknown: false                   # allow traveling in unknown space
      max_iterations: -1               # maximum total iterations to search for before failing (in case unreachable), set to -1 to disable
      max_on_approach_iterations: 1000      # maximum number of iterations to attempt to reach goal once in tolerance
      max_planning_time: 5.0                # max time in s for planner to plan, smooth
      motion_model_for_search: "MOORE"      # 2D Moore, Von Neumann
      cost_travel_multiplier: 2.0           # Cost multiplier to apply to search to steer away from high cost areas. Larger values will place in the center of aisles more exactly (if non-`FREE` cost potential field exists) but take slightly longer to compute. To optimize for speed, a value of 1.0 is reasonable. A reasonable tradeoff value is 2.0. A value of 0.0 effective disables steering away from obstacles and acts like a naive binary search A*.
      use_final_approach_orientation: false # Whether to set the final path pose at the goal's orientation to the requested orientation (false) or in line with the approach angle so the robot doesn't rotate to heading (true)
      smoother:
        max_iterations: 1000
        w_smooth: 0.3
        w_data: 0.2
        tolerance: 1e-10
```

#### Expected behavior

Shorter path when there is no cost constraint

#### Actual behavior

Non straight path when there is no cost constraint

For instance:
[image]

#### Additional information

----

SteveMacenski, 2022-04-13:
Even without the costs, that is odd. Though, as part of the README I make pretty clear everything in Smac requires a potential field to be applied, it is important. 

If you can provide me with a minimum example (map, costmap settings, etc) and starting / goal poses you see this happen in, I can take a look.

doisyg, 2022-04-17:
Thanks for your detailed answer, I am slowly looking into all that.
In an attempt to test the behavior on a lot of path (and in order to provide a minimum example), I am looking at the system tests (https://github.com/ros-planning/navigation2/tree/main/nav2_system_tests/src/planning), do you you now how the [example_result.png](https://github.com/ros-planning/navigation2/blob/main/nav2_system_tests/src/planning/example_result.png) visualization was generated ?

SteveMacenski, 2022-04-18:
Via https://github.com/ros-planning/navigation2/blob/main/nav2_system_tests/src/planning/test_planner_random_node.cpp which has its map set in https://github.com/ros-planning/navigation2/blob/bcc9aff6bd0e60f6f887cde91b8d2f23f6a1dfa9/nav2_system_tests/src/planning/CMakeLists.txt#L46 which is set to https://github.com/ros-planning/navigation2/blob/bcc9aff6bd0e60f6f887cde91b8d2f23f6a1dfa9/nav2_system_tests/maps/map.pgm which is the map generated.

doisyg, 2022-04-18:
Thx, could not find visualization related code but it was just a matter of adding 3 publishers (path, start point and end point) and increasing the number of tests to get to the same result in rviz (and adding buffer/history there):
[image]
We can already some weird path above, but this is for the default planner of the random test (NavFn, I believe), not SmacPlanner2D. I'll work on a setup that allow choosing the planner.

SteveMacenski, 2022-04-18:
I'm fairly certain that the top 3 comments on my bulleted list will take care of the issue you mention for Smac2D.

For the thing you show with NavFn, It uses the Navigation Function 2 for expanding the potential field which is smooth, but the backtracing is done seperately which can sometimes do odd things, usually towards the end of paths. Given this is only a binary field, not a costmap, there are a couple of odd "squiggles" but they're only as they're going up to / against hard walls at the start/end of goal so that is probably OK overall. 

Navfn has been around essentially since the start of ROS Navigation and the codebase is relatively complex, I don't think it would be wise to make changes to it based on a benchmarking test if in practice we don't have issue with it

doisyg, 2022-05-03:
Slow progress because of other parallel tasks, but I am currently rewriting a test process directly interacting with the planner_server through its action (and with standard costmap with inflation)

SteveMacenski, 2022-05-04:
I'm not sure what I should be pulling from those images, but always nice to see things visualized! 

I would still say that my bulleted changes to 2D are probably what you want to fix the behavior you mentione

doisyg, 2022-05-05:
Just for showing the difference between the planners and some weird paths with Smac2D. I ll investigate more and report here when I ll have a a bit of time.

SteveMacenski, 2022-05-05:
Is there a couple paths in particular in Smac 2D that you think are odd? At first glance, I'm not sure which you mean.

SteveMacenski, 2022-06-13:
@doisyg I went ahead and made the changes I outlined in the comment. See https://github.com/ros-planning/navigation2/tree/smac_fix2

Let me know if this fixes your issues / comments. I can open a PR and merge it.

doisyg, 2022-06-14:
Thx, I ll try ASAP and report

SteveMacenski, 2022-07-06:
Any thoughts / updates?

SteveMacenski, 2022-07-20:
@doisyg we did some testing (see images in the PR) and I strongly believe this fixes your reported issue, but happy to reopen if it does not fully resolve what you're seeing

doisyg, 2022-07-20:
Great! Sorry about opening the discussion without having the time nor the resources to further investigate it soon enough. I will try to report my conclusions on these changes at some point though.