ros-navigation/navigation2 #2915: Local costmap sometimes contains junk

ErikOrjehag, 2022-04-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.2 LTS
- ROS2 Version:
  - Galactic binaries
- Version or commit hash:
  - dpkg-query --show "ros-$ROS_DISTRO-navigation2"
     ros-galactic-navigation2        1.0.8-1focal.20220211.200406
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Unfortunately I have not found a reliable way to reproduce this issue. I have been able to get the issue by moving the robot rapidly in gazebo with the move tool, but it is not a reliable methods that works every run.

#### Expected behavior

The local costmap samples the global costmap at the robot position.

#### Actual behavior

The robot can run on the golf range for days without any problems but then very rarely (i dont monitor all the time but my gut feeling is that it happens maybe once a week) the local costmap samples the wrong part of the global costmap and the incorrect junk is kept in the local costmap until the robot leaves that area. As the robot moves forward the junk is moving backwards in the local costmap until it disappears over the edge of the local costmap. When turning around and going back to that area the junk is no longer there.

#### Additional information

I don't know why this happens but i've only seen it in this area of the map, which is close to a building. The robot is localizing with RTK GPS and the base station is attached to that building. I've never observed the position to be bad or flicker, but still the problem only seems to appear near that area.

Another theory could be that there is some memory or sampling problem in the local costmap layer code, but i have not been able to find it. Another this worth mentioning is that we have an obstacle on the charge station, but as the robot moves in to charge we update the global costmap and remove it. I don't know if maybe the local costmap could be confused by that. I've not seen a pattern where the noise looks like the charger, it looks more like some random obstacle from a different part of the global map.

Screenshot from 2022-03-13: The yellow cylinders mark actual obstacles that are in the global costmap, but we also see some noise in the local costmap that is not part of the global costmap.
[image]

Screenshot from 2022-04-01: Here we see the actual obstacles to the right, which are also sampled by the local costmap. To the left we have a ghost obstacle in the local costmap that should not be there.
[image]

Screenshot from 2022-04-05: Here we have a lot of noise in the lower left part of the costmap.
[image]

A camera is mounted on the building with the RTK base station pointing at the area where we observe the problems with sampling the local costmap:
[image]

SteveMacenski, 2022-04-14:
> I have been able to get the issue by moving the robot rapidly in gazebo with the move tool

So a non-real move -- I would not expect that to arbitrarily work well since you wouldn't have accurate information about where you are and you're picking up data while you're dragging the robot around without odometry updates.

> The local costmap samples the global costmap at the robot position.

You say this a few different times, which makes me believe you have a misunderstanding about what's going on. The local costmap is NOT sampling the global costmap, they have absolutely no relationship with each other. The local costmap processes raw data just like the global costmap and populates / maintains its own instance of a costmap for use by the appropriate algorithms in the controller server. They do not have to be configured the same as the global costmap and generally do actually have some differences, like rolling/fixed size or even the sensors it contains or its update rate. 

> As the robot moves forward the junk is moving backwards in the local costmap until it disappears over the edge of the local costmap. When turning around and going back to that area the junk is no longer there.

Again, the local costmap has no relationship with the global costmap. When the local costmap leaves a region, it forgets all that information (unless its in the map, then the static layer retains and that gets copied over) but anything from sensor-based observations are lost. That is expected behavior. That doesn't explain what you mean by "junk" but the fact that the "junk" is gone and doesn't come back, is correct.

I don't think you've given me much information to go off of w.r.t. the additional obstacles that are being detected. It might be useful to turn on visualizations of different layers to see who's picking up that data and what it might be representing. If it was just speckle noise, then I'd say you need to filter your sensor data. But those are big blocks of information, which makes me think maybe its picking up on the ground plane because its not a perfectly planar environment and you need to either (1) improve your RPY state estimation so its projected correctly or (2) add more ground height clearance.

ErikOrjehag, 2022-04-19:
@SteveMacenski thank you for the reply, here are some additional clearifications:

> So a non-real move -- I would not expect that to arbitrarily work well since you wouldn't have accurate information about where you are and you're picking up data while you're dragging the robot around without odometry updates.

Non-real move is the only way I have been able to sometimes reproduce the error on purpose. But it happens every now and then on the real robot when running it everyday on the field, and then it is only "real moves" and real data from a real robot. I have not found a reliable way to reproduce it on the real robot though, I have not seen a pattern, it just happens sometimes in that area of the map that is shown in my images but not often.

> When the local costmap leaves a region, it forgets all that information (unless its in the map, then the static layer retains and that gets copied over) but anything from sensor-based observations are lost.

Ah I think the confusion is that i keep saying that the local costmap samples the global costmap, but what I should have said is that the local costmap samples the static layer on /map.

Here are the configuration of the layers:

```yaml
global_costmap:
  global_costmap:
    ros__parameters:
      footprint_padding: 0.4
      footprint: "[ [0.6, 0.8], [0.6, -0.8], [-0.6, -0.8], [-0.6, 0.8] ]"
      update_frequency: 2.0
      publish_frequency: 2.0
      global_frame: map
      map_topic: map
      robot_base_frame: base_link
      use_sim_time: True
      resolution: 0.5
      trinary_costmap: False
      lethal_cost_threshold: 90
      plugins: ["static_layer","inflation_layer"]
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: true
        enabled: true
        subscribe_to_updates: true
        transform_tolerance: 0.1
        map_topic: /map
      inflation_layer:
        plugin: "relox_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 3.0
        cost_scaling_factor: 0.3
        inflate_unknown: false
        inflate_around_unknown: true
      always_send_full_costmap: true

local_costmap:
  local_costmap:
    ros__parameters:
      update_frequency: 2.0
      publish_frequency: 2.0
      global_frame: odom
      robot_base_frame: base_link
      use_sim_time: True
      rolling_window: true
      robot_radius: 0.5
      width: 10
      height: 10
      resolution: 0.1
      trinary_costmap: False
      lethal_cost_threshold: 90
      plugins: ["static_layer", "range_sensor_layer", "inflation_layer"]
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
        map_subscribe_transient_local: true
        enabled: true
        subscribe_to_updates: true
        transform_tolerance: 0.1
        map_topic: /map
      range_sensor_layer:
        plugin: "nav2_costmap_2d::RangeSensorLayer"
        enabled: false  # TODO: Ultrasonics not mounted on the Troxhammar robot yet,
        topics:         #       need to enable this later to see if it works.
          - "/ultrasonic/one"
          - "/ultrasonic/two"
        clear_on_max_reading: True
        input_sensor_type: VARIABLE
      inflation_layer:
        plugin: "relox_costmap_2d::InflationLayer"
        enabled: true
        inflation_radius: 3.0
        cost_scaling_factor: 0.55
        inflate_unknown: false
        inflate_around_unknown: true
      filters: ["speed_filter"]
      speed_filter:
        plugin: "relox_costmap_2d::SpeedFilter"
        enabled: True
        filter_info_topic: "/costmap_filter_info"
        speed_limit_topic: "/speed_limit"
      always_send_full_costmap: true
```

Notice that the ultrasonic "range_sensor_layer" is not enabled because we have not mounted the ultrasonic sensor hardware on the robot yet. So the only thing that should affect the local costmap is the "static_layer" and "inflation_layer" right? That is what makes me think that something in the "static_layer" sampling algorithm might be the problem.

By junk i mean for example in "Screenshot from 2022-04-05" that we can see cost in the local costmap that was once in the /map on top of the charger, but was removed from the /map when the robot approached the charger. Even though it is not in the /map anymore we can still see something that looks like the same obstacle in the local costmap, but it looks to be duplicated and twisted a bit so not exactly in the same spot:

Here is what the charge obstacle looks like before it is removed when the robot goes in to charge:
[image]

The "Screenshot from 2022-04-01" looks like something that could come from this area of the map but it is not easy to tell:
[image]

Here is what part of the /map looks like without the costmap visualization enabled in rviz. 
[image]

The map is not generated automatically with SLAM or anything like that. The boundary is and obstacles are measured in by hand using a gps device that we walk around with on the field and we store the measurements of the obstacles in a json file.

SteveMacenski, 2022-04-21:
What if you remove the range layer and try again, just to make sure that isn't playing a role?

ErikOrjehag, 2022-04-21:
Yes I will try that but it will probably take a week before I see the problem again if it is not fixed :) I think it is very unlikely that the ultrasonics are the problem judging by the shape of the ghost obstacles though. But might as well try just to isolate the problem :)

ErikOrjehag, 2022-04-29:
We did not have the chance to test this yet, can probably do it next week

ErikOrjehag, 2022-05-09:
I've still not had the chance to remove the range sensor layer from the yaml but also have not observed the problem for at least 20 days now so it must be very rare. I only get a notification when the robot drives into a lethal area so it might be that we have junk in the local map without me noticing because I'm not logged into the robot all the time.

SteveMacenski, 2022-05-09:
OK. In the absence of data points, I'm inclined to close this ticket but happy to reopen if we can find some issues to resolve.

closed 2022-05-09 by SteveMacenski

ErikOrjehag, 2022-05-25:
I logged into a robot today and saw this in RViz (the range layer has been removed so now we only have the static and inflation layers)

[image]

SteveMacenski, 2022-05-25:
@DLu have you seen this before? It looks like memory corruption or a wrap around error. @ErikOrjehag while I don't think this is due to your system, what's the memory consumption when this happens (e.g. are you at something like 90%+ utilization)?

Are you dynamically adjusting sizes of that costmap or other parameters around it?

ErikOrjehag, 2022-05-29:
Here is a false positive collision that our error snapshotter captured today:
[image]

Our typical memory consumption is about 5.3 GB of 7.6 GB available and we also have a lot of swap memory available that is useful when building the code.
[image]

I will talk with my colleague on monday to see if we have any special parameters on the costmap.

SteveMacenski, 2022-05-30:
You are also thrashing your CPU, I don’t know if that’s related, but  there are probably issues you’re encountering that you may not have noticed yet if you’re saturating all the cores in your CPU at 90%+ utilization. Ubuntu/Linux are not real-time operating systems and odd things do happen in those situations.