ros-navigation/navigation2 #2917: Nav2 lifecycle manager won't startup nodes

AlexeyMerzlyakov, 2022-04-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu focal 20.04.4
- ROS2 Version:
  - ROS2 rolling built with sources from Apr 13 2022
- Version or commit hash:
  - f1d25e98
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Run `nav2_lifecycle_manager` with any lifecylce node e.g.:
```
ros2 launch nav2_bringup bringup_launch.py map:=${HOME}/nav2_ws/src/navigation2/nav2_bringup/maps/turtlebot3_world.yaml | grep -e Configuring -e Activating
```

#### Expected behavior
Lifecycle nodes are going to "Configured" and "Active" state:
```
$ ros2 launch nav2_bringup bringup_launch.py map:=${HOME}/nav2_ws/src/navigation2/nav2_bringup/maps/turtlebot3_world.yaml | grep -e Configuring -e Activating

[component_container_isolated-1] [INFO] [1650040180.873389080] [lifecycle_manager_localization]: Configuring map_server
[component_container_isolated-1] [INFO] [1650040180.874164522] [map_server]: Configuring
[component_container_isolated-1] [INFO] [1650040180.919901945] [lifecycle_manager_localization]: Configuring amcl
[component_container_isolated-1] [INFO] [1650040180.920120541] [amcl]: Configuring
[component_container_isolated-1] [INFO] [1650040180.974971983] [lifecycle_manager_localization]: Activating map_server
[component_container_isolated-1] [INFO] [1650040180.975233144] [map_server]: Activating
[component_container_isolated-1] [INFO] [1650040181.081511733] [lifecycle_manager_localization]: Activating amcl
[component_container_isolated-1] [INFO] [1650040181.081762152] [amcl]: Activating
[component_container_isolated-1] [INFO] [1650040181.249707477] [lifecycle_manager_navigation]: Configuring controller_server
[component_container_isolated-1] [INFO] [1650040181.249960165] [controller_server]: Configuring controller interface
[component_container_isolated-1] [INFO] [1650040181.251433253] [local_costmap.local_costmap]: Configuring
[component_container_isolated-1] [INFO] [1650040181.370370000] [lifecycle_manager_navigation]: Configuring smoother_server
[component_container_isolated-1] [INFO] [1650040181.370620758] [smoother_server]: Configuring smoother server
[component_container_isolated-1] [INFO] [1650040181.395300276] [lifecycle_manager_navigation]: Configuring planner_server
[component_container_isolated-1] [INFO] [1650040181.395925354] [planner_server]: Configuring
[component_container_isolated-1] [INFO] [1650040181.395997608] [global_costmap.global_costmap]: Configuring
[component_container_isolated-1] [INFO] [1650040181.443931586] [planner_server]: Configuring plugin GridBased of type NavfnPlanner
[component_container_isolated-1] [INFO] [1650040181.457533277] [lifecycle_manager_navigation]: Configuring recoveries_server
[component_container_isolated-1] [INFO] [1650040181.457913917] [recoveries_server]: Configuring
[component_container_isolated-1] [INFO] [1650040181.475915198] [recoveries_server]: Configuring spin
[component_container_isolated-1] [INFO] [1650040181.487101451] [recoveries_server]: Configuring backup
[component_container_isolated-1] [INFO] [1650040181.495568349] [recoveries_server]: Configuring wait
[component_container_isolated-1] [INFO] [1650040181.506974136] [lifecycle_manager_navigation]: Configuring bt_navigator
[component_container_isolated-1] [INFO] [1650040181.507486913] [bt_navigator]: Configuring
[component_container_isolated-1] [INFO] [1650040181.594729494] [lifecycle_manager_navigation]: Configuring waypoint_follower
[component_container_isolated-1] [INFO] [1650040181.594987080] [waypoint_follower]: Configuring
[component_container_isolated-1] [INFO] [1650040181.610006747] [lifecycle_manager_navigation]: Activating controller_server
[component_container_isolated-1] [INFO] [1650040181.610493051] [controller_server]: Activating
[component_container_isolated-1] [INFO] [1650040181.610563532] [local_costmap.local_costmap]: Activating
```
#### Actual behavior
See nothing in console: Nav2 lifecycle manager will never switch nodes to configured or active state.

#### Problem analysis
Nodes are not going to configured-active states because of `LifecycleManger::startup()` routine is not being executed. This appears due to outer and inner wall timers are belonging to different callback groups:
```
$ cat nav2_lifecycle_manager/src/lifecycle_manager.cpp
...
  init_timer_ = this->create_wall_timer(
    0s,
    [this]() -> void {
      init_timer_->cancel();
      createLifecycleServiceClients();
      if (autostart_) {
        init_timer_ = this->create_wall_timer(
          0s,
[... 80 lines cut ...]
-          },
-          callback_group_);
+      std::lock_guard<std::mutex> lock(created_mutex_);
+      if (created_) {
+        init_timer_->cancel();
+        createLifecycleServiceClients();
+        if (autostart_) {
+          startup();
+        }
       }
-    });
+    },
+    callback_group_);
   auto executor = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();
   executor->add_callback_group(callback_group_, get_node_base_interface());
   service_thread_ = std::make_unique<nav2_util::NodeThread>(executor);
+
+  std::lock_guard<std::mutex> lock(created_mutex_);
+  created_ = true;
 }
 
 LifecycleManager::~LifecycleManager()
```

However, I am not totally sure that this is right solution. For example, I could miss something or we could play with callback groups that will allow us to leave current solution with two wall timers. @robotechvision, @SteveMacenski, do you have any other thoughts on that?

SteveMacenski, 2022-04-18:
I have not run into this, but I'm also using Rolling binaries, not source. The binaries stopped updates a bit ago (not super long, but also its been a few weeks) since Rolling has moved to support 22.04 and releases binaries there for the Humble release in late May.

More context https://github.com/ros-planning/navigation2/issues/2689

AlexeyMerzlyakov, 2022-04-22:
Yes, this is the most probably the problem of newer ROS2 API change. I understood what was resolved in the #2689. And I am not sure that making nested timers belonging to different callback groups will correctly work further. That is why this ticket was initiated with lock-synchronization approach proposed: to cover initially raised in #2689 problem (I've got the same testcase and played with it).

Since currently it is not related to binary-distributed ROS2 Rolling, we do not need to much care about it. If in near (or far) future this problem will appear again, we could return back to it.
I'll also plan to spend some time on WW18-19 for analysis of ROS2 changes and will try to figure-out what caused the basic example from this ticket header has stopped to work. If I will have any update on it, I will also share it here.
Until this time, you can postpone or close this issue.

SteveMacenski, 2022-04-26:
> Yes, this is the most probably the problem of newer ROS2 API change ...  we do not need to much care about it. If in near (or far) future this problem will appear again, we could return back to it.

So this went away for you already? Might be worth spending an hour and figuring out what changed in Rolling to cause that issue and finding out if that's a regression or not. If it is a regression, we need to point that out **ASAP** to OR as part of the Humble testing to see if that was intentional or not.

https://discourse.ros.org/t/branched-humble-call-for-beta-testing/25284 they're starting testing now, I'd try to see if you see it in the Humble setup that is intended for release. 

@clalancette

clalancette, 2022-04-26:
I can reproduce the issue, but I'm not that familiar with what is supposed to be happening.  That said, do feel free to open a new bug against https://github.com/ros2/rclcpp, and we can triage it during the test phase here to see if that is intentional behavior.

SteveMacenski, 2022-05-02:
@AlexeyMerzlyakov can you do this and see if this is intentional behavior? Time is a little short during the testing phase

afrixs, 2022-05-05:
The lock solution doesn't guarantee that shared pointer has been created when `startup();` is called (it could still be called between deleting the lock at the end of the node constructor and creating the shared pointer in `std::make_shared`), so I'm afraid it's not safe.

But I've had recent experiences when executor didn't check for new timers/subscriptions/etc if they are added from different thread. That could be the cause of the problem. Fortunately, rclcpp has a solution for that, you just need to notify the executor. @AlexeyMerzlyakov could you try to reset to the original code and add these after creating the inner timer?
```
auto &executor_gc = service_executor_->get_node_base_interface()->get_notify_guard_condition();
try {
  executor_gc.trigger();
} catch (const rclcpp::exceptions::RCLError & ex) {
  throw std::runtime_error("Failed to notify service executor");
}
```

afrixs, 2022-05-05:
> Maybe, we need to notify inner timer executor from custom callback_group_

I was hoping that the node base interface notifies both callback groups (since it is the node base for both of them), but apparently it does not...

> For handling it, we can use something like std::atomic_bool shared_ptr_initialized_;

Yes, that could solve it - the variable could be set to true in the "outer" timer (since as far as I know there is no "callback" which is fired when the shared_ptr gets created) and the "inner" timer can be extracted out and its callback can be repeated until `shared_ptr_initialized_` is true, i.e. something like this:

```
shared_ptr_init_timer_ = this->create_wall_timer(
    0s,
    [this]() -> void {
      init_timer_->cancel();
      shared_ptr_initialized_.store(true);
    });
if (autostart_) {
  service_startup_timer_ = this->create_wall_timer(
      0s,
      [this]() -> void {
        if (shared_ptr_initialized_.load()) {
          init_timer_->cancel();
          startup();
        }
      },
      callback_group_);
}
```

AlexeyMerzlyakov, 2022-05-06:
> But now comes to mind: we could create the service_thread_ inside the outer timer 

I think this should be the best solution: we are calling one thread executor directly from another. I've checked it on local testcase and in `nav2_lifecycle_manager`. For both cases it works fine. Also, it was checked that this solution does not cause the regression from #2689, so that should work.
Thank you for the solution proposed!
@SteveMacenski, are you OK with such solution in mainstream?

Despite it looks like the fix was found, I still not fully undestand the root cause of the problem. Working on it.

SteveMacenski, 2022-05-06:
That seems like a fine solution to me, though I would like to know why this has happened to understand if this is illustrative of a subtle bug in Humble that would be good to identify and fix ASAP before Humble's full release in 2 weeks. 

If this was an intentional change, then we have our work around we're done. If it was not, it would be good to notify folks

AlexeyMerzlyakov, 2022-05-12:
Some info on the issue. The https://github.com/ros2/rmw_fastrtps/commit/66926c7934429a4b9ec393ed128c8b6dccc8df39 commit just uncovered the hidden problem, but it is not a root cause of it. This commit contains the logic cycles for `RMW_EVENT_OFFERED_QOS_INCOMPATIBLE/RMW_EVENT_REQUESTED_QOS_INCOMPATIBLE` events which appear during the testcase run; so we have an additional delays in the end of `LifecycleManager` (or testcase) constructor when making executor and service thread. Before this commit, all events of "incompatible QoS" type were considered as "Unknown", they deallocated and never handled. So for that case we have "flew" the constructor faster and for some reasons managed to register inner timer with custom executor.

This could be easily confirmed by adding `sleep_for(1000)` milliseconds in the end of the testcase (or `LifecycleManager`) constructor. With this action even before https://github.com/ros2/rmw_fastrtps/commit/66926c7934429a4b9ec393ed128c8b6dccc8df39 testcase fails (which proves that this commit is not a root cause). Moreover, the problem appears on binary versions of Rolling, as well as on Galactic for FastRTPS. On CycloneDDS for all of these ROS2 distributions testcase passes:
```
[leha@leha-PC dev_ws]$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ./build/timer/timer_node 
[INFO] [1652375869.114717023] [timer_node]: Outer timer created
[INFO] [1652375870.116447548] [timer_node]: Hey from outer wall timer
[INFO] [1652375870.116686324] [timer_node]: Inner timer created
^C[INFO] [1652375872.401716100] [rclcpp]: signal_handler(signum=2)
[leha@leha-PC dev_ws]$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ./build/timer/timer_node
[INFO] [1652375876.817119600] [timer_node]: Outer timer created
[INFO] [1652375877.818420836] [timer_node]: Hey from outer wall timer
[INFO] [1652375877.818571282] [timer_node]: Inner timer created
[INFO] [1652375877.818786694] [timer_node]: Hey from nested wall timer
^C[INFO] [1652375880.014118853] [rclcpp]: signal_handler(signum=2)
```

Attaching updated testcase with removed `nav2_util` dependency and added sleep into the end of the testcase constructor: [dev_ws_v2.zip](https://github.com/ros-planning/navigation2/files/8681297/dev_ws_v2.zip).
Should I submit the ticket to the `rmw_fastrtps` and ask whether it is normal behavior or not?

SteveMacenski, 2022-05-12:
Yes, that would be good!

EduPonz, 2022-05-13:
If I may chip in, as @AlexeyMerzlyakov pointed out, it can only be that ros2/rmw_fastrtps@66926c7934429a4b9ec393ed128c8b6dccc8df39 has surfaced and already existing problem, whether that problem lies in nav, rmw_fastrtps, or elsewhere I do not know. I think it'd be great if you could describe more precisely what is not working according to your expectations, what are those, and probably a minimal reproducer to make understanding and debugging easier.

Thanks guys!

AlexeyMerzlyakov, 2022-05-13:
@EduPonz, I'll try to figure out the information you are requesting:
In the following snippet of code both wall timers are being executed when running with CycloneDDS, while only outer one is executed with FastRTPS:
```
  TimerNode(std::string timer_node)
  : Node(timer_node)
  {
    callback_group_ = create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive, false);

    timer1_ = this->create_wall_timer(
      0s,
      [this]() -> void {
        timer1_->cancel();
        RCLCPP_INFO(get_logger(), "Hey from outer wall timer");
        timer2_ = this->create_wall_timer(
          0s,
          [this]() -> void {
            timer2_->cancel();
            RCLCPP_INFO(get_logger(), "Hey from nested wall timer");
          },
          callback_group_);
      });

      executor_ = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();
      executor_->add_callback_group(callback_group_, get_node_base_interface());
      service_thread_ = std::make_unique<std::thread>([&]() {executor_->spin();});

      std::this_thread::sleep_for(std::chrono::milliseconds(1000));
  }
```
Our expectations are to see both timers: outer and nested are being executed as it being when running with CycloneDDS.

The simple repro-case is attached in a zip-archive to my [previous](https://github.com/ros-planning/navigation2/issues/2917#issuecomment-1125243233) message. To execute the testcase: please build `dev_ws` workspace and then run:
```
$ RMW_IMPLEMENTATION=rmw_fastrtps_cpp ./build/timer/timer_node
or
$ RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ./build/timer/timer_node
```

afrixs, 2022-05-13:
@AlexeyMerzlyakov the reason this code does not work is simply that it is not written correctly, i.e. not regarding how executors work. Inside `spin()` method executor repeatedly enters [wait for work](https://github.com/ros2/rclcpp/blob/02802bcc385c3d6d814add93618f3388d44adec7/rclcpp/src/rclcpp/executor.cpp#L750) state, waiting for any events of timers/subscriptions/... **previously** assigned to it. So if you assign a timer while the executor is already waiting for work, the executor doesn't stop waiting, because it didn't wait for new timers to be assigned, just for already existing timers to be fired (which are, in the case of `executor_`, not present, so it technically waits forever).

The executor can be waken up externally using the guard condition, as I first proposed, but seemingly, as you tested it without success, it works only when there is just one executor per node (otherwise maybe only one of the executors gets notified, which might be an issue of rclcpp to be raised).

Another way around is to avoid spinning executors before assigning work to them, which is the solution that finally worked.

AlexeyMerzlyakov, 2022-05-13:
@afrixs, thank you for the explanation. However, I still have some doubts about it. If the executor should never launch the timers registered _after_ it being created, second timer placed in any place after executor creation shouldn't work, e.g. as in following part of code:
```
    callback_group_ = create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive, false);

    timer1_ = this->create_wall_timer(
      0s,
      [this]() -> void {
        timer1_->cancel();
        RCLCPP_INFO(get_logger(), "Hey from outer wall timer");
      });

    executor_ = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();
    executor_->add_callback_group(callback_group_, get_node_base_interface());
    service_thread_ = std::make_unique<std::thread>([&]() {executor_->spin();});

    std::this_thread::sleep_for(std::chrono::milliseconds(1000));

    timer2_ = this->create_wall_timer(
      0s,
      [this]() -> void {
        timer2_->cancel();
        RCLCPP_INFO(get_logger(), "Hey from nested wall timer");
      },
      callback_group_);
```
However, it works: it does not work only if `timer2_` is placed inside `timer1_` lambda, which is strange for me.

> I think that the reason why the code worked with CycloneDDS is that it was a bit faster/slower and timer1_ callback got called before the executor_ [added handles](https://github.com/ros2/rclcpp/blob/02802bcc385c3d6d814add93618f3388d44adec7/rclcpp/src/rclcpp/executor.cpp#L744) inside the wait_for_work method

For that I've added 1 second sleep `std::this_thread::sleep_for(std::chrono::milliseconds(1000));` in the testcase before registering new `timer2_`, to exclude the possibility of timing overlays. This still works on CycloneDDS.

afrixs, 2022-05-13:
@AlexeyMerzlyakov hmm, that's interesting, this leads me to more exploration, finding that the guard condition actually gets called automatically inside the [`create_wall_timer`](https://github.com/ros2/rclcpp/blob/1037822a63330495dcf0de5e8f20544375a5f116/rclcpp/include/rclcpp/create_timer.hpp#L137)/[add_timer](https://github.com/ros2/rclcpp/blob/87d754b2190edf53a3f7b9f95812e5a80313f649/rclcpp/src/rclcpp/node_interfaces/node_timers.cpp#L47) method.

So the difference is that in your last testcase (the one with non-nested `timer2_`) the main node executor does not exist yet (it's created and assigned after the constructor in `main()`), so the guard condition works because the node has only one executor assigned.

MiguelCompany, 2022-05-13:
@afrixs 
> Node having multiple callback groups in multiple executors is unable to notify all of the executors using its guard condition

Sounds exactly like the problem ros2/rclcpp#1640 is trying to fix!

AlexeyMerzlyakov, 2022-05-19:
@afrixs, may I ask you to submit a PR with fix agreed from [this](https://github.com/ros-planning/navigation2/issues/2917#issuecomment-1118725534) comment? I think you as the author have the original right to this.

dcconner, 2022-05-19:
Just noting that I have this same problem in Galactic when I switched to rmw_fastrtps_cpp to test another issue

closed 2022-05-20 by SteveMacenski