ros-navigation/navigation2 #2919: BT_Navigator failing to load Libary when changing Namespace

isiko, 2022-04-18:
## Bug report

**Required Info:**

- Operating System:
  - Docker on Windows, Container running Ubuntu 20.04
- ROS2 Version:
  - Foxy Binaries
- Version or commit hash:
  - ros-foxy-navigation2    0.4.7-1focal.20220307.024942
- DDS implementation:
  - didn't change anything, so probably `eProsima’s Fast DDS`

#### Steps to reproduce issue
Launch file (adjust package names, etc.):
*This is mostly copied from the `nav2_bringup` package, I just added an XML_File for the BT_Navigator and the namespace*
```Python
# Copyright (c) 2018 Intel Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, GroupAction, SetEnvironmentVariable
from launch.conditions import IfCondition
from launch.substitutions import LaunchConfiguration, PythonExpression
from launch_ros.actions import Node
from launch_ros.actions import LoadComposableNodes
from launch_ros.descriptions import ComposableNode
from nav2_common.launch import RewrittenYaml

def generate_launch_description():
    # Get the launch directory
    pkg_nav = get_package_share_directory('field_robot_nav')
    pkg_share = get_package_share_directory('field_robot')

    namespace_str = "robot"

    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')
    autostart = LaunchConfiguration('autostart')
    bt_params_xml = LaunchConfiguration('bt_params')
    params_file = LaunchConfiguration('params_file')
    use_composition = LaunchConfiguration('use_composition')
    container_name = LaunchConfiguration('container_name')

    lifecycle_nodes = ['controller_server',
                       # 'smoother_server',
                       'planner_server',
                       # 'behavior_server',
                           'bt_navigator',
                       'waypoint_follower']

    # Map fully qualified names to relative ones so the node's namespace can be prepended.
    # In case of the transforms (tf), currently, there doesn't seem to be a better alternative
    # https://github.com/ros/geometry2/issues/32
    # https://github.com/ros/robot_state_publisher/pull/30
[... 543 lines cut ...]
        <ComputePathToPose goal="{goal}" path="{path}"/>
      </DistanceController>
      <FollowPath path="{path}"/>
    </PipelineSequence>
  </BehaviorTree>
</root>
```

#### Expected behavior
Nav2 Stack running in the `/robot` namespace

#### Actual behavior
Nav2 Crashing with the following Error:
```
[bt_navigator-3] [INFO] [1650307748.833286215] [bt_navigator]: Configuring
[bt_navigator-3] [ERROR] [1650307749.025385302] []: Caught exception in callback for transition 10
[bt_navigator-3] [ERROR] [1650307749.025654106] []: Original error: Could not load library: libnav2_change_goal_node_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-3] [WARN] [1650307749.025741322] []: Error occurred while doing error handling.
[bt_navigator-3] [FATAL] [1650307749.025977693] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-5] [ERROR] [1650307749.032132248] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-5] [ERROR] [1650307749.032331849] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

#### Additional information
I tried asking on `answers.ros.org`, but my question is pending moderation for some time now without anything happening...

SteveMacenski, 2022-04-18:
Please ask on ROS Answers, that is the correct place.

SteveMacenski, 2022-04-18:
But definitely remove `libnav2_change_goal_node_bt_node` from your param file, the BT nav is stopping because that doesn't exist in Foxy. You must use the appropriate parameter files / libraries for your distribution.

closed 2022-04-18 by SteveMacenski

isiko, 2022-04-18:
> But definitely remove `libnav2_change_goal_node_bt_node` from your param file, the BT nav is stopping because that doesn't exist in Foxy. You must use the appropriate parameter files / libraries for your distribution.

That's why I also posted here: I can't find that lib in my params, the file doesn't even contain the string 'change'

xinxing-max, 2022-08-25:
@SteveMacenski same problem for me. I didn't add this parameter `libnav2_change_goal_node_bt_node` in the param file at all, but still the error was displayed. And it only appeared when I added the namespace. Since this problem is not accidental, and it is not like you said, just remove `libnav2_change_goal_node_bt_node`. But the problem is, there is no such parameter in the param file, why does this error occur?

@isiko Have you made any progress? I'm working on a multi-robot project recently and I'm also having the same problem.

BATAxjh, 2022-09-01:
[lifecycle_manager-13] [INFO] [1662003008.932584225] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1662003008.934698601] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1662003009.187668684] []: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1662003009.190816521] []: Original error: Could not load library: libnav2_transform_availablemap_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-11] [WARN] [1662003009.194177208] []: Error occurred while doing error handling.
[bt_navigator-11] [FATAL] [1662003009.197183521] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-13] [ERROR] [1662003009.202547305] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-13] [ERROR] [1662003009.202713494] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

BATAxjh, 2022-09-01:
in params_file  “nav2_transform_availablemap_condition_bt_node”    change  “ nav2_transform_available_condition_bt_node”    ok!!

nav2_change_goal_node_bt_node   change to "nav2_goal_updater_node_bt_node"

maybe The versions are different

JACOBIN-SCTCS, 2023-02-06:
@SteveMacenski I am facing the same error which @isiko  has pointed out. I am using Ubuntu 20.04 and ros2 foxy

JACOBIN-SCTCS, 2023-02-06:
> @SteveMacenski same problem for me. I didn't add this parameter `libnav2_change_goal_node_bt_node` in the param file at all, but still the error was displayed. And it only appeared when I added the [...]

@xinxing-max  could you solve this issue?

xinxing-max, 2023-02-06:
Hallo JACOB JAMES K,

 yes, I habe solved this issue. I did not use the package navigation2 from deb source any more. I have downloaded the package in my workspace and changed the source code (just check the lib list in the source code). I found the source code of Nav2 from deb source and official code in github were different. You can compare them by yourself. But I use Nav2 galactic now, not foxy anymore. The issues occurred using the global path planner with Nav2 foxy, e.g. global path through the wall :).

Regard’s,
Xin

xinxing-max, 2023-02-06:
@JACOBIN-SCTCS yes, I have solved this issue. I did not use the package navigation2 from deb source any more. I have downloaded the package in my workspace and changed the source code (just check the lib list in the source code). I found the source code of Nav2 between deb source and official code in github were different. You can compare them by yourself. But I use Nav2 galactic now, not foxy anymore. The issues occurred using the global path planner with Nav2 foxy, e.g. global path through the wall :).

JACOBIN-SCTCS, 2023-02-07:
> @JACOBIN-SCTCS yes, I have solved this issue. I did not use the package navigation2 from deb source any more. I have downloaded the package in my workspace and changed the source code (just check [...]

I will try the suggestions you made. Thank you once again.

Cristian-wp, 2023-11-27:
Hi, I am using the Humble branch and I have the same problem...
```
[lifecycle_manager-14] [INFO] [1700832618.874720224] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1700832618.874903793] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1700832618.913140548] [bt_navigator]: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1700832618.913171216] [bt_navigator]: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-11] [WARN] [1700832618.913190111] [bt_navigator]: Error occurred while doing error handling.
[bt_navigator-11] [FATAL] [1700832618.913198917] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-14] [ERROR] [1700832618.913455290] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-14] [ERROR] [1700832618.913479267] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
…
…
…

lifecycle_manager-14] [INFO] [1700820547.300933970] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-14] [INFO] [1700820547.301005364] [lifecycle_manager_navigation]: Configuring controller_server
[controller_server-7] [WARN] [1700820547.301310911] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[controller_server-7] [ERROR] [1700820547.301329534] [controller_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.301832985] [lifecycle_manager_navigation]: Configuring smoother_server
[smoother_server-8] [WARN] [1700820547.302026701] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[smoother_server-8] [ERROR] [1700820547.302045041] [smoother_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.302364218] [lifecycle_manager_navigation]: Configuring planner_server
[planner_server-9] [WARN] [1700820547.302519899] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[planner_server-9] [ERROR] [1700820547.302537385] [planner_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.302776746] [lifecycle_manager_navigation]: Configuring behavior_server
[behavior_server-10] [WARN] [1700820547.302911425] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[behavior_server-10] [ERROR] [1700820547.302927600] [behavior_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
[lifecycle_manager-14] [INFO] [1700820547.303191023] [lifecycle_manager_navigation]: Configuring bt_navigator
[bt_navigator-11] [INFO] [1700820547.303381089] [bt_navigator]: Configuring
[bt_navigator-11] [ERROR] [1700820547.337725521] [bt_navigator]: Caught exception in callback for transition 10
[bt_navigator-11] [ERROR] [1700820547.337762428] [bt_navigator]: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[bt_navigator-11] [WARN] [1700820547.337784641] [bt_navigator]: Error occurred while doing error handling.
[bt_navigator-11] [ERROR] [1700820547.337798562] [bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[lifecycle_manager-14] [ERROR] [1700820547.338181273] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-14] [ERROR] [1700820547.338208415] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[lifecycle_manager-14] [INFO] [1700820593.618853682] [lifecycle_manager_navigation]: Resetting managed nodes...
[lifecycle_manager-14] [INFO] [1700820593.618894026] [lifecycle_manager_navigation]: Deactivating velocity_smoother
[velocity_smoother-13] [WARN] [1700820593.619228674] [rcl_lifecycle]: No transition matching 4 found for current state unconfigured
```
I have got this error following the [Navigating Using GPS Localization](https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html#navigating-using-gps-localization) tutorial...
@xinxing-max can you explain better what have you done please?
At the moment I have install the navigation2 package and all nav2 package as well, plus in my workspace I have the humble branch of the package.

xinxing-max, 2023-11-27:
@Cristian-wp did you build nav2 package in your workspace or just installed in /opt. Now I don't remember it very clearly. I probably know that I compiled the nav2 package in my own workspace and deleted the so shared files that could not be found from the corresponding files in nav2. In your case, you should delete libnav2_are_error_codes_active_condition_bt_node.so in the bt_navigator node file or configuration file or any file like this in bt_navigator of nav2 package.

Cristian-wp, 2023-11-27:
hi @xinxing-max thank you for the fast reply.
I have do both, first I clone the humble branch from repository:
`git clone --recursive https://github.com/ros-planning/navigation2.git -b humble`
Then I install the binary indicated on [Getting Started[¶](https://navigation.ros.org/getting_started/index.html#getting-started)](https://navigation.ros.org/getting_started/index.html#installation).
Doing that I had found that not all required package where installed, so I insall all with:
`sudo apt-get install ros-humble-nav2-*`
But I do not understand where I have do delete that libnav2_are_error_codes_active_condition_bt_node.so, plus is used as config param only in this files [nav2_multirobot_params_all.yam](https://github.com/ros-planning/navigation2/blob/3ed4c2dfa1ef9b31e117ccb5c35486b599e6b97e/nav2_bringup/params/nav2_multirobot_params_all.yaml#L73).

Whit _that I compiled the nav2 package in my own workspace and deleted the so shared files _ do you mean I have to delete them in the install and build share directory?

xinxing-max, 2023-11-27:
@Cristian-wp I was referring to [bt_navigator.cpp](https://github.com/ros-planning/navigation2/blob/3ed4c2dfa1ef9b31e117ccb5c35486b599e6b97e/nav2_bt_navigator/src/bt_navigator.cpp) before. But I just took a look at humble's version. I found that the code in github should be correct. So I think the problem should be with the params yaml file you are referring to. Just make sure that the lib file name in the yaml file is consistent with the name in bt？navigator.cpp, and there should be no errors. 

Of course, if you install two nav2 packages, make sure that the package you use is the one you want to use.

Cristian-wp, 2023-11-27:
@xinxing-max this is the [nav2_params.yaml](https://github.com/ros-planning/navigation2/blob/3ed4c2dfa1ef9b31e117ccb5c35486b599e6b97e/nav2_bringup/params/nav2_params.yaml) that I am using. I have not modify it, at the moment I am only trying to follow the tutorials.
Ok, so I have to check that all le plugin name [here](https://github.com/ros-planning/navigation2/blob/3ed4c2dfa1ef9b31e117ccb5c35486b599e6b97e/nav2_bt_navigator/src/bt_navigator.cpp#L37) are equal to [these ones](https://github.com/ros-planning/navigation2/blob/3ed4c2dfa1ef9b31e117ccb5c35486b599e6b97e/nav2_bringup/params/nav2_params.yaml#L54) right?

Cristian-wp, 2023-11-27:
@xinxing-max I have check the cpp and yaml file and there are some difference...I copy and paste the plugin list in two txt file and use diff command, this is the result:
```
diff -c bt_navigator.txt nav2_params.txt 
*** bt_navigator.txt	2023-11-27 15:02:39.882398665 +0100
--- nav2_params.txt	2023-11-27 15:03:11.982362655 +0100
***************
*** 10,19 ****
      nav2_clear_costmap_service_bt_node
      nav2_is_stuck_condition_bt_node
      nav2_goal_reached_condition_bt_node
-     nav2_initial_pose_received_condition_bt_node
      nav2_goal_updated_condition_bt_node
      nav2_globally_updated_goal_condition_bt_node
      nav2_is_path_valid_condition_bt_node
      nav2_reinitialize_global_localization_service_bt_node
      nav2_rate_controller_bt_node
      nav2_distance_controller_bt_node
--- 10,19 ----
      nav2_clear_costmap_service_bt_node
      nav2_is_stuck_condition_bt_node
      nav2_goal_reached_condition_bt_node
      nav2_goal_updated_condition_bt_node
      nav2_globally_updated_goal_condition_bt_node
      nav2_is_path_valid_condition_bt_node
+     nav2_initial_pose_received_condition_bt_node
      nav2_reinitialize_global_localization_service_bt_node
      nav2_rate_controller_bt_node
      nav2_distance_controller_bt_node
***************
*** 41,47 ****
      nav2_path_longer_on_approach_bt_node
      nav2_wait_cancel_bt_node
      nav2_spin_cancel_bt_node
-     nav2_assisted_teleop_cancel_bt_node
      nav2_back_up_cancel_bt_node
      nav2_drive_on_heading_cancel_bt_node
      nav2_is_battery_charging_condition_bt_node
--- 41,48 ----
      nav2_path_longer_on_approach_bt_node
      nav2_wait_cancel_bt_node
      nav2_spin_cancel_bt_node
      nav2_back_up_cancel_bt_node
+     nav2_assisted_teleop_cancel_bt_node
      nav2_drive_on_heading_cancel_bt_node
      nav2_is_battery_charging_condition_bt_node
+ 
```
Now I am recreating the workspace, after the compilation I try to _take as good the nav2_param.yaml_ and make the bt_navigator.cpp equal to it.

xinxing-max, 2023-11-27:
@Cristian-wp the wired thing was that there was no lib named nav2_are_error_codes_active_condition_bt_node in your yaml file. But wait and see your subsequent results.

Cristian-wp, 2023-11-27:
@xinxing-max I know, but I suspect there is a problem on this branch because @SteveMacenski [reply](https://github.com/ros-planning/navigation2/issues/3986#issuecomment-1826191314) me that: 
_You’re pulling params from the wrong branch._

But as I say, I get my package with:
`git clone --recursive https://github.com/ros-planning/navigation2.git -b humble`

So I really do not know how to do...

xinxing-max, 2023-11-27:
@Cristian-wp so now you used the nav2 in your own workspace? How did you write your source environment in the .bashrc file? I just saw that in your log, the order of the two source environments was wrong.

Cristian-wp, 2023-11-27:
@xinxing-max maybe I have found a clue...  The [tutorial](https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html#requirements) says to use [[navigation2_tutorials](https://github.com/ros-planning/navigation2_tutorials)](https://github.com/ros-planning/navigation2_tutorials) package and in [nav2_no_map_params.yaml](https://github.com/ros-planning/navigation2_tutorials/blob/24ee184246857328c3622b09e31600d6d5bea754/nav2_gps_waypoint_follower_demo/config/nav2_no_map_params.yaml#L41C24-L41C29), there is _nav2_are_error_codes_active_condition_bt_node_

xinxing-max, 2023-11-27:
@Cristian-wp This problem must lie in the source code of your nav2 or the input problem of your param yaml file. If the param file you use was right without that lib plugin. One of my suggestions is that you should completely remove the binary installation of nav2. Another thing you can try is to copy the param yaml file to another file, rename it, and then load it through the launch file. 

Now you have to make sure that the param file used in the launch file you launch is the one you want to load. It is possible that you have been using the default param file in opt, but the file you modified in your workspace has not even been loaded.