ros-navigation/navigation2 #2920: Add parameter to not run any localization in nav2 stack (amcl or slam) to enable run custom localization

hodnajit, 2022-04-19:
## Feature request

#### Feature description
Add possibility to not run localization in nav2 stack. Currently there is param "slam" so default bringup_launch starts either slam or AMCL localization. I would like to experiment with my localization, so I would like to not run slam or amcl.

#### Implementation considerations
I suggest to add a new parameter to bringup_launch, something like localization with default value true.

Thanks a lot, your work is amazing. :)

SteveMacenski, 2022-04-19:
I believe that is a good reason to have your own launch file that launches `navigation_launch.py` + your own SLAM / localization file.  The main bringup entry point is not supposed to be used in all situations by all users, more as a demonstration and sufficient for many users. If you'd like to use another localization method, we totally support that, but I'm not sure doing so in `nav2_bringup` is the right place. 

There's not really any good way of injecting a new file to run instead without a commandline argument for it, which would have to involve the package and the launch file name to find it. At that point, may as well just own the launch file and save yourself the commandline trouble.

padhupradheep, 2022-04-20:
You can checkout the ``neo_nav2_bringup`` from Neobotix for an example. If it anyways could help. We have a customized bringup files for nav2. https://github.com/neobotix/neo_nav2_bringup

closed 2022-04-20 by hodnajit